espressif/esp-idf #10771: Exception / Panic in function localtime_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/time/lcltime_r.c:123

kotyara12, 2023-02-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-espidf @ 3.50000.0 (5.0.0)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

PlatformIO, package Espressif 32 version 6.0.1

### Power Supply used.

USB

### What is the expected behavior?

After upgrading to esp-idf version 5.0.0, the localtime_r function periodically began to panic with a controller reboot. Not constantly, and randomly - once an hour or every few hours. On version 4.4.3 and below, the same code worked absolutely stably.

### What is the actual behavior?

The trace outputs the following:
```
C:\PlatformIO\security_v4225\ota>c:\Users\kotyara12\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-addr2line.exe -pfiaC -e c:\PlatformIO\security_v4225\ota\firmware\firmware.elf 0x400d4590 0x400ecdae 0x40082829 0x40083512 0x400824ca 0x40089985 0x4008f535 0x4008a40e 0x40083db2 0x40083e65 0x40083eea 0x4016631d 0x4015de08 0x40154d94 0x400d8081 0x400d80bd 0x400d8165 0x400d81b6 0x400d8242 0x400d8291 0x400d8e55 0x400d8bbd 0x400e054a
0x400d4590: __wrap_esp_panic_handler at C:/PlatformIO/libs/system/reEsp32/src/reEsp32.cpp:483
0x400ecdae: panic_handler at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system\port/panic_handler.c:192
0x40082829: xt_unhandled_exception at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system\port/panic_handler.c:223
0x40083512: _xt_user_exc at xtensa_vectors.o:?
0x400824ca: panic_abort at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system/panic.c:412
0x40089985: esp_system_abort at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system/esp_system.c:135
0x4008f535: __assert_func at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\newlib/assert.c:78
0x4008a40e: xQueueGenericSend at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel/queue.c:828 (discriminator 2)
0x40083db2: lock_release_generic at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\newlib/locks.c:192
0x40083e65: _lock_release at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\newlib/locks.c:198
0x40083eea: __retarget_lock_release at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\newlib/locks.c:334
0x4016631d: __tz_unlock at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/time/tzlock.c:50
0x4015de08: **localtime_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/time/lcltime_r.c:123**
0x40154d94: malloc_timestr_empty at C:/PlatformIO/libs/system/rStrings/src/rStrings.cpp:148
0x400d8081: rSensorItem::asStringTimeValue(sensor_value_t*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:550
0x400d80bd: rSensorItem::jsonStringTimeValue(sensor_value_t*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:582
0x400d8165: rSensorItem::jsonPartSensorValue(char const*, sensor_value_t*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:701
0x400d81b6: rSensorItem::jsonExtremums(char const*, sensor_extremums_t*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:747
0x400d8242: rSensorItem::jsonValues() at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:848
0x400d8291: rSensorItem::jsonNamedValues() at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:864
0x400d8e55: rSensorX2::getJSON() at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:2155 (discriminator 1)
0x400d8bbd: rSensor::publishData(bool) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:1585
 (inlined by) rSensor::publishData(bool) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:1560
0x400e054a: sensorsTaskExec(void*) at C:\PlatformIO\security_v4225/lib/sensors/sensors.cpp:528
```

### Steps to reproduce.

The function that sometimes crashes:
```
char * malloc_timestr_empty(const char *format, time_t value)
{
  if (value > 0) {
    struct tm timeinfo;
    char buffer[CONFIG_FORMAT_STRFTIME_BUFFER_SIZE];
    memset(&buffer, 0, sizeof(buffer));
    localtime_r(&value, &timeinfo);
    strftime(buffer, sizeof(buffer), format, &timeinfo);
    return malloc_string(buffer);
  } else {
    return malloc_string(CONFIG_FORMAT_EMPTY_DATETIME);
  }
}

char * malloc_string(const char *source) 
{
  if (source) {
    uint32_t len = strlen(source);
    #if USE_ESP_MALLOC
      char *ret = (char*)esp_malloc(len+1);
    #else
      char *ret = (char*)malloc(len+1);
    #endif
    if (ret == nullptr) {
      rlog_e(tagHEAP, "Failed to create string: out of memory!");
      return nullptr;
    }
    memset(ret, 0, len+1);
    strcpy(ret, source);
    return ret;
  };
  return nullptr;
}
```

### Debug Logs.

```plain
C:\PlatformIO\security_v4225\ota>c:\Users\kotyara12\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-addr2line.exe -pfiaC -e c:\PlatformIO\security_v4225\ota\firmware\firmware.elf 0x400d4590 0x400ecdae 0x40082829 0x40083512 0x400824ca 0x40089985 0x4008f535 0x4008a40e 0x40083db2 0x40083e65 0x40083eea 0x4016631d 0x4015de08 0x40154d94 0x400d8081 0x400d80bd 0x400d8165 0x400d81b6 0x400d8242 0x400d8291 0x400d8e55 0x400d8bbd 0x400e054a
0x400d4590: __wrap_esp_panic_handler at C:/PlatformIO/libs/system/reEsp32/src/reEsp32.cpp:483
0x400ecdae: panic_handler at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system\port/panic_handler.c:192
0x40082829: xt_unhandled_exception at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system\port/panic_handler.c:223
0x40083512: _xt_user_exc at xtensa_vectors.o:?
0x400824ca: panic_abort at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system/panic.c:412
0x40089985: esp_system_abort at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system/esp_system.c:135
0x4008f535: __assert_func at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\newlib/assert.c:78
0x4008a40e: xQueueGenericSend at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel/queue.c:828 (discriminator 2)
0x40083db2: lock_release_generic at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\newlib/locks.c:192
0x40083e65: _lock_release at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\newlib/locks.c:198
0x40083eea: __retarget_lock_release at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\newlib/locks.c:334
0x4016631d: __tz_unlock at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/time/tzlock.c:50
0x4015de08: **localtime_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/time/lcltime_r.c:123**
0x40154d94: malloc_timestr_empty at C:/PlatformIO/libs/system/rStrings/src/rStrings.cpp:148
0x400d8081: rSensorItem::asStringTimeValue(sensor_value_t*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:550
0x400d80bd: rSensorItem::jsonStringTimeValue(sensor_value_t*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:582
0x400d8165: rSensorItem::jsonPartSensorValue(char const*, sensor_value_t*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:701
0x400d81b6: rSensorItem::jsonExtremums(char const*, sensor_extremums_t*) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:747
0x400d8242: rSensorItem::jsonValues() at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:848
0x400d8291: rSensorItem::jsonNamedValues() at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:864
0x400d8e55: rSensorX2::getJSON() at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:2155 (discriminator 1)
0x400d8bbd: rSensor::publishData(bool) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:1585
 (inlined by) rSensor::publishData(bool) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:1560
0x400e054a: sensorsTaskExec(void*) at C:\PlatformIO\security_v4225/lib/sensors/sensors.cpp:528
```

KaeLL, 2023-02-15:
Seems related to 2 issues I opened that are supposedly fixed by #10649.

kotyara12, 2023-02-15:
@KaeLL, 
But, as far as I understood, the same problem was in 4.4.3? But I have no problem on 4.4.3, the devices work for 85 or more days without a single failure.
How do I get this fix if it's already in the public domain? The fact is that ESP-IDF is launched from the PlatformIO package and I don’t know how to randomly load something from the ESP-IDF repository

kotyara12, 2023-02-16:
Anyone help????
The device is already rebooting every 2-3 minutes and I can't do anything. I can't rollback this fucking 5.0.0

kotyara12, 2023-02-16:
[image]

kotyara12, 2023-02-16:
```
C:\PlatformIO\village_toilet\ota>c:\Users\kotyara12\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-addr2line.exe -pfiaC -e c:\PlatformIO\village_toilet\ota\firmware\firmware.elf 0x400d6a30 0x400ed022 0x4008243d 0x40083126 0x400820de 0x40089415 0x4008edf9 0x40089ec6 0x400839b2 0x40083a65 0x40083aea 0x4016462d 0x4015c118 0x400dac1d 0x400dadb6 0x400db86d 0x400e14c5 0x400db616 0x400d4fab 0x4017b19d 0x4017b274
0x400d6a30: __wrap_esp_panic_handler at C:/PlatformIO/libs/system/reEsp32/src/reEsp32.cpp:483
0x400ed022: panic_handler at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_system\port/panic_handler.c:192
0x4008243d: xt_unhandled_exception at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_system\port/panic_handler.c:223
0x40083126: _xt_user_exc at xtensa_vectors.o:?
0x400820de: panic_abort at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_system/panic.c:412
0x40089415: esp_system_abort at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_system/esp_system.c:135
0x4008edf9: __assert_func at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\newlib/assert.c:78
0x40089ec6: xQueueGenericSend at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel/queue.c:828 (discriminator 2)
0x400839b2: lock_release_generic at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\newlib/locks.c:192
0x40083a65: _lock_release at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\newlib/locks.c:198
0x40083aea: __retarget_lock_release at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\newlib/locks.c:334
0x4016462d: __tz_unlock at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/time/tzlock.c:50
0x4015c118: localtime_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/time/lcltime_r.c:123
0x400dac1d: rSensorItem::setRawAndConvertedValue(float, float, long long) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:215
0x400dadb6: rSensorItem::setRawValue(float, long long) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:287
0x400db86d: rSensorX2::setRawValues(float, float) at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:2003
0x400e14c5: HTU2x::readRawData() at C:/PlatformIO/libs/sensors/reHTU2x/src/reHTU2x.cpp:400
0x400db616: rSensor::readData() at C:/PlatformIO/libs/sensors/reSensor/src/reSensor.cpp:1496
0x400d4fab: sensorsTaskExec(void*) at c:\PlatformIO\village_toilet/src/sensors.cpp:1177
0x4017b19d: handler_execute at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_event/esp_event.c:137
 (inlined by) esp_event_loop_run at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_event/esp_event.c:582
0x4017b274: esp_event_loop_run_task at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_event/esp_event.c:107 (discriminator 15)
```
Crash again on localtime_r!
But I can't just rule it out!

chipweinberger, 2023-02-16:
What are you using for crash reporting? that looks useful!

kotyara12, 2023-02-17:
> What are you using for crash reporting? that looks useful!
https://kotyara12.ru/iot/remote_esp32_backtrace/

kotyara12, 2023-02-17:
It looks like I can't wait for any help...

imammy-hacomono, 2023-02-19:
I am also experiencing the same problem with v5.01.
I wish you the best of luck in resolving this!

AxelLin, 2023-02-19:
https://github.com/espressif/esp-idf/issues/9441
The fix is in master but not yet available in any stable branches now.

imammy-hacomono, 2023-02-19:
> #9441
> The fix is in master but not yet available in any stable branches now.

Hi, @AxelLin 
When will the new stave branch be released?
Because I would like to use the release branch and not the master.
Also, which is the stable version that avoids this problem?
v4.44?

AxelLin, 2023-02-20:
> > #9441
> > The fix is in master but not yet available in any stable branches now.
> 
> Hi, @AxelLin When will the new stave branch be released? Because I would like to use the release branch and not the master. Also, which is the stable version that avoids this problem? v4.44?

v4.4.4 does not include the fix.
I don't know when the fix will be available for v4.4 and v5.0 branches, I'm just a normal user.
Maybe @ESP-Marius can answer this.

ESP-Marius, 2023-02-20:
> I don't know when the fix will be available for v4.4 and v5.0 branches, I'm just a normal user.

The fix is already merged internally to all branches so it will be available at github the next time those branches are synced.

AxelLin, 2023-02-28:
> > I don't know when the fix will be available for v4.4 and v5.0 branches, I'm just a normal user.
> 
> The fix is already merged internally to all branches so it will be available at github the next time those branches are synced.

This is an important fix for stable trees, but the sync to github is so slow this time.

AndreyRoc, 2023-03-01:
Александр, посмотрите ..\esp-idf-v5.0.1\components\esp_psram\esp32\Kconfig.spiram
Предполагаю что, что-то с памятью.  
config SPIRAM_CACHE_LIBTIME_IN_IRAM
            bool "Put libc's time related functions in IRAM"
            depends on SPIRAM_CACHE_WORKAROUND
            default "y"
            help
                The functions affected by this option are: asctime, asctime_r, ctime, ctime_r, lcltime, lcltime_r,
                gmtime, gmtime_r, strftime, mktime, tzset_r, tzset, time, gettzinfo, systimes, month_lengths,
                timelocal, tzvars, tzlock, tzcalc_limits, and strptime.
                Putting these function in IRAM will allow them to be called when flash cache is disabled
                but it will also reduce the available size of free IRAM for the user application.

ПЫ.СЫ ранее использовался Espressif\esp-idf-v4.4.3\components\esp32\Kconfig 
Я бы начал с sdkconfig, filter "SPI"
P.S. My knowledge is not that strong.
[Kconfig.zip](https://github.com/espressif/esp-idf/files/10864595/Kconfig.zip)

BillBernacchi, 2023-03-03:
Looks like I am having the same problem as above. My call stack has many similarities but is slightly different because we are using IDF 5.0.1 instead of PlatformIO. I have found that if I make the follow changes to the **Component->Config-Log output** in the SDK config file, I no longer get the crash.
- Default log verbosity (Info)
- Maximum log verbosity (Same as default)
I was wonder if anyone else has tried this. This may help with resolving the issue and/or determine if our issue is the same as described above.

ESP-Marius, 2023-03-31:
Fix is already available on release/v5.0: 60f678b736dc60257a5af9ea10d73625d58bb6ee, if the issue still happens with this fix applied then please let us know.

kotyara12, 2023-03-31:
This is good, it remains to understand how to load this into all my PLATFORMIO projects?

ESP-Marius, 2023-03-31:
Sorry, I have no experience with how platform IO uses IDF. You'll probably need to either check if it is possible to use release/v5.0 branch directly in platformIO, or consult their documentation on how to apply a single patch and then apply the fix yourself.

kotyara12, 2023-06-03:
In version 5.0.2 (PlatformIO Espressif 32 (6.3.1)) this is finally fixed. Thank you!

closed 2023-06-03 by kotyara12