espressif/esp-idf #10774: [Crash][v4.4.3][ESP32-S3]  fseek() after SD card reinsertion hits /newlib/libc/stdio/fseeko.c:116

chipweinberger, 2023-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-215-g9f3f82f54b

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

no crash.

### What is the actual behavior?

crashes in fseek.

### Steps to reproduce.

1. boot with SD card inserted
2. mount with sdmmc
3. open file
4. **unplug sd card**
5. unmount vfs
6. **reinsert sd card**
7. re-mount
8. open file
9. **fseek**

### Debug Logs.

```plain
I (50204) pd midi recorder: [Open Existing] try to reopen today's midi file...
I (50204) pd midi recorder: [Open Existing] checking if path exists:
/sd/JAMC/2023/device-derrao/Jmx-A00012-Feb-15-2023-derrao/Jmx-A00012-Feb-15-2023-derrao.mid
I (50214) pd midi recorder: [Open Existing] todays asset exists
I (50214) midi file writer: open existing midi file:
/sd/JAMC/2023/device-derrao/Jmx-A00012-Feb-15-2023-derrao/Jmx-A00012-Feb-15-2023-derrao.mid

I (50224) pd midi reader: open midi file
I (50224) pd xfwrite: open reader. refcount == 1
I (50224) pd midi reader: filesize 586
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42141a44  PS      : 0x00060330  A0      : 0x82141a01  A1      : 0x3fcdb110  
0x42141a44: _fseeko_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fseeko.c:116 (discriminator 14)

A2      : 0x3fcdb978  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3c226ffc  A7      : 0x00060120  A8      : 0x82141a44  A9      : 0x3fcdb0e0  
A10     : 0x00000000  A11     : 0x3fcdb14c  A12     : 0x00060320  A13     : 0x3fca7aac  
A14     : 0x00060323  A15     : 0x00060323  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000064  LBEG    : 0x4201f3e2  LEND    : 0x4201f3f3  LCOUNT  : 0x00000003  
0x4201f3e2: pd_xfFilesize(pd_xfKey_t, pd_error_t*) at /jcommon/files/pd_xfwrite.cpp:208
 (inlined by) pd_xfFilesize(pd_xfKey_t, pd_error_t*) at /jcommon/files/pd_xfwrite.cpp:1196

0x4201f3f3: pd_xfFilesize(pd_xfKey_t, pd_error_t*) at /jcommon/files/pd_xfwrite.cpp:1198

Backtrace: 0x42141a41:0x3fcdb110 0x421419fe:0x3fcdb180 0x4201b0df:0x3fcdb1a0 0x4201f8fb:0x3fcdb1e0 0x4201faa1:0x3fcdb250 0x4201fd28:0x3fcdb280 0x4205f66c:0x3fcdb2f0 0x4205f1b5:0x3fcdb340 0x4207468c:0x3fcdb440 0x42061554:0x3fcdb570 0x420621bd:0x3fcdb5a0 0x420309bb:0x3fcdb620 0x42024985:0x3fcdb6b0
0x42141a41: _fseeko_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fseeko.c:116 (discriminator 14)

0x421419fe: _fseek_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fseek.c:89
 (inlined by) fseek at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fseek.c:99

0x4201b0df: pd_fseek(__sFILE*, long, int, pd_error_t*) at /jcommon/files/pd_file_utils.cpp:575 (discriminator 2)

0x4201f8fb: pd_xfRead(pd_xfKey_t, unsigned char*, unsigned int, pd_error_t*) at /jcommon/files/pd_xfwrite.cpp:1281

0x4201faa1: pd_xfReadItem(pd_xfKey_t, unsigned char*, unsigned int, pd_error_t*) at /jcommon/files/pd_xfwrite.cpp:1328

0x4201fd28: pd_xfReadMagicBytes(pd_xfKey_t, unsigned char*, unsigned char, pd_error_t*) at /jcommon/files/pd_xfwrite.cpp:982 (discriminator 3)

0x4205f66c: pd_readMidiHeader(pd_xfKey_t, pd_error_t*) at /jcommon/midi_file_reader/pd_midi_file_reader_utils.cpp:72

0x4205f1b5: pd_midi_file_reader_open(char const*, pd_error_t*) at /jcommon/midi_file_reader/pd_midi_file_reader.cpp:38
 (inlined by) pd_midi_file_reader_open(char const*, pd_error_t*) at /jcommon/midi_file_reader/pd_midi_file_reader.cpp:206

0x4207468c: _midi_file_writer_open(char const*, unsigned long long, pd_error_t*) at /jcommon/midi_file_writer/pd_midi_file_writer_io.cpp:248

0x42061554: pd_midi_file_writer_open(char const*, unsigned long long, pd_error_t*) at /jcommon/midi_file_writer/pd_midi_file_writer.cpp:28

0x420621bd: pd_midi_recorder_reopen_todays_asset(pd_midi_recorder_t*, midi_msg_t const*, pd_error_t*) at /jcommon/midi_recorder/pd_midi_recorder_utils.cpp:196

0x420309bb: pd_midi_recorder_write_msg(pd_midi_recorder_t*, pd_piano_t*, midi_msg_t const*, bool, pd_error_t*) at /jcommon/midi_recorder/pd_midi_recorder.cpp:135

0x42024985: midi_pipeline_task(void*) at /jcommon/midi/pd_midi_pipeline.cpp:672
```

igrr, 2023-02-16:
`EXCVADDR: 0x00000064` most likely means that the FILE* pointer passed to feel was NULL. Could you please attach the code you are using to reproduce the issue?

chipweinberger, 2023-02-16:
That was the conclusion I am arriving to as well. I will know shortly.

> Could you please attach the code

The code is too large unfortunately. 

I'll update this issue shortly.

chipweinberger, 2023-02-16:
Hmm. New crash this time. 

I call `esp_vfs_fat_sdcard_unmount()` if mounting fails. Perhaps I shouldn't do that. It's causing a crash.

```
I (54217) pd sd card: pd_sdcard_mount() called
I (54217) pd sd card: mount_generic() called. 
I (54217) pd freertos: idle task_wdt disable
I (54217) pd sd card: mount_sdmmc() 
I (54217) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (54227) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (54227) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (54227) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (54227) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (54227) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (55257) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (55257) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (55267) pd xfwrite: xfUnmount
I (55267) pd sd card: calling esp_vfs_fat_sdcard_unmount()
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0xfefefefe  PS      : 0x00060130  A0      : 0x8207d234  A1      : 0x3fcf3100  
A2      : 0x3c16ed90  A3      : 0x3de4ace0  A4      : 0xfefefefe  A5      : 0x3fca7aac  
A6      : 0x3fcf3110  A7      : 0x9609e3c6  A8      : 0x8207d08c  A9      : 0x3fcf30d0  
A10     : 0xfefefefe  A11     : 0x9609e3c6  A12     : 0x3fca7aac  A13     : 0x3fcaafdc  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0xfefefefc  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0x00000000  

Backtrace: 0x7efefefb:0x3fcf3100 0x4207d231:0x3fcf3130 0x4201d5da:0x3fcf3160 0x4201d928:0x3fcf3190 0x4201d9b1:0x3fcf3280 0x4201da2c:0x3fcf32b0 0x4200ca0b:0x3fcf32f0
0x4207d231: esp_vfs_fat_sdcard_unmount at /esp-idf/components/fatfs/vfs/vfs_fat_sdmmc.c:395

0x4201d5da: pd_sdcard_dismount(pd_error_t*) at /jamcorder-firmware/jamcorder_common/jcommon/files/pd_sdcard.cpp:317
 (inlined by) pd_sdcard_dismount(pd_error_t*) at /jamcorder-firmware/jamcorder_common/jcommon/files/pd_sdcard.cpp:597

0x4201d928: mount_generic(bool, pd_error_t*) at /jamcorder-firmware/jamcorder_common/jcommon/files/pd_sdcard.cpp:290 (discriminator 1)

0x4201d9b1: pd_sdcard_mount(pd_error_t*) at /jamcorder-firmware/jamcorder_common/jcommon/files/pd_sdcard.cpp:591

0x4201da2c: timer_detect_handler(void*) at /jamcorder-firmware/jamcorder_common/jcommon/files/pd_sdcard.cpp:96

0x4200ca0b: timer_process_alarm at /esp-idf/components/esp_timer/src/esp_timer.c:360
 (inlined by) timer_task at /esp-idf/components/esp_timer/src/esp_timer.c:386
```

chipweinberger, 2023-02-16:
> most likely means that the FILE* pointer passed to fseek was NULL

Yes this was the issue. I am no longer reproducing the original crash.

closed 2023-02-16 by chipweinberger