espressif/esp-idf #10775: xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(onvif.c.obj):(.literal.__tds__GetServices+0x0): undefined reference to `_impure_ptr'

zcj2010lzh, 2023-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF Release v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

希望贵公司可以帮忙查看此问题原因和提供解决方法
I hope that your company can help to check the cause of this problem and provide solutions

### What is the actual behavior?

希望贵公司可以帮忙查看此问题原因和提供解决方法
I hope that your company can help to check the cause of this problem and provide solutions

### Steps to reproduce.

1.build get_start/hello_world project 
2. the issue  came

### Build or installation Logs.

```plain
PS C:\Espressif\frameworks\esp-idf-v4.4.4\examples\get-started\hello_world> idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\espressif\frameworks\esp-idf-v4.4.4\examples\get-started\hello_world\build
Executing "ninja all"...
[1/6] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4.4\examples\get-started\hello_world\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Espressif/frameworks/esp-idf-v4.4.4/examples/get-started/hello_world/build/bootloader/bootloader.bin"
Bootloader binary size 0x63b0 bytes. 0xc50 bytes (11%) free.
[2/4] Linking CXX executable hello_world.elf
FAILED: hello_world.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\hello_world.elf.rsp -o hello_world.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(onvif.c.obj):(.literal.__tds__GetServices+0x0): undefined reference to `_impure_ptr'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

igrr, 2023-02-16:
> 1.build get_start/hello_world project
> ...
> libmain.a(onvif.c.obj):(.literal.__tds__GetServices+0x0): undefined reference to `_impure_ptr'

It looks like you have some additional file in the project, `onvif.c`? Perhaps you could attach the file to help us reproduce the issue?

zcj2010lzh, 2023-02-17:
Hello,yes,I added some codes in the project .These are all source codes.Link:    https://www.aliyundrive.com/s/CUdTaeo6PrP
The project is for making onvif protocol on esp32-cam.The project can run successfully on linux.when I build it with esp_idf tool.The problem occurs. please help me to fix it.Thnaks