espressif/esp-idf #10777: ESP32-C2 doesn't honor WIFI_PROMIS_FILTER_MASK_FCSFAIL filter mask

doragasu, 2023-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C2-12 Devkit

### Power Supply used.

USB

### What is the expected behavior?

When enabling RX promiscuous mode with WIFI_PROMIS_FILTER_MASK_FCSFAIL filter mask, FCS failed packets are also received.

### What is the actual behavior?

FCS failed packets are never received.

### Steps to reproduce.

Build and flash the following program on an ESP32 chip. It will receive FCS failed packets on channel 1 successfully. Then build and flash the same program on a ESP32-C2 chip. It will not receive FCS failed packets.

```C
#include <string.h>
#include <esp_wifi.h>
#include <esp_log.h>
#include <esp_event.h>
#include <nvs_flash.h>

#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <freertos/event_groups.h>

#define TAG "sniffer"

static void sniffer_cb(void* buf, wifi_promiscuous_pkt_type_t type)
{
	wifi_promiscuous_pkt_t* rx = (wifi_promiscuous_pkt_t*)buf;
	const int len = rx->rx_ctrl.sig_len;
	const uint32_t state = rx->rx_ctrl.rx_state;

	if (0x41 == state) {
		ESP_LOGI(TAG, "Got FCS failed frame, type: %d, len: %d", type, len);
	} else {
		ESP_LOGI(TAG, "Discarding frame with rx_state = 0x%lx", state);
	}
}

static void sniffer_task(void* pvParameters)
{
	wifi_promiscuous_filter_t sniffer_filter = {0};

	sniffer_filter.filter_mask |= WIFI_PROMIS_FILTER_MASK_MISC;
	sniffer_filter.filter_mask |= WIFI_PROMIS_FILTER_MASK_DATA_MPDU;
	sniffer_filter.filter_mask |= WIFI_PROMIS_FILTER_MASK_FCSFAIL;

	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(sniffer_cb));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&sniffer_filter));
	ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
	ESP_ERROR_CHECK(esp_wifi_set_channel(1, 0));
	vTaskDelete(NULL);
}

static void initialise_wifi(void)
{
	ESP_ERROR_CHECK(esp_netif_init());
	ESP_ERROR_CHECK(esp_event_loop_create_default());

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_NULL));
	ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
}

void app_main()
{
	ESP_ERROR_CHECK(nvs_flash_init());
	initialise_wifi();
	xTaskCreate(&sniffer_task, "sniffer_task", 2048, NULL, 10, NULL);
}
```

### Debug Logs.

```plain
On the ESP32-C2 chip during program run you only get the "Discarding frame" messages:

sniffer: Discarding frame with rx_state = 0x0
```

On the ESP32 chip you get these and the FCS failed frame messages:

```
I (18346) sniffer: Got FCS failed frame, type: 2, len: 1740
I (18346) sniffer: Got FCS failed frame, type: 2, len: 3667
I (18386) sniffer: Got FCS failed frame, type: 2, len: 1972
I (18386) sniffer: Got FCS failed frame, type: 2, len: 2165
I (18396) sniffer: Got FCS failed frame, type: 2, len: 3719
I (18396) sniffer: Got FCS failed frame, type: 2, len: 1859
I (18406) sniffer: Got FCS failed frame, type: 2, len: 1517
I (18426) sniffer: Discarding frame with rx_state = 0x0
```
```

zhangyanjiaoesp, 2024-04-02:
@doragasu Sorry to reply late. I have found the root cause, you should to change the XTAL to 26MHz (default is 40MHz) in the menuconfig for ESP32C2. 
[image]

By the way, you can use esptool to check the chip version of your device: `esptool.py -p /dev/ttyUSB0 chip_id`, the output like this:

> esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C2
Chip is ESP32-C2 (revision v1.1)
Features: WiFi, BLE
Crystal is 26MHz
MAC: 10:97:bd:f1:a2:7c
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-C2 has no Chip ID. Reading MAC instead.
MAC: 10:97:bd:f1:a2:7c
Hard resetting via RTS pin...

zhangyanjiaoesp, 2024-04-08:
@doragasu After changing the XTAL to 26MHz, the esp32c2 can sniffer, but the FCS error packet still can't be received, I'm debugging on this, will solve it ASAP.

closed 2024-05-17 by espressif-bot