espressif/esp-idf #10778: ESP-IDF: How to broadcast data from root to node using wifi mesh?

AvinashSinghPES, 2023-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, when we broadcast the data from node to root then it happens easily but when broadcast data from root to node then its take 20 seconds. where is the problem, please tell me.

below shown screenshot
[image]

deviousprophet, 2023-02-21:
You can try this for broadcasting within mesh
```c
esp_err_t mesh_root_send(const mesh_addr_t* dest_addr, const mesh_data_t* data) {
    static const uint8_t broadcast_addr[MAC_ADDR_LEN] = {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};

    if (memcmp(dest_addr, broadcast_addr, 6) == 0) {
        ESP_LOGI(TAG, "Broadcasting!");

        int route_table_size = 0;
        mesh_addr_t route_table[MESH_ROUTE_TABLE_SIZE];
        esp_mesh_get_routing_table((mesh_addr_t*)&route_table, MESH_ROUTE_TABLE_SIZE * 6, &route_table_size);

        for (int i = 0; i < route_table_size; i++) {
            ESP_LOGI(TAG, "Broadcast: Sending to [%d] " MACSTR, i, MAC2STR(route_table[i].addr));
            esp_err_t err = esp_mesh_send(&route_table[i], data, MESH_DATA_P2P, NULL, 0);
            if (ESP_OK != err) {
                ESP_LOGE(TAG, "Send with err code %d %s", err, esp_err_to_name(err));
            }
        }

    } else {
        // Standard P2P
        esp_err_t err = esp_mesh_send(dest_addr, data, MESH_DATA_P2P, NULL, 0);
        if (err != ESP_OK) {
            ESP_LOGE(TAG, "Send with err code %d %s", err, esp_err_to_name(err));
            return err;
        }
    }

    return ESP_OK;
}
```

And btw, avoid using loop like what you did in the screenshot. Always add a little delay in the loop (not only when route_table_size < 10) or only send data when you need it.

AvinashSinghPES, 2023-02-23:
Thanks,
@deviousprophet

zhangyanjiaoesp, 2023-03-13:
@AvinashSinghPES Has the problem been solved？

closed 2023-04-12 by Alvin1Zhang