espressif/esp-idf #10779: CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y, eth program will crash

ZHOUZIYANG-g, 2023-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32 
when I use the exmaple: **examples/ethernet/basic**

I turn on the switch
Found in:(idfv4.4)Component config  --->ESP32-specific  --->SPI RAM config  --->
or (idfv5.0)Component config → ESP PSRAM → Support for external, SPI-connected RAM → SPI RAM config
 Allow .bss segment placed in external memory (NEW)

**If you turn on this switch, the program will crash.
If you turn off this switch, the program will run normally.**

LOG:
.......
[0;32mI (1299) esp_psram: Adding pool of 4092K of PSRAM memory to heap allocator[0m
[0;32mI (1307) spi_flash: detected chip: generic[0m
[0;32mI (1311) spi_flash: flash io: dio[0m
[0;33mW (1315) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (1329) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (1339) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (1349) system_api: Base MAC address is not set[0m
[0;32mI (1349) system_api: read default base MAC address from EFUSE[0m
[0;32mI (1379) esp_eth.netif.netif_glue: 90:38:0c:ae:b9:83[0m
[0;32mI (1379) esp_eth.netif.netif_glue: ethernet attached to netif[0m
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400f5389  PS      : 0x00060030  A0      : 0x800e46bc  A1      : 0x3ffc00a0  
A2      : 0x00000000  A3      : 0x3ffc00f0  A4      : 0x00000000  A5      : 0x003fffff  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x3f800944  A9      : 0x3ffc0090  
A10     : 0x000001fe  A11     : 0x3ffc00f0  A12     : 0xffffffff  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400f5386:0x3ffc00a0 0x400e46b9:0x3ffc00d0 0x400e478c:0x3ffc00f0 0x4008cfcd:0x3ffc0120

ELF file SHA256: 0be6e1776fa63b06

Rebooting...

**How to turn on the switch while ensuring that the program runs normally
THANKS**

Icarus113, 2023-02-16:
It's a null pointer issue, what did you do?

ZHOUZIYANG-g, 2023-02-16:
> It's a null pointer issue, what did you do?
> 
> The error log shows that the CPU is trying to load from a NULL pointer. Maybe you can further check this.

> It's a null pointer issue, what did you do?
> 
> The error log shows that the CPU is trying to load from a NULL pointer. Maybe you can further check this.

Thanks, I'll follow this . and try to find a way to turn on the switch without crashing the program

Alvin1Zhang, 2023-03-07:
Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

ZHOUZIYANG-g, 2023-03-08:
> I've fixed this out. Because I set phy_config.reset_gpio_num to 17, the pin caused a conflict.

closed 2023-03-08 by ZHOUZIYANG-g

Alvin1Zhang, 2023-03-08:
Thanks for sharing the updates.