espressif/esp-idf #10781: ESP32 RS485 Communication Max485

maialbe2823, 2023-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I am using a MAX485 in order to achieve RS485 communication for ESP32 and other devices. However, ESP32 doesn't see any data present but there is. It keeps saying that it has no data on the receiving end while in truth there is.
I have check lots of samples regarding rs485 but I haven't been able to make any of them work. I have a an exact setup as the following:
[image]

Any other information need it please let me know. Thanks

### What is the actual behavior?

It should read all the data available and send as well.

### Steps to reproduce.

[image]

### Build or installation Logs.

```plain
Log monitor keeps saying that there is no data available.
```

alisitsyn, 2023-02-17:
Hello @maialbe2823 ,

The RS485 communication mode is supported by esp-idf driver and tested. Let us start to check from your connection diagram first then we can go ahead. I see on your connection diagram that your `blue board` which is RS485 transceiver is connected incorrectly to your board. For example:  RO - receiver output is connected to TX2, DI - data (transmitter) input is connected to RX2 which is opposite direction. Please check the connection and add your code with the configuration of UART pins your are using to check the functionality of the RS485 interface. Also I would suggest you to use the [RS485 echo example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo_rs485) to check the communication interface.

Please also refer to: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#overview-of-rs485-specific-communication-0ptions

Once the communication issue is resolved we can follow with your Bacnet MS/TP issue. 
Please also take a look to [Modbus serial port driver](https://github.com/espressif/esp-modbus/blob/master/freemodbus/port/portserial_m.c#L320) which can be useful for the Bacnet MS/TP port.

maialbe2823, 2023-02-17:
I’ll start changing the connection and checking everything else. I’ll let you know. Thanks.

maialbe2823, 2023-03-08:
@alisitsyn I have made the changes you mentioned but it did not work.
Below is the UART config that I am using:
#define UART0_TXD           (17)
#define UART0_RXD           (16)
#define UART0_RTS           (18)

#define BUF_SIZE            (1024)
#define RD_BUF_SIZE         (BUF_SIZE)
#define PACKET_READ_TICS    (500 / portTICK_RATE_MS) 

static void rs485_baud_rate_configure(void)
{
    uart_config_t uart0_config = {
        .baud_rate = Baud_Rate,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .rx_flow_ctrl_thresh = 122,
        .source_clk = UART_SCLK_APB,
    };
    ESP_LOGI(TAG, "RS485 start and configure UART.");
    uart_driver_install(UART_NUM_1, BUF_SIZE * 2, 0, 20, &uart_queue, 0);
    // Configure UART parameters
    uart_param_config(UART_NUM_1, &uart0_config);
    uart_set_pin(UART_NUM_1, UART0_TXD, UART0_RXD, UART0_RTS, UART_PIN_NO_CHANGE);
    uart_set_mode(UART_NUM_1, UART_MODE_RS485_HALF_DUPLEX);

    //Reads uart and puts data to FIFO Buffer
    xTaskCreate(uart_event_task, "uart_event_task", 2048, NULL, 12, NULL);
  
}

One thing that I am checking is that ESP32 is not switching my RTS pin which is GPIO 18 from 0 to 3.3v(0-1) so it just gets stuck receiving data. However; even that data I get it is not the full data I am missing a bit and sometimes even more.
How can I tackle this? Thanks.

alisitsyn, 2023-03-16:
Hi @maialbe2823,

Depending on your ESP32 module you are using the pins 16, 17 can be used for other purpose and are not available for UART.
In the ESP32-WROVER modules Pin 16 is used as the chip select for PSRAM, and 17 is used for the clock signal. Please use the original [rs485 echo example](https://github.com/espressif/esp-idf/blob/c77b5752ef9e4e40c2e819d511b1dd7b83be2996/examples/peripherals/uart/uart_echo_rs485/main/rs485_example.c) with the default pins first then you can experiment with the code. The pin 18 should not be a problem if it is not configured for other purpose (peripheral). Please try again with default pins and then prepare the logic analyzer or oscilloscope diagram of RTS, TXD, RXD pins to find the issue on your side.
Thanks.

maialbe2823, 2023-03-16:
I made such changes, however it still works the same as it did with pins 16 and 17. I can read data over and over but it whenever it is processing such data it just says there's none. Also, for the transmitting part it just sends separates bits instead of sending a whole 8 bit frame as I have it. It is like its not capable of sending more than one bit in one single frame. What is that for?

alisitsyn, 2023-03-16:
@maialbe2823,

I just checked this again in order to check if some update made a regression but it works just fine with the v4.4 and later.
Again, please send the following artifacts:
1. oscillogram or logic analyzer picture of your signals, RXD, TXD, RTS, A, B on RS485 transceiver,
2. your connection diagram or a clear picture of connections, 
3. the log for the rs485 echo example. 
4. Check the VCC pin of the RS485 transceiver = 3.3V and the transceiver is 3.3V compatible.
It can work properly if your RS485 transceiver board works correctly. However, it could be broken after your incorrect connection in the previous step. Change the transceiver to new one and check it again later. 
Thanks.

alisitsyn, 2023-03-28:
Hi @maialbe2823,

Could you give some update for this issue? Is this possible to send some picture of the signals from the TXD, RXD, RTS, D+, D- pins with the rs485_echo example?

maialbe2823, 2023-03-28:
The trials I am doing are using a basic oscilloscope and I am getting ESP to send data however it is like if the 8 bytes frame is too slow for other BACnet devices. 
Does the RS485_echo sample supports uart events in order to read data into a buffer and while also have it transmit 8 bytes?
Also, which RS485 transceiver do you recommend? This is the one I am using :https://www.amazon.com/MAX485-Conversion-Transceiver-Development-Accessories/dp/B0B3M456QG/ref=sr_1_5?crid=1P6KU1I7UO9MS&keywords=rs485+transceiver+esp32&qid=1680025014&sprefix=rs485+transceiver+esp32%2Caps%2C107&sr=8-5
Thank you.

alisitsyn, 2023-03-31:
> The trials I am doing are using a basic oscilloscope and I am getting ESP to send data however it is like if the 8 bytes frame is too slow for other BACnet devices.

The simple and chip [logic analyzer](https://www.amazon.com/HiLetgo-Analyzer-Ferrite-Channel-Arduino/dp/B077LSG5P2/ref=zg_bs_5011666011_sccl_1/133-0146753-3168264?psc=1) can help to debug what happened during communication on your side. The RS485 communication has been tested up to 1Mbit. However,  the RS485 direction control is performed in the driver by software to mitigate the HW issues and can cause some delays. This usually is not be a problem for communication. I propose to test your communication interface with simple USB-RS485 adapter + PC first then you can check the communication with your devices.

> Does the RS485_echo sample supports uart events in order to read data into a buffer and while also have it transmit 8 bytes?

The  RS485_echo is very simple and just simply receive the data then sends it back to external interface as a strings. It uses the UART driver in HALF DUPLEX mode and its events and the ring buffers for RX/TX, in some conditions in can include gaps in the transmitted frame which is not a problem for this example. The [uart events example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_events) can be used to monitor what happened.

> Also, which RS485 transceiver do you recommend? Since my Bacnet devices are RS485 5v not 3.3v. This is the one I am using 

The driver you are using is 5V based driver. However it can work normally with VCC = 3.3V for short distance. However it is not recommended to use it with 5V VCC without level shifter with conjunction with ESP32. For better result I would recommend you to use the RS485 driver with galvanic isolation line [ADM2484](https://www.analog.com/media/en/technical-documentation/data-sheets/adm2484e.pdf) which will provide correct level shifting on ESP32 side and the same time will allow to solve many communication problems with you BACnet devices. There are many equivalent IC and design options like [here](https://www.ti.com/lit/ml/slla416c/slla416c.pdf?ts=1680206822451).

maialbe2823, 2023-04-03:
Got it. I'll try the above for the next few days and will let you know. Thanks again.

maialbe2823, 2023-04-29:
I have achieve a correct level shifter. Since this now my Bacnet devices do see sometimes a whole frame package. However, I am not able to keep esp uart working completely right while transmitting. I think that the problem is within the initialization of the uart itself. The receiving is good though as I see the other device's data. But the ESP is not responding in the way is it supposed to as the uart is not working well for transmitting and after sometime it just keeps transmitting and shuts up the tx of the bacnet device.
Any suggestion on how can I fix the uart configuration in order to have working correctly for rs485 protocol?
Thank you.

alisitsyn, 2023-05-02:
@maialbe2823 ,

You can refer to [initialization of serial port for modbus](https://github.com/espressif/esp-modbus/blob/master/freemodbus/port/portserial_m.c#L262) for more information on how to transmit and receive data from your RS485 based devices.
[Here](https://github.com/alisitsyn/modbus_support/blob/feature/driver_collision_detect/rs485_collision_detector/main/rs485_collision_detector.c#L227)  is the simple RS485 driver based on HAL which supports the collision detection.

Hope this can help to resolve your issues.

maialbe2823, 2023-05-02:
Got it, I'll try it out and see what happens. Thank you.

preveen-stack, 2023-05-23:
@maialbe2823 can you update the status. I have been using ESP32 MODBUS-RS485 with Texas Instruments SN65HVD72DR and SN65HVD1785 transceiver from 2020 and had never had any issues

alisitsyn, 2024-04-09:
@maialbe2823,

The issue will be closed because no any update for a long time. Feel free to reopen the issue if you have some update.

closed 2024-04-09 by alisitsyn