espressif/esp-idf #10782: ESP32-C2 / ESP8684 softAP not working / visible

tsVestor, 2023-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3405-g8ba96ab873

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP8684-DevKitM-1-H4

### Power Supply used.

USB

### What is the expected behavior?

The WiFi softAP Example should start an AP which is visible to other devices. The same example works on ESP32-C3 but not on ESP32-C2. Same behaviour with diffrent ESP8684 and also with a ESP8684 based custom board. With ESP32-C3 everything works as expected.

### What is the actual behavior?

The device log shows that the AP is active and no error but the wifi network is not visible to other devices.

### Steps to reproduce.

1. Use ESP8684 dev board
2. set target to esp32c2
3. flash and moniror WiFi softAP Example
4. use any device and try to connect to the AP

### Debug Logs.

```plain
I (379) main_task: Calling app_main()
I (389) wifi softAP: ESP_WIFI_MODE_AP
I (399) pp: pp rom version: de8c800
I (399) net80211: net80211 rom version: de8c800
I (409) wifi:wifi driver task: 3fcb748c, prio:23, stack:6144, core=0
W (409) wifi: [ieee80211_ioctl.c,2100] wifi firmware version: 5b8642d
W (409) wifi: [ieee80211_ioctl.c,2101] wifi certification version: v7.0
W (419) wifi: [ieee80211_ioctl.c,2102] config NVS flash: enabled
W (419) wifi: [ieee80211_ioctl.c,2103] config nano formating: enabled
W (429) wifi: [esf_buf.c,423] Init data frame dynamic rx buffer num: 32
W (429) wifi: [esf_buf.c,430] Init management frame dynamic rx buffer num: 32
W (439) wifi: [esf_buf.c,437] Init management short buffer num: 32
W (449) wifi: [esf_buf.c,455] Init dynamic tx buffer num: 32
W (449) wifi: [esf_buf.c,477] Init static tx FG buffer num: 2
I (459) wifi:Init static rx buffer size: 1600
I (459) wifi:Init static rx buffer num: 10
I (469) wifi:Init dynamic rx buffer num: 32
I (469) wifi_init: rx ba win: 6
I (469) wifi_init: tcpip mbox: 32
I (479) wifi_init: udp mbox: 6
I (479) wifi_init: tcp mbox: 6
I (479) wifi_init: tcp tx win: 5744
I (489) wifi_init: tcp rx win: 5744
I (489) wifi_init: tcp mss: 1440
I (499) wifi_init: WiFi IRAM OP enabled
I (499) wifi_init: WiFi RX IRAM OP enabled
I (509) phy_init: phy_version 210,e134770,Oct 20 2022,15:26:13
W (539) wifi: [hal_mac.c,606] WDEVBBRXHUNG:0xc00000f0, en:1, timer:240(3840 us)
W (539) wifi: [hal_mac.c,608] WDEVBBTXHUNG:0x800000f0, en:1, timer:240(3840 us)
W (539) wifi: [hal_mac.c,610] WDEVBBCCAHUNG:0xf0, en:0, timer:240(3840 us)
W (549) wifi: [hal_mac.c,615] WDEVHUNGRECOVER:0xffff2710, rxHungRecover[en:1, timer:32767 cycles(409 us)], txHungRecover[en:0, timer:10000 cycles(125 us)]
W (559) wifi: [ieee80211.c,474] mode : softAP (08:3a:8d:40:d7:e5)
W (569) wifi: [ieee80211_power.c,111] Total power save buffer number: 16
W (579) wifi: [ieee80211_ets.c,89] Init max length of beacon: 752/752
W (579) wifi: [ieee80211_ets.c,89] Init max length of beacon: 752/752
I (589) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (599) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword channel:1
I (609) main_task: Returned from app_main()
```

### More Information.

I also tried v5.0, v5.1-dev and master without success.

Xiehanxin, 2023-02-17:
hi, I have tried three IDF versions, v5.0, v5.1 and master(9092b9396) but can't reproduce your problem. I set an esp8684 as AP and a ESP32 as a station. The following is my screetshot
[image]
[image]
[image]

tsVestor, 2023-02-19:
Hi @Xiehanxin, Thank you for testing. I tried it again on a new windows device and on wsl ubuntu with ESP-IDF v5.0.1. It still does only work on C3. After rebuilding it for C2, console output looks fine but no wifi visible. I also checked if there is any RF emission, but there isnt any. Same for the station example. Tested with 3 diffrent ESP8684 Modules on stock dev board and custom hardware. Do you have any ideas?

`esp\esp-idf-v5.0.1\examples\wifi\getting_started\softAP>git log -1
commit a4afa44435ef4488d018399e1de50ad2ee964be8 (HEAD, tag: v5.0.1)
Author: Aditya Patwardhan <aditya.patwardhan@espressif.com>
Date:   Wed Feb 15 17:09:42 2023 +0530

    versions: Update version to 5.0.1`
[image]
[image]

larryli, 2023-03-19:
Maybe the cause of the 26Mhz crystal on the ESP8684-MINI/ESP8684-WROOM

JavierPascual, 2023-03-26:
Same bug for me with ESP32-C3 and SoftAP correctly active: it does not appear as visible in my mac, windows or mobile devices.

Xiehanxin, 2023-03-27:
hi, @JavierPascual, could you please show more details, like log, idf version

ustccw, 2023-03-27:
@tsVestor could you please switch to 26MHz XTAL and give it a try.

`idf.py menuconfig` -> `Component config` -> `Hardware Settings` -> `Main XTAL Config` -> `Main XTAL frequency` -> `26 MHz`

JavierPascual, 2023-04-03:
Would be possible to change the title of this bug? Because it exists also in ESP32-S3 stable processors...

ramiws, 2023-04-16:
> hi, @Xiehanxin:
> 
> Tested with: IDF Version "5.0.1" (yesterday) and "master" (today): commit [81e1e65](https://github.com/espressif/esp-idf/commit/81e1e6599553a646a689ad51e32a5d48b34cfec5) (HEAD -> master, origin/master, origin/HEAD)
> 
> I only have "(X) 40 Mhz" XTAL Config option available in menuconfig (ESP32-S3-WROOM-1 module).
> 
> I'm using SoftAP and SoftAP+STA like years ago... but now, the AP SSID is not visible (and neither I can't connect like with a hidden SSID)

If you go back to 5.0 then it works?

SLASH-JEET, 2023-04-20:
> Would be possible to change the title of this bug? Because it exists also in ESP32-S3 stable processors...
hello, Did you manage to solve the issue in S3 becuz I am facing the same issue in my S3 
I have tried it on my custom development board and it works perfectly but when I am trying it on the other S3 based designer it doesn't seem to it can receive and connect to other network but SoftAP not working

Beauxrel, 2023-05-11:
Did you get this solved? Im having the issue now too :(. And i swear it use to work

SLASH-JEET, 2023-05-11:
> Did you get this solved? Im having the issue now too :(. And i swear it use to work

so basically in my case, it was a custom designed PCB which had an issue with antenna trace, it was designed wrong so as soon as I made a cut in antenna trace to break that track it started working

SLASH-JEET, 2023-05-11:
> > > Would be possible to change the title of this bug? Because it exists also in ESP32-S3 stable processors...
> > 
> > 
> > hello, Did you manage to solve the issue in S3 becuz I am facing the same issue in my S3 I have tried it on my custom development board and it works perfectly but when I am trying it on the other S3-based design it doesn't seem to work it can receive and connect to other network but SoftAP not working
> 
> Hi! I still have this horrible bug. Today, I'm going to try again with the 5.0 IDF version... keep you informed!

Do have a custom-designed board or are you using a ready-made ESP32-S3 breakout board?

youxingz, 2023-05-15:
@ustccw 666666 Thanks for your comment, it helps me a lot! I change the frequency from 40MHz to 26 MHz, and it works now!

> @tsVestor could you please switch to 26MHz XTAL and give it a try.
> 
> `idf.py menuconfig` -> `Component config` -> `Hardware Settings` -> `Main XTAL Config` -> `Main XTAL frequency` -> `26 MHz`

Beauxrel, 2023-05-17:
Hey I aactually fixed this by using the 2.4Ghz frequency on my Router.

Beauxrel, 2023-06-28:
I also solved the issue by using the 2.5 ghz band on my router. this was literally all I needed to do

Alvin1Zhang, 2023-10-22:
Thanks for sharing the updates, feel free to reopen.

closed 2023-10-22 by Alvin1Zhang

eavcodes, 2024-02-29:
> @tsVestor could you please switch to 26MHz XTAL and give it a try.
> 
> `idf.py menuconfig` -> `Component config` -> `Hardware Settings` -> `Main XTAL Config` -> `Main XTAL frequency` -> `26 MHz`

For we it was such case. I've changed config to XTAL 26Mhz and Wifi AP works like a charm! Thank you!