espressif/esp-idf #10783: ESP32 RMT loop support

diablo1281, 2023-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I am looking for an answer and can't find it anywhere. I was preparing a driver for buzzer based on RMT.  Seems that ESP32 do not support loop in RMT.
I couldn't find why. Is there a workaround other than using simple PWM instead of RTM?

suda-morris, 2023-02-17:
@diablo1281 
ESP32 RMT does support **infinite loop** transfer. But unlike other esp chips, esp32 can't control the length of the loop. 

To drive a musical buzzer, the hardware at least should has the feature of the so-called `SOC_RMT_SUPPORT_TX_LOOP_COUNT`, unfortunately, esp32 doesn't have that ability.

```
examples/peripherals/rmt/musical_buzzer:
  disable:
    - if: SOC_RMT_SUPPORT_TX_LOOP_COUNT != 1
```

diablo1281, 2023-02-17:
Thanks for answer!

closed 2024-12-18 by suda-morris