espressif/esp-idf #10784: MQTT SSL示例或者MQTT SSL_DS示例，CA证书问题

jiafeng-1, 2023-02-17:
### Is your feature request related to a problem?

## MQTT SSL示例或者MQTT SSL_DS示例，CA证书问题
我想使用SSL/TLS做加密通讯,但是示例都是提前下载好的证书，链接到了执行程序中。

如果我的服务器器证书过期了，这将无法链接mqtts服务器，请问除了将证书写死，如何通过自动更新证书呢？

因为我在MQTTX客户端中并没有提前下载这个证书，也是随时可以通讯的。

### Describe the solution you'd like.

可自动更新CA证书，而不是提前手动下载链接到可执行程序中

AdityaHPatwardhan, 2023-02-20:
Hi @jiafeng-1 The CA certificate usually has a large expiry so it is not supposed to be updated frequently. That is why it has been stored in the esp_secure_cert_partition.
But in a case where the CA certificate is expired and needs replacement then I think that can be managed in the OTA update.

Just to note here, Currently, in the [esp_secure_cert_mgr](https://github.com/espressif/esp_secure_cert_mgr) we don't support any addition to the existing esp_secure_cert partition. but that feature is discussed in the discussion phase internally. Until that any change regarding the CA certificate should be managed through the OTA upgrade.

AdityaHPatwardhan, 2023-03-08:
Hi @jiafeng-1  Any update on this?

mahavirj, 2023-03-20:
@jiafeng-1 

- By default we keep `MBEDTLS_HAVE_TIME_DATE` configuration disabled, please see reasoning here:
https://github.com/espressif/esp-idf/blob/4f0769d2ed074ef770c6432565d6e5610124e9ea/components/mbedtls/Kconfig#L517-L523
- Application can manage the CA certificate separately (and decoupled from application) by storing it in say NVS partition. Its updates then can be managed separately from the main application image

I will go ahead and close the issue. Please feel free to comment if you still have any questions.

closed 2023-03-20 by mahavirj