espressif/esp-idf #10785: Cannot debug in VSCode with ESP32-S3 and built-in JTAG

mbastida123, 2023-02-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

I get an error when trying to debug with esp-idf debug on VSCode.

This is part of my `settings.json`:
```
    "idf.openOcdConfigs": [
        "board/esp32s3-bridge.cfg"
    ],
    "idf.adapterTargetName": "esp32s3"
```
`launch.json`

```
{
    "version": "0.2.0",
    "configurations": [
      {
        "type": "espidf",
        "name": "Launch",
        "request": "launch",
        "debugPort": 9998,
        "logLevel": 4,
        "mode": "auto",
        "verifyAppBinBeforeDebug": false,
        "tmoScaleFactor": 2
      }
    ]
  }
```

If I run `openocd -f board/esp32s3-builtin.cfg` from terminal everything seems to be in order:

> Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : only one transport option; autoselect 'jtag'
> Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
> Info : esp_usb_jtag: capabilities descriptor set to 0x2000
> Warn : Transport "jtag" was already selected
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
> Info : esp_usb_jtag: serial (68:B6:B3:48:87:04)
> Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
> Info : clock speed 40000 kHz
> Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
> Warn : target esp32s3.cpu1 examination failed
> Info : starting gdb server for esp32s3.cpu0 on 3333
> Info : Listening on port 3333 for gdb connections
> Info : [esp32s3.cpu0] Target halted, PC=0x40041A79, debug_reason=00000000
> Error: [esp32s3.cpu0] Timed out waiting for CPU to be reset, target state=2
> Info : [esp32s3.cpu0] Reset cause (1) - (Power on reset)
> Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
> Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
> Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
> Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
> Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000

[... 668 lines cut ...]
> Exception in thread Thread-1:
> Traceback (most recent call last):
>   File "threading.py", line 932, in _bootstrap_inner
> Exception in thread Thread-2:
> Traceback (most recent call last):
>   File "threading.py", line 932, in _bootstrap_inner
>   File "threading.py", line 870, in run
>   File "threading.py", line 870, in run
>   File "subprocess.py", line 1366, in _readerthread
>   File "subprocess.py", line 1366, in _readerthread
> OSError: [Errno 22] Invalid argument
> OSError: [Errno 22] Invalid argument
> [Debug Adapter]
> 2023-02-17 10:01:06,245 - Debug Adapter (main) - DEBUG - Closing the Socket
> 2023-02-17 10:01:06,245 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "list index out of range\n", "source": {}}, "seq": 28}
> 
> 2023-02-17 10:01:06,245 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
> 2023-02-17 10:01:06,245 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the Socket\n", "source": {}}, "seq": 29}
> 
> 2023-02-17 10:01:06,247 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
> 2023-02-17 10:01:06,247 - Debug Adapter (main) - DEBUG - Adapter is Stopped
> DEBUG_ADAPTER_STOPPED
> 2023-02-17 10:01:06,247 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
> 2023-02-17 10:01:06,248 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
> [Stopped] : ESP-IDF Debug Adapter

brianignacio5, 2023-02-20:
It seems that `Failed to wait for target state %d! Current state %d" %` I would suggest to update the launch.json to increase the timeout `"tmoScaleFactor": 2` for a bigger factor. Using the logLevel 4 would also show more timeout issues if you think the application is working fine reduce the logLevel to 2.

mbastida123, 2023-02-20:
> It seems that `Failed to wait for target state %d! Current state %d" %` I would suggest to update the launch.json to increase the timeout `"tmoScaleFactor": 2` for a bigger factor. Using the [...]

Hi @brianignacio5 and thanks for the reply.

incressing tmoScaleFactor to 4 didin't yield any result.

However using openOCD I see this output now:

> Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : only one transport option; autoselect 'jtag'
> Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
> Info : esp_usb_jtag: capabilities descriptor set to 0x2000
> Warn : Transport "jtag" was already selected
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
> Info : esp_usb_jtag: serial (68:B6:B3:48:87:04)
> Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
> Info : clock speed 40000 kHz
> Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
> Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
> Warn : target esp32s3.cpu1 examination failed
> Info : starting gdb server for esp32s3.cpu0 on 3333
> Info : Listening on port 3333 for gdb connections
> Info : [esp32s3.cpu0] Debug controller was reset.
> Error: libusb_bulk_write error: LIBUSB_ERROR_PIPE
> Info : [esp32s3.cpu0] Core was reset.
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: Failed to exec JTAG queue!
> Error: Failed to read debug stubs address location (-104)!
> Polling target esp32s3.cpu0 failed, trying to reexamine
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: [esp32s3.cpu0] Unexpected OCD_ID = 0b339fd2
> Examination failed, GDB will be halted. Polling again in 100ms
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: Failed to exec JTAG queue!
> Error: Failed to read debug stubs address location (-104)!
> Polling target esp32s3.cpu0 failed, trying to reexamine
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: [esp32s3.cpu0] Unexpected OCD_ID = 0b339fd2
> Examination failed, GDB will be halted. Polling again in 300ms
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: Failed to exec JTAG queue!
> Error: Failed to read debug stubs address location (-104)!
> Polling target esp32s3.cpu0 failed, trying to reexamine
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: [esp32s3.cpu0] Unexpected OCD_ID = 0b339fd2
> Examination failed, GDB will be halted. Polling again in 700ms
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
> Error: missing data from bitq interface
> Error: Failed to exec JTAG queue!
> Error: Failed to read debug stubs address location (-104)!
> Polling target esp32s3.cpu0 failed, trying to reexamine
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: [esp32s3.cpu0] Unexpected OCD_ID = 0b339fd2
> Examination failed, GDB will be halted. Polling again in 1500ms
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: Failed to exec JTAG queue!
> Error: Failed to read debug stubs address location (-104)!
> Polling target esp32s3.cpu0 failed, trying to reexamine
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO
> Error: missing data from bitq interface
> Error: [esp32s3.cpu0] Unexpected OCD_ID = 0b339fd2
> Examination failed, GDB will be halted. Polling again in 3100ms
> shutdown command invoked
> Error: libusb_bulk_write error: LIBUSB_ERROR_IO

Is this `LIBUSB_ERROR_IO` normal?

brianignacio5, 2023-02-20:
Something's off here in openOCD result. Maybe there is some information in the [openOCD troubleshooting FAQ](https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ). Could you please take a look @gerekon ?

mituritsyn, 2023-08-03:
> Is this `LIBUSB_ERROR_IO` normal?

you need to connect with boot shorted.

Faced same issue any solutions?

erhankur, 2025-03-11:
@mbastida123 I see different config files in the logs and issue description. If you still have an issue, could you please create another ticket in the [OpenOCD-ESP32](https://github.com/espressif/openocd-esp32) repo with the latest logs and detailed description.

closed 2025-03-11 by erhankur