espressif/esp-idf #10786: Crash in bta_sys_event during BTDeinit

ztefanjo, 2023-02-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

BTDeinit to work without crashes. btu_task_shut_down should ensure that the BTU_TASK is stopped before removing it.

### What is the actual behavior?

See attached coredump. The BTU_TASK is trying to dispatch an event while the BTC_TASK shuts down the BTU_TASK. 

[core-dump_51.1_Wifi-BLE-App_42.17_230214-150221.elf.txt](https://github.com/espressif/esp-idf/files/10765307/core-dump_51.1_Wifi-BLE-App_42.17_230214-150221.elf.txt)

### Steps to reproduce.

Not known, seen only once so far.

### More Information.

We execute all BT-related functions from the same task and should not have anything ongoing when doing BTDeinit.

zhp0406, 2024-02-28:
Is the issue still present in the latest version?

zhp0406, 2024-02-28:
thanks for your feedback,If the issue persists, please report it to us.

closed 2024-02-28 by Alvin1Zhang