espressif/esp-idf #10788: esptool.py flash fails with secure boot v2 on a ESP32-S3-WROOM-1

mhgue, 2023-02-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1-3-g886e98a2c1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Ability to flash applications with correct signature.

### What is the actual behavior?

1.) With --no-stub flash fails with: A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)
2.) Without --no-stub flash fails with: A fatal error occurred: Failed to start stub. There was no response.

### Steps to reproduce.

Enable secure boot V2 on a ESP32-S3 as described in [How To Enable Secure Boot V2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/secure-boot-v2.html#secure-boot-v2-howto)

### Build or installation Logs.

```plain
Uploading stub...
Running stub...

A fatal error occurred: Failed to start stub. There was no response.
Try increasing timeouts, for more information see: https://docs.espressif.com/projects/esptool/en/latest/esptool/configuration-file.html
CMake Error at run_serial_tool.cmake:55 (message):
  
  /home/mg/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/dec/mg/src/DX3_DSX_ESP/libs/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed

FAILED: CMakeFiles/flash 

=========================================================================================
or without stub:

Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)
CMake Error at run_serial_tool.cmake:55 (message):
  
  /home/mg/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/dec/mg/src/DX3_DSX_ESP/libs/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed

FAILED: CMakeFiles/flash
```

### More Information.

Using head of **release/v5.0** branch of IDF, **components/bootloader/Kconfig.projbuild** contains
```
config SECURE_BOOT
	...
        select ESPTOOLPY_NO_STUB if !IDF_TARGET_ESP32 && !IDF_TARGET_ESP32S2
```
and thus for IDF_TARGET_ESP32**S3** activating `SECURE_BOOT` will activate `ESPTOOLPY_NO_STUB=y` too and therefore the `esptool` is called with option `--no-stub` and fails with:
```
Erasing flash...
A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)
```
If changing the Kconfig entry to exclude S3 from `NO_STUB` too (like done for S2):
```
config SECURE_BOOT
	...
        select ESPTOOLPY_NO_STUB if !IDF_TARGET_ESP32 && !IDF_TARGET_ESP32S2 && !IDF_TARGET_ESP32S3
```
the option `--no-stub` is not used, but after enabling secure boot the `esptool` now fails with:
```
Uploading stub...
Running stub...

A fatal error occurred: Failed to start stub. There was no response.
```
In my application secure boot is only needed to verify OTA updates. Protection against direct access to the HW is not needed. Is there a way to allow flash stubs even with secure boot enabled? Or is there a way to enable flashing without stubs on a ESP32-S3-WROOM-1?

radimkarnis, 2023-02-23:
Hello @mhgue,
thanks for the report. This issue might not be related to the stub flasher - you can see flashing fails even in the ROM bootloader.

This could be an esptool regression, could you please help with verification?
You can try to install an older esptool version with `pip install esptool==3.3.2` in your ESP-IDF environment a try the operations again. Please let me know hoe that went.

I will investigate more, I have a similar report of this happening on the C3 with active secure boot.

mhgue, 2023-02-23:
Hello @radimkarnis 
thank you for answering. Currently I used the esptool v4.5 . Now with
```
Installing collected packages: esptool
  Attempting uninstall: esptool
    Found existing installation: esptool 4.5
    Uninstalling esptool-4.5:
      Successfully uninstalled esptool-4.5
  DEPRECATION: esptool is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for esptool ... done
Successfully installed esptool-3.3.2
```
I do get from idf.py
```
The following Python requirements are not satisfied:
esptool~=4.5 - (esptool 3.3.2 (/home/mg/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages), Requirement.parse('esptool~=4.5'))
```
Try with direct call to esptool
```
$ esptool.py --chip esp32s3 -p /dev/ttyUSB1 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size keep 0x20000 build/dx3-dsx-esp.bin
esptool.py v3.3.2
Serial port /dev/ttyUSB1
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:fd:6d:90
Uploading stub...
Running stub...
Traceback (most recent call last):
  File "/home/mg/.espressif/python_env/idf5.0_py3.10_env/bin/esptool.py", line 5482, in <module>
    _main()
  File "/home/mg/.espressif/python_env/idf5.0_py3.10_env/bin/esptool.py", line 5475, in _main
    main()
  File "/home/mg/.espressif/python_env/idf5.0_py3.10_env/bin/esptool.py", line 4805, in main
    esp = esp.run_stub()
  File "/home/mg/.espressif/python_env/idf5.0_py3.10_env/bin/esptool.py", line 946, in run_stub
    p = self.read()
  File "/home/mg/.espressif/python_env/idf5.0_py3.10_env/bin/esptool.py", line 438, in read
    return next(self._slip_reader)
StopIteration
```
and without STUB
```
$ esptool.py --chip esp32s3 -p /dev/ttyUSB1 -b 460800 --no-stub --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size keep 0x20000 build/dx3-dsx-esp.bin
esptool.py v3.3.2
Serial port /dev/ttyUSB1
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:fd:6d:90
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00020000 to 0x002f0fff...
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)
```
So unfortunately no significant difference.

mahavirj, 2023-02-24:
@mhgue 

> In my application secure boot is only needed to verify OTA updates. Protection against direct access to the HW is not needed.

Please check documentation section regarding secure OTA without secure boot [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/secure-boot-v2.html#signed-app-verification-without-hardware-secure-boot) and [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/ota.html#secure-ota-updates-without-secure-boot).

mhgue, 2023-02-24:
@mahavirj Thank you. I will have a look, this might be a workaround for me.
Regardless, I'd like to continue using the already irreversibly secure booting device, to assist @radimkarnis in clarifying this issue as described.

radimkarnis, 2023-03-01:
Hi @mhgue,
thanks again for trying to debug the issue. We are actively trying to reproduce it but without any luck. In our tests, everything works as intended.

The `01060000: Operation or feature not supported` error code from the ROM seems to be connected to the SPI Flash chip operations. That leads me to believe maybe this is not an ESP issue, but a problem with the flash chip.

We can try to narrow the problem down. For starters - could you please try the same process on a different host machine? Thank you.

mhgue, 2023-03-02:
Hi @radimkarnis,
thank you again for investigating this issue. I have no special or extra flash chip. It is a ready to use ESP32-S3-WROOM-1-n16r8 module with 16 MB flash right from the market. There is absolutely nothing special with it. Is there a way to identify the flash chip build inside the WROOM?
To try the process on a differnet host machine will take some time, because I usually only use Ubuntu Linux, no MacOS and no others capable of using the IDF. But maybe I can find a way with a virtual machine.
If it belongs to the flash chip, it would be interesting to try a module with a differnet chip. What module exactly have your used for your tries? In your tests are there both way working, with and without stub?
Thank you.

Ramakrishna247, 2023-03-05:
Hi,

I'am also facing similar issue with the ESP32 S3 N16R8 , my host machine is Windows 11, using VS code . tried IDF versions 4.4, 4.4.3 and  5.0
I have enabled the secure boot and Flash encryption , I'am able to flash the application in the first OTA partition , but if i'am trying to flash the application manually in to 2nd OTA partition i'am getting issue as  bellow

esptool.py v3.3.2
Serial port COM3
Connecting....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:d6:f2:70
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x0000b000 to 0x0000bfff...
Flash will be erased from 0x0001d000 to 0x0001efff...
Flash will be erased from 0x00220000 to 0x00310fff...
Erasing flash...
Took 0.06s to erase flash block
Wrote 3072 bytes at 0x0000b000 in 0.1 seconds (169.1 kbit/s)...
Erasing flash...
Took 0.09s to erase flash block
Wrote 8192 bytes at 0x0001d000 in 0.4 seconds (172.1 kbit/s)...
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)
---------------------------------------------------------------------------------------------------------------------------------
My partition table is as below.

Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs_key,data,nvs_keys,0xc000,4K,encrypted
nvs,data,nvs,0xd000,64K,
otadata,data,ota,0x1d000,8K,
phy_init,data,phy,0x1f000,4K,
ota_0,app,ota_0,0x20000,2M,
ota_1,app,ota_1,0x220000,2M,
file,data,spiffs,0x420000,12160K,
*******************************************************************************

Also I have configured PSRAM as octal OPI , 40 MHZ and serial flash as  QIO , 80 MHZ, i have tried different configurations of PSRAM  and FLASH but no luck. 
Please help me to resolve the issue.

Also i tried to update binary via OTA,   Iam not able to do OTA also as soon as binary gets downloaded and second stage bootloader verification of  downloaded app is successful a stackover flow is happening.

mhgue, 2023-03-06:
Hi @radimkarnis,
I think the best way to reproduce this for you, is to use my HW. I have got the permission to send you the board. On the backside there is the 6-pole connector for the ESPprog. If you give me a postal address, I will send you the board.

gm-jiang, 2023-03-09:
@mhgue Note that if secure boot v2 is enabled, then the ROM code will refuse to execute the stub even if the stub has been downloaded into RAM. Because we think it is not safe to execute stub. The correct logic should be bootROM(trusted) + efuse (trusted) ---> verify 2ndbootload ---> verify application.

gm-jiang, 2023-03-09:
For errors `A fatal error occurred: Failed to enter Flash download mode (result was 01060000: Operation or feature not supported)` . In fact, this is a normal download process (without `--no-stub` ),  I don't think it has anything to do with secureboot, the error code show that the esptool message is ok,but the runing result is wrong(Erase flash failed).

Ramakrishna247, 2023-03-09:
Hi,

@gm-jiang  is there any other way to flash the 2nd partition, also why esptool.py not able to erase the second partition ?

gm-jiang, 2023-03-10:
@Ramakrishna247 From what I can see so far, there are some reasons for flash erasing failure.
1. start_addr + len > flash chip size
2. start_addr not align to sector size
3. Unlock flash failed
4. Erase flash sector/block failed

If possible, could you connect a logic analyzer to see the signal on the SPI bus, this can determine whether the flash chip is hung up.

gm-jiang, 2023-03-13:
Related with https://github.com/espressif/esp-idf/issues/10959#issuecomment-1466037168

mhgue, 2023-03-17:
Hi @gm-jiang,
concerning [your first comment](https://github.com/espressif/esp-idf/issues/10788#issuecomment-1461839217): yes that is the reason why I tried both (with and without stub), because I saw that `components/bootloader/Kconfig.projbuild` disables stub usage for S3 but not for S2. So the decision to consider it unsafe to execute stubs obviously was made in between and I just wanted to try if it was just a wrong config entry. In general I perfectly agree with that, but it makes operation without stub indispensable.

concerning [your second comment](https://github.com/espressif/esp-idf/issues/10788#issuecomment-1461862588): Good point. I will try flash without stub on a device without secure boot.

Emill, 2023-03-20:
@gm-jiang, for ESP32-S3, when the `SECURE_BOOT_EN` bit has been enabled in efuse, is code execution from RAM totally disabled in all possible means when the chip is booted in uart download mode, or can e.g. a signed RSA signature be appended (in RAM) so that it allows to run the stub?

gm-jiang, 2023-03-21:
@gm-jiang Good question, but the bootROM does not support the feature of verifying the stub.

mahavirj, 2023-03-24:
@Emill 

> Regarding the "security feature" to not allow execution of code passed in download mode, it's a bit strange that arbitrary register reads/writes are allowed (which basically gives full access anyway) but not code execution

Yes, ideally the secure DL mode along with the secure boot should control the stub execution over UART. We have this addressed in our recent chips like C6/H2 but for S3 as you rightly mentioned that only secure boot configuration disallows the stub execution.

Just a note that we always recommend enabling secure DL mode along with any security features, please see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/security.html#debug-interfaces

radimkarnis, 2023-06-27:
This has been fixed in esptool v4.6. Running the install script again should install the latest available version.

If you are using the secure download mode, the flash size has to be explicitly set in menuconfig.

Closing this issue. Thanks everyone for the investigation.

radimkarnis, 2024-03-14:
Hello @discreetmayor,

The solution is mentioned above:

> If you are using the secure download mode, the flash size has to be explicitly set in menuconfig.

Please change `--flash_size keep` to a specific flash size. That should fix the issue.

MosamXu, 2024-05-18:
> 你好 ，
> 
> 解决方案如上所述：
> 
> > 如果您使用的是安全下载模式，则必须在 menuconfig 中显式设置闪存大小。
> 
> 请更改为特定的闪光灯尺寸。这应该可以解决问题。`--flash_size keep`

无法解决。

>esptool -p com4 -b 1152000 --no-stub --before default_reset --after no_reset write_flash --flash_mode dio --flash_freq 80m --flash_size keep 0x120000 enc/enc-signed-A23036_App.bin --encrypt
esptool.py v4.7.0
Serial port com4
Connecting...
Failed to get PID of a device on com4, using standard reset sequence.
.
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
Changing baud rate to 1152000
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00120000 to 0x001e7fff...
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01640000: Unknown result)