espressif/esp-idf #10789: E (453969) wifi:fail to alloc timer, type=1..

Rp777, 2023-02-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Device:          ESP WROOM 32
IDF Version:   V5.0.0
IDE:                VSCode (Extension: Espressif IDF v1.5.1)

I am running 2 tasks one for wifi and another one for mqtt.
After 5-9 minutes I get the below errors:

E (453969) wifi:fail to alloc timer, type=1..
E (459669) wifi:fail to alloc timer, type=5..
E (463289) wifi:fail to alloc timer, type=3..
.[0;31mE (471789) transport_base: tcp_write error, errno=No more processes.[0m..
.[0;31mE (471789) mqtt_client: Writing failed: errno=11.[0m..
.[0;31mE (471789) mqtt_client: Error sending ping.[0m..
.[0;31mE (471789) mqtt_client: Can't send ping, disconnected.[0m..
.[0;32mI (471799) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED.[0m..
I (483789) wifi:bcn_timout,ap_probe_send_start..
I (492289) wifi:ap_probe_send over, resett wifi status to disassoc..
I (492289) wifi:state: run -> init (c800)..
I (492289) wifi:pm stop, total sleep time: 0 us / 490663333 us..
..
W (492289) wifi:<ba-del>idx..
I (492289) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1..
.[0;32mI (492299) WIFI station: CB STA Disconnected.!!.[0m..
.[0;32mI (492309) WIFI station: CB connect to the STA fail .[0m..

Can any one point me out why wifi  is getting disconnected?
After esp_wifi_init() I am calling esp_wifi_set_ps(WIFI_PS_NONE);

Any document reference would also help.

Thanks in advance.!

Rp777, 2023-02-17:
Updated with Version 5.0.1 with the latest build relaese of esp-idf Getting below error:

I (461313) wifi:bcn_timeout,ap_probe_send_start..
I (463813) wifi:ap_probe_send over, resett wifi status to disassoc..
I (463813) wifi:state: run -> init (c800)..
I (463813) wifi:pm stop, total sleep time: 0 us / 462170833 us..
..
I (463823) wifi:<ba-del>idx..
I (463823) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1..
.[0;31mE (463833) transport_base: tcp_write error, errno=Software caused connection abort.[0m..
.[0;32mI (463833) WIFI station: CB STA Disconnected.!!.[0m..
.[0;32mI (463843) WIFI station: CB connect to the STA fail .[0m..
.[0;31mE (463843) mqtt_client: Writing failed: errno=113.[0m..
.[0;32mI (463853) MQTTS_EXAMPLE: MQTT_EVENT_ERROR.[0m..
.[0;31mE (463853) MQTTS_EXAMPLE: Last error captured as transport's socket errno: 0x71.[0m..
.[0;32mI (463863) MQTTS_EXAMPLE: Last errno string (Software caused connection abort).[0m..
.[0;31mE (463873) mqtt_client: Error sending ping.[0m..
.[0;31mE (463873) mqtt_client: Can't send ping, disconnected.[0m..
.[0;32mI (463883) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED.[0m..

Sometimes the error is random:
1) Sometimes I get below error:
                    E (453969) wifi:fail to alloc timer, type=1. .
                    E (459669) wifi:fail to alloc timer, type=5..
                    E (463289) wifi:fail to alloc timer, type=3
2) Sometimes I get this :
                    I (461313) wifi:bcn_timeout,ap_probe_send_start..
                    I (463813) wifi:ap_probe_send over, resett wifi status to disassoc..
                    I (463813) wifi:state: run -> init (c800)..
                    I (463813) wifi:pm stop, total sleep time: 0 us / 462170833 us..

Rp777, 2023-02-23:
Hi,
Can Anyone can help me out on this??
Any document I could refer?

Xiehanxin, 2023-02-28:
hi, @Rp777，it seems that there is insufficient ram. Could you please print the rainmain memory by esp_get_free_internal_heap_size

Rp777, 2023-03-02:
> hi, @Rp777，it seems that there is insufficient ram. Could you please print the rainmain memory by esp_get_free_internal_heap_size

Hi @Xiehanxin Sorry for delayed reply..

.[0;32mI (1307) spi_flash: detected chip: generic.[0m 
.[0;32mI (1310) spi_flash: flash io: dio.[0m 
.[0;32mI (1316) cpu_start: Starting scheduler on PRO CPU..[0m 
.[0;32mI (0) cpu_start: Starting scheduler on APP CPU..[0m 
.[0;32mI (1325) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations.[0m 
.[0;32mI (1335) MQTTS_EXAMPLE: [APP] Startup...[0m 
.[0;32mI (1335) MQTTS_EXAMPLE: [APP] Free memory: 4422984 bytes.[0m 
.[0;32mI (1345) MQTTS_EXAMPLE: [APP] IDF version: v5.0.1-dirty.[0m

I am allocating heap on PSRAM.
I Guess memory is not the issue

Rp777, 2023-04-25:
Hi @Xiehanxin  Sorry for delayed response.
I re-checked.

Seems like the the psram  memory is piling up.
It disconnects after few hours.
If I increase the task delay it runs for much more time.

Logs for reference:
14:45:13.414  E (5642001) wifi:sta is connecting, return error..
14:45:15.794  .[0;32mI (5644411) WIFI station: CB retry to connect to the STA.[0m..
14:45:15.820  I (5644411) wifi:flush txq..
14:45:15.820  I (5644421) wifi:stop sw txq..
14:45:15.820  I (5644421) wifi:lmac stop hw txq..
14:45:15.900  .[0;32mI (5644521) WIFI station: [APP] Free memory: 2676 bytes.[0m..
14:45:15.932  I (5644521) wifi:mode : sta (34:94:54:e6:58:24)..
14:45:15.932  I (5644521) wifi:enable tsf..
14:45:15.932  .[0;32mI (5644521) WIFI station: CB connect to the STA fail .[0m..
14:45:15.932  E (5644521) wifi:sta is connecting, return error..
14:45:18.339  .[0;32mI (5646931) WIFI station: CB retry to connect to the STA.[0m..
14:45:18.339  I (5646931) wifi:flush txq..
14:45:18.339  I (5646931) wifi:stop sw txq..
14:45:18.339  I (5646941) wifi:lmac stop hw txq..
14:45:18.419  .[0;32mI (5647041) WIFI station: [APP] Free memory: 2676 bytes.[0m..

mhdong, 2024-03-22:
Hi @Rp777 Sorry for delayed response.
We have recently fixed some memory leakage issues. Please use the latest release branch to test and see if the problem still exists.
https://github.com/espressif/esp-idf/tree/release/v5.0

closed 2024-05-13 by Sherry616