espressif/esp-idf #10790: Example Onewire DS18B20 does not work

d3ads1lence, 2023-02-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I have a custom board to which one DS18B20 sensor is connected and I assumed that I would take an [example ](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/onewire_ds18b20) how to work with the DS18B20 sensor load it into the device and see the temperature in the log. 

### What is the actual behavior?

But instead of this sensor is not detected on the bus.

### Steps to reproduce.

1. Take an example from the esp-idf repository.
2. Change the sensor pin.
3. Load the firmware into the device.

### Debug Logs.

```plain
I (253) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (253) onewire_rmt: RMT Tx channel created for 1-wire bus
I (263) gpio: GPIO[0]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (273) onewire_rmt: RMT Rx channel created for 1-wire bus
I (283) example: 1-wire bus installed
E (283) onewire_rmt: no device present on 1-wire bus
I (293) example: 0 device found on 1-wire bus
I (293) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (303) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (313) example: 1-wire bus deleted
```

### More Information.

Using a logic analyzer, I found that before ESP sends the sensor reset signal, the bus signal is set to zero at 32 ms.
[image]
I google this strange behavior and have found that is [well know issue](https://github.com/espressif/esp-idf/issues/9449).

:white_check_mark: I have workaround, that works for me: add `onewire_bus_write_bit(handle, 0);` in the end of _onewire_new_bus_rmt_ function.

I created this report because:
1) I don't understand why there is an example in the repository that doesn't work, I think it can be fixed.
2) To help someone else who will investigate this problem not to waste a lot of time on such nonsense.

wuyuanyi135, 2023-02-18:
For me, the RMT onewire driver fails on boot constantly. During device searching I wrap the search in a for loop and it exits until device count is satisfied otherwise retry.

danclarke, 2023-02-27:
For me it goes even further, I can never pick up a DS18B20. It works fine with an Arduino Uno. The workaround and loops didn't work for me.

[image]

```I (320) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (330) onewire_rmt: RMT Tx channel created for 1-wire bus
I (340) gpio: GPIO[5]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (340) onewire_rmt: RMT Rx channel created for 1-wire bus
I (350) example: 1-wire bus installed
E (360) onewire: no devices participating in search
I (360) example: 0 device found on 1-wire bus
E (1170) onewire: no devices participating in search
I (1170) example: 0 device found on 1-wire bus
E (1970) onewire: no devices participating in search
I (1970) example: 0 device found on 1-wire bus```

danclarke, 2023-02-28:
Turns out my probe needed some more voltage! The vanilla example still doesn't work, but the fix by @d3ads1lence worked for me.

DChe47, 2023-03-08:
This example works somehow strangely.
I use VSCode+PlatformIO. 
I wanted to check the new DS18B20 sensor and link and upload this example, but it didn't work, it was also "0 device found on 1-wire bus".
I sent to change the parameters in Munuconfig and suddenly the sensor began to work well.
I remember then I changed Log level - Verbose, RMT Config - Suppress legasy driver depricated varning, RMT Config - Enable debug log (Debug -0g was already) and set the correct CPU frecquensy (I have this 160MHz). Perhaps I changed something else, now I no longer remember.
Now the sensor is working and the temperature is displayed.
If I change anything in the configuration, then the sensor is not found again.
I think that we need to find problem in time delays.

Может кому-нибудь это поможет.

TimofeyPro, 2023-03-10:
I have the same issue on ESP32-DevKitC V4: ESP32-WROOM-32E using ESP-IDF v5.0.1: 0 device found on 1-wire bus.
Several attempts to launch the App were successful, but in most cases the program does not found my two DS18B20.

suda-morris, 2023-04-19:
@d3ads1lence Thank you for reporting this issue! indeed, I can reproduce the issue on my local side. And thanks for sharing your workaround! That also worked. 

Some follow-up work that we will take time to do in the soon future:

* Upload the onewire_bus component to the registry: https://components.espressif.com/
* set up some runner to test the DS18B20 example

closed 2023-05-08 by espressif-bot