espressif/esp-idf #10791: Problem with larger Strings mbedtls_aes_crypt_cbc concurency on ISR/Timer

filzek, 2023-02-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32 wrover psiram 

### Power Supply used.

External 5V

### What is the expected behavior?

mbedtls aes_crypt_cbc very large strings are breaking ISR concurrent calls.

Core  1 register dump:
PC      : 0x40093534  PS      : 0x00060633  A0      : 0x80098031  A1      : 0x3ffbfbb0
0x40093534: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:275

A2      : 0x00000000  A3      : 0x749f128c  A4      : 0xf0599579  A5      : 0x00000000
A6      : 0x00e04cc0  A7      : 0xe04cc000  A8      : 0xe04cc000  A9      : 0x3ffbfb10
A10     : 0x0000000a  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00001000
A14     : 0x3ffbfb03  A15     : 0x00000001  SAR     : 0x00000008  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4009352d  LEND    : 0x4009354f  LCOUNT  : 0x0f059956
0x4009352d: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:272

0x4009354f: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:290

Backtrace: 0x40093531:0x3ffbfbb0 0x4009802e:0x3ffbfbc0 0x40098b5a:0x3ffbfbe0 0x400842e6:0x3ffbfc10 0x400843e5:0x3ffbfc40 0x401507a5:0x3ffbfc60 0x4015084c:0x3ffbfc80 0x400d97c9:0x3ffbfcb0 0x40084091:0x3ffbfcd0 0x4000bfed:0x3ffee5f0 0x4009b6e9:0x3ffee600 0x40178a16:0x3ffee620 0x40178b3d:0x3ffee640 0x400dd0b4:0x3ffee680 0x400ebb0c:0x3fff7970
0x40093531: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:274

0x4009802e: prvCopyDataFromQueue at D:/Dropbox/esp-idf/components/freertos/queue.c:2326

0x40098b5a: xQueueReceiveFromISR at D:/Dropbox/esp-idf/components/freertos/queue.c:1968

0x400842e6: lock_acquire_generic at D:/Dropbox/esp-idf/components/newlib/locks.c:142

0x400843e5: _lock_acquire at D:/Dropbox/esp-idf/components/newlib/locks.c:163

0x401507a5: adc1_rtc_mode_acquire at D:/Dropbox/esp-idf/components/driver/adc_common.c:406

0x4015084c: adc1_get_raw at D:/Dropbox/esp-idf/components/driver/adc_common.c:431
 (inlined by) adc1_get_raw at D:/Dropbox/esp-idf/components/driver/adc_common.c:427

0x400d97c9: periodic_timer_adc_callback at D:/Dropbox/esp-idf/Dev/PRODUCAO/main/user_main.c:2180
 (inlined by) periodic_timer_adc_callback at D:/Dropbox/esp-idf/Dev/PRODUCAO/main/user_main.c:2170

0x40084091: _xt_lowint1 at D:/Dropbox/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1114

0x4009b6e9: vPortClearInterruptMaskFromISR at D:/Dropbox/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at D:/Dropbox/esp-idf/components/freertos/port/xtensa/port.c:332

0x40178a16: esp_aes_release_hardware at D:/Dropbox/esp-idf/components/mbedtls/port/aes/block/esp_aes.c:68

0x40178b3d: esp_aes_crypt_cbc at D:/Dropbox/esp-idf/components/mbedtls/port/aes/block/esp_aes.c:259

TOTAL SIZE TO BE ENCRYPTED STRLEN from a STRING [37475]
invove from: mbedtls_aes_crypt_cbc

### What is the actual behavior?

execute everything right, not crash because the system is on load, even on interruption ISR running, so, the AES should be slowed and others functions keeped.

with smallers strings it is working perfect.

### Steps to reproduce.

4 interruptions running.
ADC reading while timer interrupt
gpio interrupt on every 8000us

### Debug Logs.

```plain
no logs.
```

### More Information.

The problems is on the routing of MBED with all others functions runnign together.

AdityaHPatwardhan, 2023-02-27:
Hi @filzek Can you please provide a small sample code that may help in reproducing the issue?
Thanks,
Aditya

filzek, 2023-02-27:
run adc on timer handle reading by 100 times on each of 6 adc channel. every 1/32 sec, to timer will be dispatch 32 times per second. read 100 times each of 6 channels. 

in another task also run udp multicast sender and receive.
in another task gpio interrupt handle.

in a http task try to encrypt a psiram object of 1mb of pure text.
crash happen.

we tried everything to solve but was unable.

it giver

AdityaHPatwardhan, 2023-04-04:
Hi @filzek Can you please provide a small test code.
I am unable to reproduce this at my end.
The issue you see looks like that it is only reproducible under certain conditions.
If you could provide some small test build then it would definitely speed up the process of providing a fix.
Thanks,
Aditya

AdityaHPatwardhan, 2023-04-24:
Hi @filzek I am unable to reproduce the issue at my end.
Please provide a small test code if possible.
I will close down the issue. Feel free to re-open the issue if the problem is not solved at your end.
Thanks,
Aditya

closed 2023-04-24 by AdityaHPatwardhan