espressif/esp-idf #10798: Linking CXX executable IDF 4.4.4

Jurand1, 2023-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I am using ESP IDF 4.4.4 and the same problem I had with IDF v5.0. I am using idf and mdf components along. Since two weeks I have been trying to solve linking executable problem but without any succses. Please read below-

1/6] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\Janek\Projekty\VSC_proj\Espnow_Slave_1.1\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/Janek/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/Janek/Projekty/VSC_proj/Espnow_Slave_1.1/build/bootloader/bootloader.bin"
Bootloader binary size 0x63b0 bytes. 0xc50 bytes (11%) free.
[2/4] Linking CXX executable Espnow_Slave_1.1.elf
FAILED: Espnow_Slave_1.1.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\Espnow_Slave_1.1.elf.rsp -o Espnow_Slave_1.1.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x1c): undefined reference to `app_main'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
C:/Users/Janek/esp/esp-idf/components/freertos/port/port_common.c:135: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Thanks for replynig and any sugetions to resolve problem.

igrr, 2023-02-19:
Please check if your app_main function is defined with C linkage: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/cplusplus.html#defining-app-main-in-c

Jurand1, 2023-02-19:
Thanks for replying but I did it before and repeat today. It is not working for me. Error is the same as I wrote before.

igrr, 2023-02-19:
To debug the such an error, i would suggest to do the following steps:

1. Check whether the source file where the app_main function is defined is actually compiled. You can do so, for example, by introducing some syntax error in the file and checking that this error is reported during the compilation process.
2. Find the object file produced from that source file and get the list of symbols in that file using xtensa-esp32-elf-nm. Do you see app_main in the list of symbols?

Jurand1, 2023-02-20:
Hello, I did first point I commented wi-fi componet to be missed and building error was not reported. Compiler finnished its job with the same error as I reported it yesterday.

igrr, 2023-02-20:
I see, this probably means that the source file where app_main is defined is not compiled. Since we can't see your project, please share some further information:
* In which file and in which component is app_main defined?
* What is the contents of the CMakeLists.txt of that component?

Jurand1, 2023-02-20:
Hi. App_main is defined in get_started.c in component called main.
This is Cmake file from "main" folder:

_idf_component_register(SRCS "get_started.c"
                    INCLUDE_DIRS "include"
                    REQUIRES
                    espnow_chain device_inout
                    espnow_wireless common_var HDC1080 espnow_utility
                    esp_timer mcommon mconfig mwifi
)_
This is Cmake folder from Project folder:
_cmake_minimum_required(VERSION 3.16)
set(COMPONENT_DIRS "$ENV{IDF_PATH}/components" )
set(EXTRA_COMPONENT_DIRS "${CMAKE_CURRENT_LIST_DIR}/components" "${CMAKE_CURRENT_LIST_DIR}/components_mdf")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(Espnow_Slave_1.1)_

igrr, 2023-02-20:
```
set(COMPONENT_DIRS "$ENV{IDF_PATH}/components" )
set(EXTRA_COMPONENT_DIRS "${CMAKE_CURRENT_LIST_DIR}/components" "${CMAKE_CURRENT_LIST_DIR}/components_mdf")
```

I think this sets the components to just the `$IDF_PATH/components`, as well as `components` and `components_mdf` subdirectories of the project. So the `main` component is not built.

I think setting COMPONENT_DIRS here is unnecessary, since the default behavior of the build system is to include the components found in: IDF /components subdirectory, project /components subdirectory, and project /main subdirectory.

You can simply it as:

```cmake
cmake_minimum_required(VERSION 3.16)
set(EXTRA_COMPONENT_DIRS "${CMAKE_CURRENT_LIST_DIR}/components_mdf")
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(Espnow_Slave_1.1)
```

Afterwards, check CMake output where the list of components being built is printed. (Look for `-- Components: `)
If everything is correct, `main` should be on the list.

Jurand1, 2023-02-21:
Hi, I did as you suggested and I removed that line 

_set(COMPONENT_DIRS "$ENV{IDF_PATH}/components" )_

I got error:

_[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
.....Updating lock file at C:/Users/Janek/Projekty/VSC_proj/Espnow_Slave_1.1\dependencies.lock
Processing 1 dependencies:
[1/1] idf (4.4.4)
-- DEBUG: Use esp-modbus component folder: C:/Users/Janek/esp/esp-idf/components/freemodbus.
CMake Error at C:/Users/Janek/esp/esp-idf/tools/cmake/component.cmake:237 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "coap" candidates: "espressif__coap, coap"

Call Stack (most recent call first):
  C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:486 (__component_get_requirements)
  C:/Users/Janek/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:21 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Janek/Projekty/VSC_proj/Espnow_Slave_1.1/build/CMakeFiles/CMakeOutput.log".
←[31mFAILED: ←[0mbuild.ninja
C:\Users\Janek\.espressif\tools\cmake\3.20.3\bin\cmake.exe --regenerate-during-build -SC:\Users\Janek\Projekty\VSC_proj\Espnow_Slave_1.1 -BC:\Users\Janek\Projekty\VSC_proj\Espnow_Slave_1.1\build 
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1_

Thanks for replying

igrr, 2023-02-21:
> requirement: "coap" candidates: "espressif__coap, coap"

This means that one of the components in your project has a dependency (declared in idf_component.yml file) on espressif/coap component from the IDF Component Registry (https://components.espressif.com/components/espressif/coap). At the same time, IDF v4.4.x has `coap` component built-in, which leads to this conflict. Please check the components in your project (their idf_component.yml files) and see if any of them has a dependency on `coap`.

Jurand1, 2023-02-21:
I had coap file in my project and one was removed. Undefined refrence error "app_main" still is.

igrr, 2023-02-23:
Okay, in that case going back to the previous list of troubleshooting steps:

1. Can you check if the `main` component is built? (Check if it is present in the `-- Components: ` list printed by CMake when you run `idf.py reconfigure`, or introduce some syntax error in main/CMakeLists.txt and see if an error is reported.)
2. Is the source file where app_main is defined built? (For example, introduce an error in that source file and check if the error is reported.)
3. If the source file is built, is the `app_main` defined in the object file, and does it have C linkage there? (Output of `xtensa-esp32-elf-nm build/esp-idf/main/CMakeFiles/__idf_main.dir/YOUR_MAIN_SOURCE_FILE.c.obj` should include `<number> T app_main`)

Jurand1, 2023-02-23:
Thanks for replying. I lost hope to get any solution to solve my problem I deleted all files associated with Esp idf. I installed new version Espidf 5.0.1. and decided to rewrite some components and app for new version. I started new mdf project and problem is 
- I did not add any components to check if app get_started.c will build-no building with include($ENV{MDF_PATH}/project.cmake). Error can not resolve component "driver/i2c"
 
CMake Error at C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'driver/i2c'.
Call Stack (most recent call first):
  C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
  C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:574 (__build_expand_requirements)
  C:/Users/Janek/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:10 (project)
- I added line-list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}\\components\\driver\\i2c) with or without quotation and built with include($ENV{MDF_PATH}/project.cmake). I got error

Diagnostic info: C:/Users/Janek/esp/esp-idf/tools/split_paths_by_spaces.py was invoked in C:\Users\Janek\Projekty\VSC_proj\espidf_v5.0.1\Test_v5_Espnow with arguments: ['--var-name=EXTRA_COMPONENCMake Error at C:/Users/Janek/esp/esp-idf/tools/cmake/project.cmake:100 (message):
  Failed to parse EXTRA_COMPONENT_DIRS, see diagnostics above
Call Stack (most recent call first):
  C:/Users/Janek/esp/esp-idf/tools/cmake/project.cmake:239 (paths_with_spaces_to_list)
  C:/Users/Janek/esp/esp-idf/tools/cmake/project.cmake:327 (__project_init)
  CMakeLists.txt:10 (project).

- I tried to build example project with include($ENV{IDF_PATH}/tools/cmake/project.cmake)
 I got error-

CMake Error at C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'mcommon'.
Call Stack (most recent call first):
  C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
  C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:574 (__build_expand_requirements)
  C:/Users/Janek/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:10 (project)

- I wrote line at main Cmakelist-set(EXTRA_COMPONENT_DIRS $ENV{MDF_PATH}/components/mcommon) and error-

CMake Error at C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'driver/i2c'.
Call Stack (most recent call first):
  C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
  C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  C:/Users/Janek/esp/esp-idf/tools/cmake/build.cmake:574 (__build_expand_requirements)
  C:/Users/Janek/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:10 (project).

- I changed line at main Cmakelist-set(EXTRA_COMPONENT_DIRS $ENV{MDF_PATH}/components/mcommon $ENV{IDF_PATH}\\components\\driver\\i2c) and error again-

Running cmake in directory c:\users\janek\projekty\vsc_proj\espidf_v5.0.1\test_v5_espnow\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 c:\users\janek\projekty\vsc_proj\espidf_v5.0.1\test_v5_espnow"...
-- Found Git: C:/Espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
CMake Error at C:/Users/Janek/esp/esp-idf/tools/cmake/project.cmake:243 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
  C:/Users/Janek/esp/esp-idf/components/driver/i2c
Call Stack (most recent call first):
  C:/Users/Janek/esp/esp-idf/tools/cmake/project.cmake:327 (__project_init)
  CMakeLists.txt:10 (project).
Thanks for reading and wait for any reply.

igrr, 2023-02-24:
I think the issue might be that ESP-MDF is compatible with IDF v4.3.1, not with v5.0.1: https://github.com/espressif/esp-mdf#quick-start

The issue about ESP-MDF incompatibility with IDF v5.x was reported in the ESP-MDF repository here: https://github.com/espressif/esp-mdf/issues/325.

Jurand1, 2023-02-24:
Hi, thanks for your help. In this case I need to downgrade ESP IDF to v4.1. I hope ESP-MDF compability with v5.0 will be improved as soon as possible.

0xjakob, 2023-04-17:
@Jurand1 Please not that IDF v4.1 is not maintained anymore. IDF version v4.3 would be the suggested one. Unless there is something else that we can do from the IDF side, could we close this issue then?

AxelLin, 2023-04-17:
> @Jurand1 Please not that IDF v4.1 is not maintained anymore. IDF version v4.3 would be the suggested one. 

@0xjakob 
If you still maintain v4.3 and v4.4, you probably want to fix this: https://github.com/espressif/esp-idf/pull/7754

closed 2023-04-18 by Jurand1