espressif/esp-idf #10799: add support for single pair Ethernet PHY

gkasprow, 2023-02-19:
### Is your feature request related to a problem?

10Base-T1S becomes more and more popular as a CAN replacement in industry and automotive.
It's an ideal solution for short-range IoT networks, epecially where 2.4GHz band is overcorowded

### Describe the solution you'd like.

Support for either LAN8670 PHY or NCN26010 MAC+PHY

### Describe alternatives you've considered.

Of course standard Ethernet works fine but it requires switch and a lot of cabling if one needs to build a sensor network with let's say 16 devices.

### Additional context.

Here is the use case - a photonic sensor where SPE is used for both data transfer and power
https://github.com/sinara-hw/DiPho/issues/18

kostaond, 2023-02-24:
Hi @gkasprow, thanks for submitting the feature request. It really could be interesting to support 10Base-T1S. Could you please check this [article](https://blog.espressif.com/esp-idf-custom-ethernet-phy-driver-quick-guide-45f9d23de0ad) first?  Would it be applicable to 10Base-T1S PHY chips?

gkasprow, 2023-02-24:
It looks so. These chips have a standard RMII interface. Only registers differ.

gkasprow, 2023-02-24:
I have [LAN8670 eval](https://eu.mouser.com/ProductDetail/Microchip-Technology/EV06P90A?qs=tlsG%2FOw5FFhZQakMhwEgnQ%3D%3D) in my lab as well as [USB dongle](https://eu.mouser.com/new/microchip/microchip-evb-lan8670-usb-board/)
we are curently developing open source PoE router based on RPI 0.

morgana2313, 2023-07-11:
@gkasprow: we also want to use the LAN8670 with esp-idf. Is there any code available from https://github.com/sinara-hw/DiPho/issues/18 available, from which we can learn how to setup the proper LAN8670 chip registers ?

We have the EVB-LAN8670-USB working with Linux raspberry pi.

gkasprow, 2023-07-11:
We were using NCN26010 with STM32 eval board.
There is example code for STM32 and other platforms provided by Onsemi
https://www.onsemi.com/pub/Collateral/EXAMPLE%20SOFTWARE:%20STM32%20IMPLEMENTATION%20.ZIP

morgana2313, 2023-07-15:
We got an ESP32 + LAN8551 replying to ping packets send by the RPi with the *LAN8650/1 OPEN Alliance TC6 Driver*. It gives ping times of ±50ms. Between two RPi's the SPE ping times are ±1ms.

What would be normal ping times for a 10Base-T1S network?

gkasprow, 2023-07-15:
It depends on how the ICMP stack is implemented. Did you use polling or interrupts?
With interrupts, it should be a much lower value. If you use polling, 50ms could be possible.

morgana2313, 2023-07-15:
Dank voor uw bericht.

Ik ben afwezig tot 7 augustus 2023 en lees sporadisch mijn mail.

Voor dringende zaken kunt u contact opnemen met mijn collega's via 070 891 9 891 of via ***@***.***

baybal, 2023-07-21:
> Hi @gkasprow, thanks for submitting the feature request. It really could be interesting to support 10Base-T1S. Could you please check this [article](https://blog.espressif.com/esp-idf-custom-ethernet-phy-driver-quick-guide-45f9d23de0ad) first? Would it be applicable to 10Base-T1S PHY chips?

Current chip revisions of LAN867* have buggy PLCA. I was recommended NCN26000, but it's nowhere to be found.

gkasprow, 2023-07-21:
NCN26010 chips are easily available [now](https://www.digikey.com/en/products/detail/NCN26010XMNTXG/488-NCN26010XMNTXGCT-ND/17038954?curr=usd&utm_campaign=buynow&utm_medium=aggregator&utm_source=octopart). Availability issue was a few months ago.
Devkits are also [available ](https://octopart.com/ncn26010xmnevb-onsemi-129969137?r=sp).

baybal, 2023-07-21:
> NCN26010 chips are easily available [...]

NCN26010 is SPI and NCN26000 is MII. You can only realistically use it with a much higher end MCU, while RMII is a lingua franca for ethernet in microcontrollers, including sub $1 ones.

gkasprow, 2023-07-21:
Oh,I didn't know Onsemi has RMII version. I thought you made a typoo. Anyway, SPI is better because RMII consumes most of ESP32 IOs.

baybal, 2023-07-23:
> Oh,I didn't know Onsemi has RMII version. I thought you made a typoo. Anyway, SPI is better because RMII consumes most of ESP32 IOs.

But then you lose all the power & performance benefit of the onboard MAC

gkasprow, 2023-07-24:
We arr limited to 10Mbit anyway so that is not critical.

kostaond, 2023-12-06:
https://github.com/espressif/esp-eth-drivers/pull/19

Indastri, 2024-03-28:
The fix was merged. Closing the issue. If there will be any more problems do not hesiate to reopen this or create a new issue.
Cheers!

closed 2024-03-28 by Indastri