espressif/esp-idf #10800: OTA crashes when FreeRTOS SMP is enabled

sosiskus, 2023-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-32-4MB

### Power Supply used.

USB

### What is the expected behavior?

I want to make OTA while using FreeRTOS SMP, and I would like the update to be successful. I tried to use esp_https_ota() in task that was created by xTaskCreatePinnedToCore()

### What is the actual behavior?

When I run this programm and try to update random errors appear when OTA perform is starting for example
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x400800c0  PS      : 0x00060d34  A0      : 0x00000000  A1      : 0x3ffbf16c
0x400800c0: _WindowUnderflow8 at C:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel-SMP/portable/xtensa/xtensa_vectors.S:1915

A2      : 0x3ffb2154  A3      : 0x0007a120  A4      : 0x8008a628  A5      : 0x3ffb1a30
A6      : 0x00000001  A7      : 0x3ffc69e4  A8      : 0x8015f985  A9      : 0x3ffc1130
A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x000001b0  SAR     : 0x00000020  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
Guru Meditation Error: Core  0 panic'ed (Double exception).

Core  0 register dump:
PC      : 0x40088d6a  PS      : 0x00040836  A0      : 0x400ea138  A1      : 0x3ffbeeac
0x40088d6a: _xt_context_save at C:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel-SMP/portable/xtensa/xtensa_context.S:203

0x400ea138: wifi_softap_get_config at ??:?

A2      : 0x3ffb0040  A3      : 0x3ffbeee4  A4      : 0x00000001  A5      : 0x3ffbeee0
A6      : 0x00060d34  A7      : 0x00040024  A8      : 0x400803c0  A9      : 0x3ffbf08c
0x400803c0: _DoubleExceptionVector at C:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel-SMP/portable/xtensa/xtensa_vectors.S:471

A10     : 0x00060a36  A11     : 0x00040026  A12     : 0x3ffc1150  A13     : 0x4008158c
0x4008158c: xt_highint4 at C:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/soc/esp32/highint_hdl.S:234

A14     : 0x00000000  A15     : 0xffffffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000002
EXCVADDR: 0xffffffe0  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x40088d67:0x3ffbeeac |<-CORRUPTED
0x40088d67: _xt_context_save at C:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel-SMP/portable/xtensa/xtensa_context.S:202

or
[... 315 lines cut ...]
mode:DIO, clock div:1
load:0x3fff0030,len:7000
load:0x40078000,len:15456
load:0x40080400,len:3840
0x40080400: _init at ??:?

entry 0x4008064c
I (160) boot: ESP-IDF v5.0-dirty 2nd stage bootloader
I (161) boot: compile time 23:16:08
I (161) boot: chip revision: v1.0
I (165) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (172) boot.esp32: SPI Speed      : 80MHz
I (177) boot.esp32: SPI Mode       : DIO
I (181) boot.esp32: SPI Flash Size : 4MB
W (186) boot.esp32: PRO CPU has been reset by WDT.
W (191) boot.esp32: WDT reset info: PRO CPU PC=0x40088da0
0x40088da0: _xt_context_save at C:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel-SMP/portable/xtensa/xtensa_context.S:217

W (197) boot.esp32: WDT reset info: APP CPU PC=0x400818a7
0x400818a7: panicHandler at C:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/panic_handler.c:217
```

### More Information.

I tried disabling Freertos SMP and everything works just fine also enabling "use freertos only in one core" fixes the issue. But I need both cores for my project, so how can I use SMP and OTA together so they do not cause errors

Dazza0, 2023-02-20:
@sosiskus You should be able to disable `CONFIG_FREERTOS_SMP` and still get SMP.

The `CONFIG_FREERTOS_SMP` option's naming is actually a bit misleading. It actually enables the [SMP implementation of the FreeRTOS Kernel](https://www.freertos.org/symmetric-multiprocessing-introduction.html) that is written/maintained by Amazon. There's a port of this kernel in ESP-IDF (enabled via the `CONFIG_FREERTOS_SMP`), but this port is still in an experimental/beta state.

If you disable `CONFIG_FREERTOS_SMP`, you will be running IDF FreeRTOS instead (a modified version the FreeRTOS kernel for the ESP chips). This also supports SMP and is currently the default version of the FreeRTOS kernel used in ESP-IDF.

We already have pending commit updating the description of the `CONFIG_FREERTOS_SMP` to avoid this confusion.

Dazza0, 2023-02-27:
@sosiskus

Given that the CPU is crashing at `_WindowUnderflow8` and `_xt_context_save`, it seems like some form of stack corruption is occurring. For now, I suggest you disable `CONFIG_FREERTOS_SMP` (you'll still get SMP functionality, but via the IDF FreeRTOS implementation instead) as the `CONFIG_FREERTOS_SMP` is still an experimental feature.

I've updated the description of the `CONFIG_FREERTOS_SMP` option ([commit](https://github.com/espressif/esp-idf/commit/c126b910b685c24316c2cbb62cdc34c408192196)), and the FreeRTOS documentation ([commit](https://github.com/espressif/esp-idf/commit/c94e0a9892e9aa2005f38e97c5b921ad30c9b3a2)) to describe the different implementations of FreeRTOS.

I'll update this issue again when the option/documentation updates have been backported to v5.0.

closed 2023-02-27 by Dazza0