espressif/esp-idf #10801: ESP32-S3 WIFI configure not storage on NVS Flash

pnt325, 2023-02-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board: ESP32-S3 WROOM-2 V1.2

### Power Supply used.

USB

### What is the expected behavior?

Get wifi configure by funciton `esp_wifi_get_config` to check that wifi is provisioned or not.

### What is the actual behavior?

`wifi_config_t` ssid alway null. After multiple reset MCU and WIFI connected to AP.

```c
    wifi_config_t __cfg;
    ESP_ERROR_CHECK(esp_wifi_get_config(WIFI_IF_STA, &__cfg));
    ESP_LOGW(TAG, "Is provisioning: %s", strlen((const char*)__cfg.sta.ssid)?"YES":"NO");
```

[image]

### Steps to reproduce.

Run the example wifi station then add wifi AP to run at the same time.

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3810,len:0x167c
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x2e98
entry 0x403c9904
I (32) boot: ESP-IDF v5.0 2nd stage bootloader
I (32) boot: compile time 15:44:48
I (32) boot: chip revision: v0.1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : SLOW READ
[... 178 lines cut ...]
I (690) wifi:Init max length of beacon: 752/752
I (690) wifi station: wifi_init_sta finished.
I (1000) wifi:ap channel adjust o:1,1 n:3,1
I (1000) wifi:new:<3,0>, old:<1,1>, ap:<3,1>, sta:<3,0>, prof:1
I (1970) wifi:state: init -> auth (b0)
I (1980) wifi:state: auth -> assoc (0)
I (1980) wifi:state: assoc -> run (10)
I (2000) wifi:connected with xxxxxxxxxxxx, aid = 16, channel 3, BW20, bssid = xxxxxxxxxxxxx
I (2000) wifi:security: WPA2-PSK, phy: bgn, rssi: -33
I (2010) wifi:pm start, type: 1

I (2010) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
W (2020) wifi:<ba-add>idx:0 (ifx:0, xxxxxxxxxxxxx), tid:5, ssn:2, winSize:64
I (2080) wifi:BcnInt:102400, DTIM:1
I (2960) esp_netif_handlers: sta ip: 192.168.0.37, mask: 255.255.255.0, gw: 192.168.0.1
I (2960) wifi station: got ip:192.168.0.37
I (2960) wifi station: connected to ap SSID:xxxxxxxxxxxx password:xxxxxxx
W (3200) wifi:<ba-add>idx:1 (ifx:0, xxxxxxxxxxxxx), tid:0, ssn:0, winSize:64
```

### More Information.

The example run success on ESP32-C3 and ESP32-S3. It's also get the wifi configure success on ESP32-C3. 

Then I just think there is problem with ESP32-S3

igrr, 2023-02-20:
> Octal Flash Mode Enabled
> ...
> I (403) spi_flash: flash io: dio

Please check if you have configured the flash mode correctly for your module, enabling CONFIG_ESPTOOLPY_OCT_FLASH option in menuconfig. See https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#how-to-configure-flash-and-psram for more details.

pnt325, 2023-02-21:
It's work for me.
Thank you very much @igrr 

[image]

closed 2023-02-21 by pnt325