espressif/esp-idf #10802: esp32_mock.h

yueduz, 2023-02-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

"#ifdef __linux__" should not be used to load "esp32_mock.h" in "components/lwip/port/esp32/include/lwipopts.h". I learned that mocks are used for simulation testing, but I'm developing normally on Linux and don't want to do testing.

This gives me a red squiggle under vscode in Linux.

[image]

0xjakob, 2023-02-22:
Hi @yueduz

Mocks can be used whenever they are useful. This can e.g. also be the case if you develop on Linux and need to satisfy a dependency that does not exist on Linux.

Do you also face a build error or is it just the red squiggle that you are concerned about?

KaeLL, 2023-02-22:
microsoft/vscode-cpptools#9680

david-cermak, 2023-03-02:
@yueduz I agree that it's wrong to conditionally compile based on this `__linux__` macro. This issue has already been fixed in fa97004faf03240a1446321a0547eb0ee697e0d5 with the official support of linux target for lwip.

> This gives me a red squiggle under vscode in Linux.

Would it help if you include:
```cpp
#include <sys/socket.h>
```
instead of `lwip/sockets.h" ?

yueduz, 2023-03-02:
Then I think it's okay to shut him down

closed 2023-03-02 by yueduz

xdevelnet, 2023-04-10:
I am not sure that this issue should be closed. I'm using platformio and vscode, and have the save error that is related to esp32_mock.h

And I'm not using lwip/sockets.h, but using sys/socket.h

[image]

@0xjakob @david-cermak 

Can you give me a proper direction about solving this? I mean, it's not disturbing me that much because it just compiles fine, but I'm trying to understand why editor highlights it.

P.S. After few weeks of using platformio I'm thinking that it was a bad idea. It's not actually helping much, but causes difficulties all the time.

0xjakob, 2023-04-10:
> I'm trying to understand why editor highlights it

@xdevelnet The issue seems to be that the IDEs cannot correctly chose the compiler for the syntax checking, see the  the issue linked in the [comment above](https://github.com/espressif/esp-idf/issues/10802#issuecomment-1440956630). I'm not using any of these IDEs, though, so the variance of my comment might be high.

yueduz, 2023-04-10:
> > 我试图理解为什么编辑器突出显示它
> 
> @xdevelnet[问题似乎是 IDE 无法正确选择编译器进行语法检查，请参阅上面评论](https://github.com/espressif/esp-idf/issues/10802#issuecomment-1440956630)中链接的问题。不过，我没有使用这些 IDE 中的任何一个，所以我的评论可能会有很大差异。

Are you using the latest version of the library?

xdevelnet, 2023-04-10:
> @xdevelnet The issue seems to be that the IDEs cannot correctly chose the compiler for the syntax checking, see the the issue linked in the [comment [...]

Thanks for the reference! At least I can understand why it's happening.

xdevelnet, 2023-04-10:
> > > 我试图理解为什么编辑器突出显示它
> > 
> > 
> > @xdevelnet[问题似乎是 IDE 无法正确选择编译器进行语法检查，请参阅上面评论](https://github.com/espressif/esp-idf/issues/10802#issuecomment-1440956630)中链接的问题。不过，我没有使用这些 IDE 中的任何一个，所以我的评论可能会有很大差异。
> 
> Are you using the latest version of the library?

Yes. Here is the line from my platromio.ini:

```
platform = https://github.com/platformio/platform-espressif32.git
```

They are using 6.1.0 version if esp-idf.

yueduz, 2023-04-10:
> Yes. Here is the line from my platromio.ini:
>
> ```
> platform = https://github.com/platformio/platform-espressif32.git
> ```
>
> They are using 6.1.0 version if esp-idf.

platform使用的idf版本并不一定是最新版本。请确保fa97004提交的内容包括在内。

yueduz, 2023-04-10:
> Yes. Here is the line from my platromio.ini:
>
> ```
> platform = https://github.com/platformio/platform-espressif32.git
> ```
>
> They are using 6.1.0 version if esp-idf.

The idf version used by the platform is not necessarily the latest version.  Please make sure the fa97004 submission is included.

yueduz, 2023-04-10:
Another way to eliminate the red line is to set the target platform in the c/c++ extension in vscode to windows or macos (the cpp extension does not yet support setting the platform to arm).  If you don't manually set the target platform, the extension will automatically set it to the native platform type for you.