espressif/esp-idf #10804: adc_digi_read_bytes() messes the order of channels between read chuncks

ves011, 2023-02-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

adc_digi_read_bytes() for multiple channels and given number of samples shall return the channel values in the same order  in each data chunk returned by function call.
e.g. for 2 channel ADC1 only, i expect 
chunk1 = **data_chn_a**, **data_chn_b**,  **data_chn_a**, **data_chn_b**,  **data_chn_a**, **data_chn_b**, ...
if chunk1 ends with **data_chn_a**, chunk2 shall start with **data_chn_b**,  data_chn_a, data_chn_b,  data_chn_a, data_chn_b, ...
if chunk1 ends with **data_chn_b**, chunk2 shall start with **data_chn_a**, data_chn_b,  data_chn_a, data_chn_b,  data_chn_a, ...
In this way the assembled buffer will not miss any sample.

### What is the actual behavior?

Instead the following chunk start randomly with data_chn_a or data_chn_a messing the order in the assembled buffer.
Maybe some samples are lost, which for my case is very critical

### Steps to reproduce.

this is a sample code to reporoduce
```
typedef union
    {
    uint8_t result[NR_SAMPLES * ADC_RESULT_BYTE];
    adc_digi_output_data_t presult[NR_SAMPLES];
    } res_t;

#define    SENSOR_ADC_CHANNEL        (4)
#define CURRENT_ADC_CHANNEL        (3)
static uint16_t adc1_chan_mask = BIT(SENSOR_ADC_CHANNEL) | BIT(CURRENT_ADC_CHANNEL);
static uint16_t adc2_chan_mask = BIT(0);
static adc_channel_t channel[] = {CURRENT_ADC_CHANNEL, SENSOR_ADC_CHANNEL};
static void continuous_adc_init(uint16_t adc1_chan_mask, uint16_t adc2_chan_mask, adc_channel_t *channel, uint8_t channel_num)
    {
    adc_digi_init_config_t adc_dma_config =
        {
        .max_store_buf_size = NR_SAMPLES * ADC_RESULT_BYTE,
        .conv_num_each_intr = NR_SAMPLES,
        .adc1_chan_mask = adc1_chan_mask,
        .adc2_chan_mask = adc2_chan_mask,
        };
    ESP_ERROR_CHECK(adc_digi_initialize(&adc_dma_config));

    adc_digi_configuration_t dig_cfg =
        {
        .conv_limit_en = ADC_CONV_LIMIT_EN,
        .conv_limit_num = 0,
        //.sample_freq_hz = 20 * 1000,
        .sample_freq_hz = 4 * 1000,
        .conv_mode = ADC_CONV_SINGLE_UNIT_1,
        .format = ADC_DIGI_OUTPUT_FORMAT_TYPE2,
        };

    adc_digi_pattern_config_t adc_pattern[4] = {0};
    dig_cfg.pattern_num = channel_num;
    for (int i = 0; i < channel_num; i++)
        {
        uint8_t unit = GET_UNIT(channel[i]);
        uint8_t ch = channel[i] & 0x7;
        adc_pattern[i].atten = ADC_ATTEN_DB_11;
        adc_pattern[i].channel = ch;
        adc_pattern[i].unit = unit;
        adc_pattern[i].bit_width = SOC_ADC_DIGI_MAX_BITWIDTH;

        //ESP_LOGI(TAG, "adc_pattern[%d].atten is :%x", i, adc_pattern[i].atten);
        //ESP_LOGI(TAG, "adc_pattern[%d].channel is :%x", i, adc_pattern[i].channel);
        //ESP_LOGI(TAG, "adc_pattern[%d].unit is :%x", i, adc_pattern[i].unit);
        }
    dig_cfg.adc_pattern = adc_pattern;
    ESP_ERROR_CHECK(adc_digi_controller_configure(&dig_cfg));
    }

void app_main()
     {
...

    continuous_adc_init(adc1_chan_mask, adc2_chan_mask, channel, sizeof(channel) / sizeof(adc_channel_t));
...
       res_t res;
       index = 0;
       ret = ESP_OK;
       uint32_t b2read = NR_SAMPLES * ADC_RESULT_BYTE;
       adc_digi_start();
       while(index < NR_SAMPLES * ADC_RESULT_BYTE && ret == ESP_OK)
           {
           ret = adc_digi_read_bytes(res.result + index, b2read, &ret_num, 1000);
           if (ret == ESP_ERR_TIMEOUT || ret == ESP_ERR_INVALID_STATE)
               break;
           ESP_LOGI(TAG, "read chunk %d %d", index, ret_num);
           index += ret_num;
           b2read -= ret_num;
           }
       adc_digi_stop();
...
}
```
If you look in res.presult at chunk boundary from time to time you'll see the order of values is changed.

igrr, 2023-02-20:
This might be the same issue as https://github.com/espressif/esp-idf/issues/10636.

ves011, 2023-02-23:
hi @Icarus113,
Thx for taking your time to look into this issue!
Here is a summary of my results after 24 cycles with NR_SAMPLES = 400

- each cycle has 5 chunks with different length
(summary extracted from the log)
Line   369: I (17:01:57.648) ADC OP: read chunk 0 208
	Line   370: I (17:01:57.672) ADC OP: read chunk 208 388
	Line   371: I (17:01:57.697) ADC OP: read chunk 596 400
	Line   372: I (17:01:57.722) ADC OP: read chunk 996 400
	Line   373: I (17:01:57.747) ADC OP: read chunk 1396 204
	Line   775: I (17:02:00.018) ADC OP: read chunk 0 196
	Line   776: I (17:02:00.042) ADC OP: read chunk 196 388
	Line   777: I (17:02:00.067) ADC OP: read chunk 584 400
	Line   778: I (17:02:00.092) ADC OP: read chunk 984 400
	Line   779: I (17:02:00.117) ADC OP: read chunk 1384 216
	Line  1181: I (17:02:02.378) ADC OP: read chunk 0 184
	Line  1182: I (17:02:02.402) ADC OP: read chunk 184 384
	Line  1183: I (17:02:02.427) ADC OP: read chunk 568 400
	Line  1184: I (17:02:02.452) ADC OP: read chunk 968 400
	Line  1185: I (17:02:02.477) ADC OP: read chunk 1368 232
	Line  1587: I (17:02:04.738) ADC OP: read chunk 0 168
	Line  1588: I (17:02:04.762) ADC OP: read chunk 168 388
	Line  1589: I (17:02:04.787) ADC OP: read chunk 556 400
	Line  1590: I (17:02:04.812) ADC OP: read chunk 956 400
	Line  1591: I (17:02:04.837) ADC OP: read chunk 1356 244
	Line  1993: I (17:02:07.088) ADC OP: read chunk 0 156
	Line  1994: I (17:02:07.112) ADC OP: read chunk 156 388
	Line  1995: I (17:02:07.137) ADC OP: read chunk 544 400
	Line  1996: I (17:02:07.162) ADC OP: read chunk 944 400
	Line  1997: I (17:02:07.187) ADC OP: read chunk 1344 256
	Line  2403: I (17:02:09.448) ADC OP: read chunk 0 144
	Line  2404: I (17:02:09.472) ADC OP: read chunk 144 388
	Line  2405: I (17:02:09.497) ADC OP: read chunk 532 400
	Line  2406: I (17:02:09.522) ADC OP: read chunk 932 400
	Line  2407: I (17:02:09.547) ADC OP: read chunk 1332 268
	Line  2809: I (17:02:11.808) ADC OP: read chunk 0 132
	Line  2810: I (17:02:11.832) ADC OP: read chunk 132 384
	Line  2811: I (17:02:11.857) ADC OP: read chunk 516 400
	Line  2812: I (17:02:11.882) ADC OP: read chunk 916 400
	Line  2813: I (17:02:11.907) ADC OP: read chunk 1316 284
	Line  3215: I (17:02:14.168) ADC OP: read chunk 0 116
	Line  3216: I (17:02:14.191) ADC OP: read chunk 116 376
	Line  3217: I (17:02:14.216) ADC OP: read chunk 492 400
	Line  3218: I (17:02:14.241) ADC OP: read chunk 892 400
	Line  3219: I (17:02:14.266) ADC OP: read chunk 1292 308
	Line  3621: I (17:02:16.528) ADC OP: read chunk 0 92
	Line  3622: I (17:02:16.552) ADC OP: read chunk 92 388
	Line  3623: I (17:02:16.577) ADC OP: read chunk 480 400
	Line  3624: I (17:02:16.602) ADC OP: read chunk 880 400
	Line  3625: I (17:02:16.627) ADC OP: read chunk 1280 320
	Line  4027: I (17:02:18.888) ADC OP: read chunk 0 80
	Line  4028: I (17:02:18.912) ADC OP: read chunk 80 388
	Line  4029: I (17:02:18.937) ADC OP: read chunk 468 400
	Line  4030: I (17:02:18.962) ADC OP: read chunk 868 400
	Line  4031: I (17:02:18.987) ADC OP: read chunk 1268 332
	Line  4433: I (17:02:21.248) ADC OP: read chunk 0 68
	Line  4434: I (17:02:21.272) ADC OP: read chunk 68 388
	Line  4435: I (17:02:21.297) ADC OP: read chunk 456 400
	Line  4436: I (17:02:21.322) ADC OP: read chunk 856 400
	Line  4437: I (17:02:21.347) ADC OP: read chunk 1256 344
	Line  4839: I (17:02:23.618) ADC OP: read chunk 0 56
	Line  4840: I (17:02:23.642) ADC OP: read chunk 56 388
	Line  4841: I (17:02:23.667) ADC OP: read chunk 444 400
	Line  4842: I (17:02:23.692) ADC OP: read chunk 844 400
	Line  4843: I (17:02:23.717) ADC OP: read chunk 1244 356
	Line  5226: I (17:02:25.948) ADC OP: read chunk 0 44
	Line  5227: I (17:02:25.972) ADC OP: read chunk 44 388
	Line  5228: I (17:02:25.997) ADC OP: read chunk 432 400
	Line  5229: I (17:02:26.022) ADC OP: read chunk 832 400
[... 181 lines cut ...]
	else
		{
		ESP_LOGI(TAG, "cannot take adc mutex");
		esp_restart();
		}
...
    if (ret == ESP_OK)
		{
		indexd = indexc = 0;
		c_val = calloc(NR_SAMPLES, sizeof(uint16_t));
		if(!c_val)
			{
			ESP_LOGI(TAG, "not enough memory to allocate intermediate vector");
			return ESP_FAIL;
			}
		for(i = 0; i < NR_SAMPLES; i++)
			{
			ESP_LOGI(TAG, "%d %d %d", i, res.presult[i].type2.channel, res.presult[i].type2.data);
			if(res.presult[i].type2.channel == channel[0])
				c_val[indexc++] = res.presult[i].type2.data;
                      }
               }
...

```

ves011, 2023-03-01:
Hi

Now its clear samples are lost in between chunks.
In the picture below should be a 50Hz sinus signal. between chunk 1 and 2  some samples are lost :(
These are the chunks:
```
I (50433) --------------------- ADC read loop:: 72 / 268 /268 / 0
I (50433) ADC DMA: read chunk
I (50453) --------------------- ADC read loop:: 73 / 364 /632 / 0
I (50453) ADC DMA: read chunk
I (50473) --------------------- ADC read loop:: 74 / 400 /1032 / 0
I (50473) ADC DMA: read chunk
I (50503) --------------------- ADC read loop:: 75 / 400 /1432 / 0
I (50503) ADC DMA: read chunk
I (50523) --------------------- ADC read loop:: 76 / 168 /1600 / 0
```
[image]

Erlkoenig90, 2023-05-16:
I am experiencing the same issue on `ESP32-S3` with ESP-IDF `v5.0.2`. Samples are missing and the samplerate is lower than expected. I think I found the cause for it:

The ADC DMA callback [restarts](https://github.com/espressif/esp-idf/blob/bfe07cbef90d50cde089f8f0695c919dd7a73d1c/components/esp_adc/adc_continuous.c#L355) the DMA transfer after each DMA run, i.e. after the DMA completed all descriptors from the linked descriptor list. If this callback is not called fast enough, the ADC will produce more samples with DMA disabled/finished before the DMA transfer is restarted, so those samples will be lost.

There is a simple solution: Turn the descriptor list into a ring/loop, by [setting the last descriptors `next` pointer](https://github.com/espressif/esp-idf/blob/bfe07cbef90d50cde089f8f0695c919dd7a73d1c/components/hal/adc_hal.c#L253) to point to the first descriptor instead to `NULL`. This will make the ADC run continuously, restarting at the beginning of the buffer after finishing.

The [call to `adc_hal_digi_start`](https://github.com/espressif/esp-idf/blob/bfe07cbef90d50cde089f8f0695c919dd7a73d1c/components/esp_adc/adc_continuous.c#L355) in the DMA callback can then simply be removed. Directly [after copying the data](https://github.com/espressif/esp-idf/blob/bfe07cbef90d50cde089f8f0695c919dd7a73d1c/components/esp_adc/adc_continuous.c#L329) from the DMA buffer to the ringbuffer, we need to make the DMA descriptor available again for DMA by setting its `owner` field to `1`. Otherwise, the DMA will trigger an error. This should probably be wrapped in its own ADC HAL function.

If the software still can't keep up with the ADC sampling, the DMA buffer will overrun, i.e. the DMA will attempt to use a descriptor with its `owner` field set to `0`. To uncover such an issue, we should catch the `DSCR_ERR` interrupt and print an error message. To do this cleanly we should add another callback to the GDMA and use it from the continuous ADC driver.

I submitted a pull request #11411 which does all of this, including new APIs for setting the `owner` field to `1` and providing a callback fro the `DSCR_ERR` interrupt. This fixes the issue for me.

commit 2a8b9fb6 (2023-06-08): Fix #10804 by running continuous ADC DMA in endless loop instead of restarting after each run (descriptor chain) to avoid losing samples. Use descriptor error callback for GDMA to check for DMA buffer overrun.
commit 85ce6abd (2023-09-09): Fix #10804 by running continuous ADC DMA in endless loop instead of restarting after each run (descriptor chain) to avoid losing samples. Use descriptor error callback for GDMA to check for DMA buffer overrun.
closed 2023-06-08 by espressif-bot