espressif/esp-idf #10805: ESP-IDF v4.4.4 basic Blink example doesn't build "out of the box"

JBHunterI, 2023-02-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

ESP-IDF v4.4.4 basic "blink" example builds straight "out of the box" with a simple "idf.py build" command.

### What is the actual behavior?

"idf.py build" fails.

### Steps to reproduce.

1. Follow Programming Guide to create a "blink" example project in a new directory.
2. Build with "idf.py build" command for a ESP32-C3 DevKit target.
3. Observe the error displayed at the end of the failed build process.

### Build or installation Logs.

```plain
"idf.py build" results in:
`
  Syntax error in cmake code at

    C:/App/ESP/IDF/esp-idf-v4.4.4/tools/cmake/component.cmake:297

  when parsing string

    .;C:\App\ESP\IDF\esp-idf-v4.4.4/examples/common_components/led_strip/include

  Invalid character escape '\A'.
Call Stack (most recent call first):
  C:/App/ESP/IDF/esp-idf-v4.4.4/tools/cmake/component.cmake:473 (__component_add_include_dirs)
  main/CMakeLists.txt:5 (idf_component_register)

-- Configuring incomplete, errors occurred!
`
```

### More Information.

Why do I have to do this (below) in line 1 of the blink example's .\main\CMakeLists.txt and replace line 8 with line 9, yet line 6 is not a problem?

1:  string(REPLACE "\\\\" "/" IDF_PATH_NO_BACKSLASH $ENV{IDF_PATH})
2:  #message(STATUS "--> original is " $ENV{IDF_PATH} " and modified is " ${IDF_PATH_NO_BACKSLASH})
3:  #message(FATAL_ERROR)
4:
5:  idf_component_register(SRCS "blink_example_main.c"
6:                              "$ENV{IDF_PATH}/examples/common_components/led_strip/led_strip_rmt_ws2812.c"
7:                         INCLUDE_DIRS "."
8:  #                            "$ENV{IDF_PATH}/examples/common_components/led_strip/include"
9:                              "${IDF_PATH_NO_BACKSLASH}/examples/common_components/led_strip/include"
10:                        PRIV_REQUIRES "driver"                # for led_strip
11:                       )

Actually, why isn't line 6 and everything including/below line 8 already in the example to ensure it builds successfully out-of-the-box, instead of giving a "fatal error: led_strip.h: No such file or directory" message?

BTW, this missing line in .\CMakeLists.txt right after the cmake_minimum_required() also prevents the blink example from building successfully but doesn't seem to mind the backslashes:
    set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/led_strip)

So, why does the path passed to idf_component_register()'s INCLUDE_DIRS subcommand not handle backslashes?

igrr, 2023-02-21:
Hi @JBHunterI,

Normally when you define INCLUDE_DIRS of a component (or a CMake library, in general), you only need to specify the public and private include directories of that specific library. When you are using other libraries (such as led_strip in this case), you need to specify a dependency on that other library, then CMake will automatically add public include directories of the dependency (and all the transitive dependencies) to the list of include directories of the current library.

So I would suggest to modify your main component CMakeLists.txt file like this:

```c++
idf_component_register(SRCS "blink_example_main.c"
                       INCLUDE_DIRS "."
                       PRIV_REQUIRES led_strip)
```
(if your code uses `driver` component as well, change that to `PRIV_REQUIRES driver led_strip`.)

The original CMakeLists.txt file for the blink example in v4.4 doesn't declare any dependencies, which inside `main` component is equivalent as have dependencies on all components in the build ([docs](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-guides/build-system.html#main-component-requirements)).

Now, back to the original problem:

Could you please try to build the blink example without any modifications again, and then capture the following:
* complete output of `idf.py build`
* contents of build/CMakeCache.txt

and attach both to this issue. (Ideally, as text files.)

igrr, 2023-02-21:
As to your last question, i'm not sure, to be honest. I did a quick test and the behavior is the same for a plain CMake project on Windows, without ESP-IDF. I guess it is not normally a problem because library include paths are very rarely (if ever) specified as absolute paths. Most of the times the paths are relative to the CMakeLists.txt file, i.e. inside the library directory.

JBHunterI, 2023-02-21:
I couldn't find anything on [CMake web site](https://cmake.org/) because idf_component_register() is something specific to ESP-IDF, so I'm not sure how you could test the behavior "for a plain CMake project on Windows, without ESP-IDF". It's not that the path was absolute because writing the complete absolute path without backslashes was how I got around this problem. I just got tired of manually changing it for the different versions/installations of IDF. Like you say, hopefully I won't have to add any include paths using environment or CMake variables. I appreciate your clarifications.

closed 2023-02-21 by JBHunterI