espressif/esp-idf #10807: Urgent WiFi PM stop error. While in examples\wifi\getting_started\station\

abhi-ratnman, 2023-02-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

While running the wifi getting started example in station mode. The below error is thrown. It says PM stop working. The issue is pertaining to some ESP32 modules. In a batch of 10 this issue was found in 4 ESP32 while other 6 ESP32 are working on the same code. 

`I (27) boot: ESP-IDF -128-NOTFOUND 2nd stage bootloader
I (27) boot: compile time 19:45:20
I (27) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1304ch ( 77900) map
I (138) esp_image: segment 1: paddr=00023074 vaddr=3ffb0000 size=037c4h ( 14276) load
I (144) esp_image: segment 2: paddr=00026840 vaddr=40080000 size=097d8h ( 38872) load
I (160) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6f960h (457056) map
I (326) esp_image: segment 4: paddr=0009f988 vaddr=400897d8 size=0b164h ( 45412) load
I (355) boot: Loaded app from partition at offset 0x10000
I (355) boot: Disabling RNG early entropy source...
I (367) cpu_start: Pro cpu up.
I (367) cpu_start: Starting app cpu, entry point is 0x400811ac
0x400811ac: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (381) cpu_start: Pro cpu start user code
I (381) cpu_start: cpu freq: 160000000
I (381) cpu_start: Application information:
I (385) cpu_start: Project name:     wifi_station
I (391) cpu_start: App version:      1
I (395) cpu_start: Compile time:     Dec 16 2022 19:44:50
I (401) cpu_start: ELF file SHA256:  f7a1689d6cd49c70...
I (407) cpu_start: ESP-IDF:          -128-NOTFOUND
I (413) heap_init: Initializing. RAM available for dynamic allocation:
I (420) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (426) heap_init: At 3FFB74D0 len 00028B30 (162 KiB): DRAM
I (432) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (439) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (445) heap_init: At 4009493C len 0000B6C4 (45 KiB): IRAM
I (453) spi_flash: detected chip: generic
I (456) spi_flash: flash io: dio
W (460) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (474) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (543) wifi station: ESP_WIFI_MODE_STA
I (553) wifi:wifi driver task: 3ffc0198, prio:23, stack:6656, core=0
I (553) system_api: Base MAC address is not set
I (553) system_api: read default base MAC address from EFUSE
I (573) wifi:wifi firmware version: 2c582c7
I (573) wifi:wifi certification version: v7.0
I (573) wifi:config NVS flash: enabled
I (573) wifi:config nano formating: disabled
I (573) wifi:Init data frame dynamic rx buffer num: 32
I (583) wifi:Init management frame dynamic rx buffer num: 32
I (583) wifi:Init management short buffer num: 32
I (593) wifi:Init dynamic tx buffer num: 32
I (593) wifi:Init static rx buffer size: 1600
I (593) wifi:Init static rx buffer num: 10
I (603) wifi:Init dynamic rx buffer num: 32
I (603) wifi_init: rx ba win: 6
I (613) wifi_init: tcpip mbox: 32
I (613) wifi_init: udp mbox: 6
I (613) wifi_init: tcp mbox: 6
I (623) wifi_init: tcp tx win: 5744
I (623) wifi_init: tcp rx win: 5744
I (633) wifi_init: tcp mss: 1440
I (633) wifi_init: WiFi IRAM OP enabled
I (633) wifi_init: WiFi RX IRAM OP enabled
I (643) wifi:Set ps type: 0

I (653) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (743) wifi:mode : sta (b4:8a:0a:42:79:a8)
I (743) wifi:enable tsf
I (743) wifi station: wifi_init_sta finished.
I (843) wifi:new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
I (843) wifi:state: init -> auth (b0)
I (883) wifi:state: auth -> assoc (0)
I (893) wifi:state: assoc -> run (10)
I (943) wifi:connected with Darshna, aid = 18, channel 13, BW20, bssid = aa:56:fc:d0:54:f6
I (943) wifi:security: WPA2-PSK, phy: bgn, rssi: -59
I (953) wifi:pm start, type: 0

I (963) wifi:state: run -> init (5a0)
I (963) wifi:pm stop, total sleep time: 0 us / 12120 us

I (963) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (973) wifi station: retry to connect to the AP
I (973) wifi station: connect to the AP fail
I (1043) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (1053) wifi:state: init -> auth (b0)
I (1053) wifi:state: auth -> assoc (0)
I (1063) wifi:state: assoc -> run (10)
I (1083) wifi:connected with Darshna, aid = 19, channel 13, BW20, bssid = aa:56:fc:d0:54:f6
I (1083) wifi:security: WPA2-PSK, phy: bgn, rssi: -58
I (1093) wifi:pm start, type: 0

I (1103) wifi:state: run -> init (5a0)
I (1103) wifi:pm stop, total sleep time: 0 us / 6676 us

I (1103) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (1103) wifi station: retry to connect to the AP
I (1113) wifi station: connect to the AP fail
I (1153) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (1153) wifi:state: init -> auth (b0)
I (1153) wifi:state: auth -> assoc (0)
I (1163) wifi:state: assoc -> run (10)
I (1183) wifi:connected with Darshna, aid = 20, channel 13, BW20, bssid = aa:56:fc:d0:54:f6
I (1193) wifi:security: WPA2-PSK, phy: bgn, rssi: -58
I (1193) wifi:pm start, type: 0

I (1203) wifi:state: run -> init (5a0)
I (1203) wifi:pm stop, total sleep time: 0 us / 10325 us

I (1203) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (1213) wifi station: retry to connect to the AP
I (1213) wifi station: connect to the AP fail
I (1253) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (1253) wifi:state: init -> auth (b0)
I (1253) wifi:state: auth -> assoc (0)
I (1273) wifi:state: assoc -> run (10)
I (1293) wifi:connected with Darshna, aid = 21, channel 13, BW20, bssid = aa:56:fc:d0:54:f6
I (1293) wifi:security: WPA2-PSK, phy: bgn, rssi: -56
I (1293) wifi:pm start, type: 0

I (1303) wifi:state: run -> init (5a0)
I (1303) wifi:pm stop, total sleep time: 0 us / 8658 us

I (1303) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (1313) wifi station: retry to connect to the AP
I (1313) wifi station: connect to the AP fail
I (1353) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (1353) wifi:state: init -> auth (b0)
I (1363) wifi:state: auth -> assoc (0)
I (1373) wifi:state: assoc -> run (10)
I (1393) wifi:connected with Darshna, aid = 22, channel 13, BW20, bssid = aa:56:fc:d0:54:f6
I (1393) wifi:security: WPA2-PSK, phy: bgn, rssi: -55
I (1403) wifi:pm start, type: 0

I (1413) wifi:state: run -> init (5a0)
I (1413) wifi:pm stop, total sleep time: 0 us / 10375 us

I (1413) wifi:new:<13,0>, old:<13,0>, ap:<255,255>, sta:<13,0>, prof:1
I (1423) wifi station: retry to connect to the AP
I (1423) wifi station: connect to the AP fail`

abhi-ratnman, 2023-02-22:
The same code works with other router but not with this one. 

Logs when ESP32 does not connect with the AP

` 

> I (2085) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
> D (2095) phy_init: loading PHY init data from application binary
> D (2105) nvs: nvs_open_from_partition phy 0
> D (2105) nvs: nvs_get cal_version 4
> D (2115) nvs: nvs_get_str_or_blob cal_mac
> D (2115) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
> D (2125) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
> D (2125) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
> D (2135) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
> D (2145) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
> D (2145) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
> D (2155) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
> D (2165) nvs: nvs_get_str_or_blob cal_data
> D (2175) nvs: nvs_close 4
> D (2175) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
> D (2175) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
> D (2185) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
> D (2195) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
> D (2195) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
> D (2205) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
> D (2215) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
> D (2305) wifi:filter: set rx policy=0
> I (2305) wifi:mode : sta (b4:8a:0a:42:79:a8) + softAP (b4:8a:0a:42:79:a9)
> I (2305) wifi:enable tsf
> D (2305) wifi:filter: set rx policy=1
> D (2305) wifi:connect status 0 -> 0
> D (2315) wifi:filter: set rx policy=9
> I (2315) wifi:Total power save buffer number: 16
> I (2315) wifi:Init max length of beacon: 752/752
> I (2325) wifi:Init max length of beacon: 752/752
> D (2325) event: running post WIFI_EVENT:2 with handler 0x400f1fac and context 0x3ffc05f8 on loop 0x3ffbf458
> 0x400f1fac: wifi_default_action_sta_start at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_wifi/src/wifi_default.c:67
> 
> D (2335) wifi_init_default: wifi_start esp-netif:0x3ffc07a0 event-id2
> D (2345) wifi_init_default: WIFI mac address: b4 8a a 42 79 a8
> D (2345) esp_netif_handlers: esp_netif action has started with netif0x3ffc07a0 from event_id=2
> D (2355) esp_netif_lwip: check: remote, if=0x3ffc07a0 fn=0x400ef900
> 0x400ef900: esp_netif_start_api at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/lwip/esp_netif_lwip.c:700
> 
> 
> I (2365) pWifi: wifi_init_sta finished.
> D (2365) esp_netif_lwip: esp_netif_start_api 0x3ffc07a0
> D (2375) esp_netif_lwip: esp_netif_get_hostname esp_netif:0x3ffc07a0
> D (2385) esp_netif_lwip: check: local, if=0x3ffc07a0 fn=0x400f0338
> 0x400f0338: esp_netif_update_default_netif_lwip at C:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/lwip/esp_netif_lwip.c:188
> 
> 
> D (2385) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffc07a0
> D (2395) esp_netif_lwip: call api in lwip: ret=0x0, give sem
> D (2395) event: running post WIFI_EVENT:2 with handler 0x400db1c8 and context 0x3ffc7968 on loop 0x3ffbf458
> 0x400db1c8: event_handler at C:/Espressif/frameworks/esp-idf-v4.4/projects/temp_mqtt/components/pWifi/pWifi.c:125
> 
> I (2405) pWifi: WIFI_EVENT_STA_START
> D (2415) wifi:Start wifi connect
> D (2415) wifi:connect status 0 -> 0
> D (2415) wifi:connect chan=0
> D (2425) wifi:first chan=6
> D (2425) wifi:connect status 0 -> 1
> D (2425) wifi:filter: set rx policy=3
> D (2435) wifi:clear scan ap list
> D (2435) wifi:start scan: type=0x50f, priority=2, cb=0x40145cc8, arg=0x0, ss_state=0x1, time=132968, index=0
> 0x40145cc8: cnx_start_handoff_cb at ??:?
> 
> D (2445) wifi:perform scan: ss_state=0x9, chan<6,0>, dur<0,120>
[... 309 lines cut ...]
> I (5168) app_main: Free heap size 201kB  Filedata = 0
> I (6178) app_main: Free heap size 201kB  Filedata = 0
> D (6558) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffc0818
> D (6558) esp_netif_lwip: if0x3ffc0798 ip changed=1
> D (6558) event: running post IP_EVENT:0 with handler 0x400f1e30 and context 0x3ffc0724 on loop 0x3ffbf450
> 0x400f1e30: lwip_recv_tcp_from at C:/esp/esp-idf/components/lwip/lwip/src/api/sockets.c:1137
> 
> D (6568) wifi_init_default: Got IP wifi default handler entered
> D (6568) esp_netif_handlers: esp_netif action got_ip with netif0x3ffc0798 from event_id=0
> I (6578) esp_netif_handlers: sta ip: 192.168.0.108, mask: 255.255.255.0, gw: 192.168.0.1
> D (6588) event: running post IP_EVENT:0 with handler 0x400db1c8 and context 0x3ffc7994 on loop 0x3ffbf450
> 0x400db1c8: event_handler at C:/projects2/genAuto/components/pWifi/pWifi.c:141 (discriminator 13)
> 
> D (6598) pWifi: GOT IP_EVENT_STA_GOT_IP
> I (6598) pWifi: Got IP: 192.168.0.108
> I (6608) pWifi: Internet check start.
> I (6618) pWifi: Resolved IPv4 address: 142.250.183.196
> I (7178) pWifi: start MQTT
> D (7178) MQTT_CLIENT: MQTT client_id=ESP32_7c0C30
> D (7178) event: created event loop 0x3ffce8ac
> D (7178) MQTT_CLIENT: Core selection disabled
> I (7178) app_main: Free heap size 183kB  Filedata = 0
> D (7178) event: running post MQTT_EVENTS:7 with handler 0x400d82e8 and context 0x3ffcf1ec on loop 0x3ffce8ac 

`

shreyassheth1, 2023-02-23:
Hey from logs it can be observed that esp32 is able to connect with following APs
- Pegasus  `(2658) wifi:connected with Pegasus, aid = 8, channel 5, 40D, bssid = 04:95:e6:5d:a2:58`
- Galaxy Group 0  `(2675) wifi:connected with Galaxy Group O, aid = 2, channel 6, 40D, bssid = 60:a4:b7:06:c4:07`

The potential problem observed with AP whose ssid is Darshna could be that AP is not able to clear the state of connecting esp32
which can be seen in the following logs 
- I (943) wifi:connected with Darshna, aid = 18, channel 13, BW20, bssid = aa:56:fc:d0:54:f6
- I (1083) wifi:connected with Darshna, aid = 19, channel 13, BW20, bssid = aa:56:fc:d0:54:f6
- I (1183) wifi:connected with Darshna, aid = 20, channel 13, BW20, bssid = aa:56:fc:d0:54:f6
- I (1293) wifi:connected with Darshna, aid = 21, channel 13, BW20, bssid = aa:56:fc:d0:54:f6
- I (1393) wifi:connected with Darshna, aid = 22, channel 13, BW20, bssid = aa:56:fc:d0:54:f6

Here, after a successful connection with Darshna after 10 milliseconds, from the logs  `I (1303) wifi:state: run -> init (5a0)` which implies that AP sends Disassociation frame to the station with reason code ASSOC_TOO_MANY. To debug it, I will require a sniffer capture to validate whether esp32 is transmitting assoc req even after receiving assoc resp or AP was unable to handle new connection.

To guide you through the setup for capturing sniffer [setup](https://github.com/espressif/esp-idf/issues/8687#issuecomment-1105070843)

abhi-ratnman, 2023-02-23:
Hi @shreyassheth1

The issue here is that one of the ESP32 is able to connect to Darshna by using WiFi station mode example with darshna, but other ESP32 is not able to connect with Darshna using the same WiFi station mode example. Is is possible that the router will allow one ESP32 to connect and will not allow other?

Also check the link below. They are also facing a similar issue. 
[issue 6427](https://github.com/espressif/esp-idf/issues/6427)

Also attaching the wire shark and log output for your refrence
[
[image]
[logs.txt](https://github.com/espressif/esp-idf/files/10814290/logs.txt)
[image]
](url)

shreyassheth1, 2023-02-24:
Hey @abhi-ratnman looks like there is some problem capturing sniffer
Please follow these steps
Check for your interface name using `iwconfig` mostly it is `wlan0` 
`ifconfig wlan0 down`
`iwconfig wlan0 mode monitor`
`ifconfig wlan0 up`
`iwconfig wlan0 channel 13`
`wireshark`
Please make sure you are selecting the interface same as `wlan0` in wireshark and please attach the whole capture here if possible (pcapng) 

Also, I would request you to test this, `The issue here is that one of the ESP32 is able to connect to Darshna by using WiFi station mode example with darshna, but other ESP32 is not able to connect with Darshna using the same WiFi station mode example` if this is easily reproducible please try to reproduce it and when this happens `ESP32 is not able to connect with Darshna` please try to connect a completely new client(laptop, mobile) to this AP which has never connected to this network.
The reason to test this is we think AP has reached its limit for handling new stations and is not able to clear previous states.
Also, can you check for both esp32 connecting with another router?

abhi-ratnman, 2023-02-24:
@shreyassheth1 

There is no issue with the limit of WiFi AP. The AP is connected to only 1 other device. Have tested the ESP32 that is not connecting with AP with 3 diffrent AP (mobile hotspot, laptop hotspot diffrent router). Have tested everything possible but still this will not connect. 

Have tested both ESP32 connection with other 3 router the same issue persist. I have been facing this issue since last 4-5 days. Have tried and exhausted all the possible reason that should cause this. But the final conclusion I can come on is it is working on one ESP32 and it is not working on the other one. If you are based in India I can ship the device to you and you can check it. But that will delay the process further. 

Also regarding testing if you wish I can give you access to a computer with both the ESP32 connected to it and show it to you via team viewer and if you wish to run some test I can allocate the PC to you for your testing.

shreyassheth1, 2023-02-25:
Hey @abhi-ratnman thank you for your co-ordination 
Can you please try these things before we move forward
1 ) Sniffer capture, for an interval of the start of the connection to disassociation.
2) Try `idf.py erase-flash` and then flash the example again.
3) Use this `esp_wifi_set_ps(WIFI_PS_NONE);` in the example code to turn off power save mode.

idea--list, 2023-03-05:
@shreyassheth1  @abhi-ratnman 
I am also struggling for a while with WiFi reliability issues when both auto light sleep feature is enabled and also `esp_wifi_set_ps(WIFI_PS_MAX_MODEM)` is used.
In my particular case i am logging data to files and send those periodically over WiFi (using FTP). The data sensor issues interrupt signals ca. every 30ms, then the ESP32S3 module stores data to the file on an SD card. Is a time period over, the file is send to the cloud. This works bullet-proof when the proposed fix for the bug 3.14 published in the errata is also applied.

However as soon as i set `esp_wifi_set_ps(WIFI_PS_MAX_MODEM)` to extend battery life of my project, then i always observe these:
- After succesfully sending some files, there is something happening and no more files are being sent for about 10 minutes.
- During that phase also power consumption is elevated and it feels as if interrupt detection would only function partially.
- After such a period power consumption normalizes and also files are being sent again over FTP.
- However such periods can occur several times during a session.

Up until now i could not find out what exactly is happening and as a consequence i have no idea how to solve or at least circumvent the issue. Obviously interrupt handling of ESP32s is not perfect, already published solutions seem to work fine when auto light sleep is disabled and `WIFI_PS_NONE` is used. However when using auto light sleep and `WIFI_PS_MAX_MODEM`, then things tend to break again. I know these are just observations/findings but i could not narrow it down up until now. Hope it still makes some sense for those who maintain WiFi related part of ESP-IDF.

shreyassheth1, 2024-03-14:
@idea--list
Hey we tried reproducing the same scenario and were not able to see the issue, if you are still facing the issue can you please provide us with the following info which will help us to debug the issue.
1) Enable debug logs under `menuconfig` -> `Component config` -> `Log output` -> `Default log verbosity(set to debug)`
2) Please share your `sdkconfig`
3) If possible please help us with capturing packets using sniffer
Here are some resources that might help you to capture 80211 packets.
If you are using a Windows machine, try: https://www.youtube.com/watch?v=As9BUoOMk_A
If you are using an iOS machine, try: https://www.macupdate.com/app/mac/56809/airtool/old-versions
If you are using a Linux machine, try:
```
ifconfig wlp0s20f3 down
iwconfig wlp0s20f3 mode monitor
ifconfig wlp0s20f3 up
iwconfig wlp0s20f3 channel 3
wireshark
```

Sherry616, 2024-05-15:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. 
Thanks for using our Espressif product.

closed 2024-05-15 by Sherry616