espressif/esp-idf #10808: in hfp_ag ,how to play the sound from earphone when send single wave to earphone at same time

zhou88mang, 2023-02-21:
### Is your feature request related to a problem?

idf version 4.4.4
i use the exmple (hfp_ah)  from idf,  config  sco path  HCI  , 
 BR/EDR Sync(SCO/eSCO) Max Connections = 1 ,
enable WBS,
esp_bd_addr_t hf_peer_addr = {0x41, 0x42, 0x01, 0xB6, 0x36,0x4F};//my earphone mac address

i used UDA1334A   I2S Stereo Decoder  to connect the esp32, so i want to hear sound from UDA1334A and also hear sound from my headset.

### Describe the solution you'd like.

now , i  just hear sine wave from my headset , but what i heared from UDA1334A were  all noise  

### Describe alternatives you've considered.

`#define SAMPLE_RATE     (44100)
#define I2S_NUM         (1)
//#define WAVE_FREQ_HZ    (100)
#define PI              (3.14159265)
#define I2S_BCK_IO      (GPIO_NUM_5)
#define I2S_WS_IO       (GPIO_NUM_25)
#define I2S_DO_IO       (GPIO_NUM_26)
#define I2S_DI_IO       (-1)

// init i2s
    i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_TX,
        .sample_rate = SAMPLE_RATE,
        .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
        .communication_format = I2S_COMM_FORMAT_STAND_MSB,
        .dma_buf_count = 6,
        .dma_buf_len = 60,
        .use_apll = false,
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1                                //Interrupt level 1
    };
    i2s_pin_config_t pin_config = {
        .mck_io_num = I2S_PIN_NO_CHANGE,
        .bck_io_num = I2S_BCK_IO,
        .ws_io_num = I2S_WS_IO,
        .data_out_num = I2S_DO_IO,
        .data_in_num = I2S_DI_IO                                               //Not used
    };
    i2s_driver_install(I2S_NUM, &i2s_config, 0, NULL);
    i2s_set_pin(I2S_NUM, &pin_config);

//bt_app_hf_incoming_cb
static void bt_app_hf_incoming_cb(const uint8_t *buf, uint32_t sz)
{
     size_t i2s_bytes_write = 0;
    
     i2s_set_clk(1,16000, 16, 2);
     i2s_write(1, buf, sz, &i2s_bytes_write, 100);
    // printf("i2s_bytes_write = %d\n",i2s_bytes_write);
}
//bt_app_hf_outgoing_cb
static uint32_t bt_app_hf_outgoing_cb(uint8_t *p_buf, uint32_t sz)
{
    size_t item_size = 0;
    uint8_t *data;
    if (!s_m_rb) {
        return 0;
    }
    vRingbufferGetInfo(s_m_rb, NULL, NULL, NULL, NULL, &item_size);
    if (item_size >= sz) {
        data = xRingbufferReceiveUpTo(s_m_rb, &item_size, 0, sz);
        memcpy(p_buf, data, item_size);
        vRingbufferReturnItem(s_m_rb, data);
        return sz;
    } else {
        // data not enough, do not read\n
        return 0;
    }
    return 0;
}`

i can not figure out how to set the value of sample rate of i2s ,maybe this is the one  reason of my problem, and when i config sco path pcm , I can  hear sound from my headset clearly. 

### Additional context.

this is log

`hfp_ag>con
I (9031) BT_APP_HF: APP HFP event: CONNECTION_STATE_EVT
I (9031) BT_APP_HF: --connection state CONNECTED, peer feats 0x0, chld_feats 0x0
E (9071) BT_BTC: btc_hf_cb_handler: Invalid index 14868
I (9141) BT_APP_HF: APP HFP event: CIND_RESPONSE_EVT
I (9141) BT_APP_HF: --CIND Start.
I (9151) BT_APP_HF: APP HFP event: CONNECTION_STATE_EVT
I (9161) BT_APP_HF: --connection state SLC_CONNECTED, peer feats 0x29f, chld_feats 0x4010
I (9241) BT_APP_HF: APP HFP event: UNKNOW_AT_CMD
I (9241) BT_APP_HF: --UNKOW AT CMD: +XAPL=000D-0001-0101,2
I (9391) BT_APP_HF: APP HFP event: VOLUME_CONTROL_EVT
I (9401) BT_APP_HF: --Volume Target: MICROPHONE, Volume 9
I (9401) BT_APP_HF: APP HFP event: VOLUME_CONTROL_EVT
I (9411) BT_APP_HF: --Volume Target: SPEAKER, Volume 15

hfp_ag>
hfp_ag>
hfp_ag> cona
Connect Audio
I (13831) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT
I (13831) BT_APP_HF: --Audio State connecting
hfp_ag> W (13841) BT_BTM: BTM Remote does not support 3-EDR eSCO
I (13841) BT_APP_HF: APP HFP event: BCS_EVT
I (13851) BT_APP_HF: --Consequence of codec negotiation: Use MSBC
E (13851) BT_BTM: btm_sco_connected, handle 180
I (13851) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT
I (13871) BT_APP_HF: --Audio State connected_msbc
W (14131) BT_APPL: bta_ag_sco_read_cback: status(1)
W (14211) BT_APPL: bta_ag_sco_read_cback: status(2)
W (14481) BT_APPL: bta_ag_sco_read_cback: status(2)
W (14501) BT_APPL: bta_ag_sco_read_cback: status(1)
W (14801) BT_APPL: bta_ag_sco_read_cback: status(1)
W (15231) BT_APPL: bta_ag_sco_read_cback: status(1)
W (15491) BT_APPL: bta_ag_sco_read_cback: status(1)
W (15761) BT_APPL: bta_ag_sco_read_cback: status(2)
W (15821) BT_APPL: bta_ag_sco_read_cback: status(1)
W (15851) BT_APPL: bta_ag_sco_read_cback: status(2)
W (16421) BT_APPL: bta_ag_sco_read_cback: status(2)
W (17261) BT_APPL: bta_ag_sco_read_cback: status(2)
W (17771) BT_APPL: bta_ag_sco_read_cback: status(1)
W (18571) BT_APPL: bta_ag_sco_read_cback: status(1)
W (18841) BT_APPL: bta_ag_sco_read_cback: status(1)
W (19151) BT_APPL: bta_ag_sco_read_cback: status(1)
W (20651) BT_APPL: bta_ag_sco_read_cback: status(1)
W (20681) BT_APPL: bta_ag_sco_read_cback: status(1)
W (20861) BT_APPL: bta_ag_sco_read_cback: status(1)
W (21001) BT_APPL: bta_ag_sco_read_cback: status(1)
W (21041) BT_APPL: bta_ag_sco_read_cback: status(1)
W (21071) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(50 150)
W (21081) BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x24
E (21091) BT_APPL: bta_dm_pm_btm_status hci_status=36
W (21201) BT_APPL: bta_ag_sco_read_cback: status(1)
W (21761) BT_APPL: bta_ag_sco_read_cback: status(1)
W (22191) BT_APPL: bta_ag_sco_read_cback: status(1)
W (22501) BT_APPL: bta_ag_sco_read_cback: status(1)
W (23061) BT_APPL: bta_ag_sco_read_cback: status(1)
W (23221) BT_APPL: bta_ag_sco_read_cback: status(2)
W (23291) BT_APPL: bta_ag_sco_read_cback: status(2)
W (23811) BT_APPL: bta_ag_sco_read_cback: status(2)
W (24031) BT_APPL: bta_ag_sco_read_cback: status(2)
W (24081) BT_APPL: bta_ag_sco_read_cback: status(2)
W (24191) BT_APPL: bta_ag_sco_read_cback: status(2)
W (25141) BT_APPL: bta_ag_sco_read_cback: status(1)
W (25151) BT_APPL: bta_ag_sco_read_cback: status(1)
W (25181) BT_APPL: bta_ag_sco_read_cback: status(2)
W (27291) BT_APPL: bta_ag_sco_read_cback: status(2)
W (27801) BT_APPL: bta_ag_sco_read_cback: status(1)
W (28201) BT_APPL: bta_ag_sco_read_cback: status(2)
W (28521) BT_APPL: bta_ag_sco_read_cback: status(2)
W (30211) BT_APPL: bta_ag_sco_read_cback: status(1)
W (30261) BT_APPL: bta_ag_sco_read_cback: status(2)
W (30861) BT_APPL: bta_ag_sco_read_cback: status(2)
W (30981) BT_APPL: bta_ag_sco_read_cback: status(2)
W (31281) BT_APPL: bta_ag_sco_read_cback: status(1)
W (34591) BT_APPL: bta_ag_sco_read_cback: status(1)
W (35881) BT_APPL: bta_ag_sco_read_cback: status(2)
W (36191) BT_APPL: bta_ag_sco_read_cback: status(2)
W (36701) BT_APPL: bta_ag_sco_read_cback: status(2)
W (37121) BT_APPL: bta_ag_sco_read_cback: status(2)
W (37231) BT_APPL: bta_ag_sco_read_cback: status(2)
`

esp-qing, 2024-02-23:
Hi, @zhou88mang  Sorry for replying so late, could you please tell me if this issue still exists?

Alvin1Zhang, 2024-07-24:
Thanks for reporting, seems this is pretty similar to https://github.com/espressif/esp-idf/issues/11937, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2024-07-24 by Alvin1Zhang