espressif/esp-idf #10809: MODBUS invalid response issue after migrating to esp-idf v5.0

AshvajitPXy, 2023-02-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 | Custom board

### Power Supply used.

USB

### What is the expected behavior?

I have implemented custom MODBUS master to read/write multiple registers over UART to an onboard STM32 controller. This implemenation was done in esp-idf version v4.3 and migrated to v4.4 also. In pervious version MODBUS master is working fine .

### What is the actual behavior?

After migrating to esp-idf v5.0. MODBUS is working fine for few minutes then it show Invalid response from MODBUS slave and stays for few minutes and   after soft reset or after some time automatically it starts working again. 

### Steps to reproduce.

//User operation function to read slave values and check alarm
void master_operation_func(void)
{
	esp_err_t err = ESP_OK;
	/*-------------------------------------*/
	mb_param_request_t request;		// Instance created for multiple read / write registers
	/*-------------------------------------*/

	//ESP_LOGI(MASTER_TAG, "Start modbus...");
	switch (state)
	{
	case 0:	
			//  MODBUS Read Schedule
		/*-------------------------------------*/
		request.command = MB_READ_HOLD_REG;
		request.slave_addr = MB_DEVICE_ADDR1;
		request.reg_start = HPID_MBREG;
		request.reg_size = 51;
		err = mbc_master_send_request(&request, (uint8_t*)(MBRegRx_Buff));	// Request to read multiple holding registers
		if (err == ESP_OK)
		{
			DataSync_flag++;
			ESP_LOGI(MASTER_TAG, "Read active parameters part :%d ",DataSync_flag);
			
		}
		else
		{	
			ESP_LOGE(MASTER_TAG, "=== Read active value failure 1 ===");
		}
		/*-------------------------------------*/
		/*-------------------------------------*/
		request.command = MB_READ_HOLD_REG;
		request.slave_addr = MB_DEVICE_ADDR1;
		request.reg_start = HMI_ID_MBREG;
		request.reg_size = 37;
		err = mbc_master_send_request(&request, (uint8_t*)(MBRegRx_Buff + HMI_ID_CID));	// Request to read multiple holding registers
		if (err == ESP_OK)
		{
			DataSync_flag++;
			ESP_LOGI(MASTER_TAG, "Read active parameters part :%d ", DataSync_flag);
		}
		else
		{
			ESP_LOGE(MASTER_TAG, "=== Read active value failure 2 ===");
		}
		/*-------------------------------------*/
		/*-------------------------------------*/
		request.command = MB_READ_HOLD_REG;
		request.slave_addr = MB_DEVICE_ADDR1;
		request.reg_start = APP_STATUS_FLAG_MBREG;
		request.reg_size = 9;
		err = mbc_master_send_request(&request, (uint8_t*)(MBRegRx_Buff + APP_STATUS_WORD_CID));
		if (err == ESP_OK)
		{
			DataSync_flag++;
			ESP_LOGI(MASTER_TAG, "Read active parameters part :%d ", DataSync_flag);
		}
		else
		{
			ESP_LOGE(MASTER_TAG, "=== Read active value failure 3 === %d", err);
		}
    	
    	// update HMI command word from read
        for(int i = 0 ; i < 16 ; i++)
    	{
        	if (MBRegRx_Buff[HMI_CMD_WORD_CID] & (1 << i))
            	MBCommandWordRx[i] = 1;
        	else
            	MBCommandWordRx[i] = 0;
    	}
    	// Update fault history
    	for(int i = 0 ; i < 4 ; i++)
    	{
        	FaultHistory[i] = (MBRegRx_Buff[FAULT_HIST_CID] >> (4 * i)) & 0x0F;
    	}
		ESP_LOGW(MASTER_TAG, "=== State change to MB Write ===");

		if ((DataSync_flag == 3) && SyncCheckForID())
		{
			DataSync_flag = 0;
			state++; // Set to write only when 3 writes are Successfull and sync with App Controller is completed
			ESP_LOGW(MASTER_TAG, "=== State change to MB Write ===");
		}
		else
		{
			ESP_LOGW(MASTER_TAG, " State change failed , flag :%d MBReg : %d.", DataSync_flag, MBRegRx_Buff[SYNCHRONIZE_CID]);
    		DataSync_flag = 0;
		}
		/*-------------------------------------*/
	break;
	case 1 :	// MODBUS Write Schedule
		/*-------------------------------------*/
		MBReg_Buff[PRESS_RANGE_CID] = 1;		// TODO : Default value to be given at startup 
        // update HMI command word for write
        for(int i = 0 ; i < 16 ; i++)
    	{
        	if (MBCommandWord[i])
            	MBReg_Buff[HMI_CMD_WORD_CID] |= 1 << i;
        	else
            	MBReg_Buff[HMI_CMD_WORD_CID] &= ~(1 << i);
    	}

        	
		request.command = MB_WRITE_HOLD_REG;
		request.slave_addr = MB_DEVICE_ADDR1;
		request.reg_start = COMMAND_WORD_MBREG;
		request.reg_size = 15;
		err = mbc_master_send_request(&request, (uint8_t*)(MBReg_Buff + COMMAND_WORD_MBREG_CID));
		if (err == ESP_OK)
		{
			ESP_LOGI(MASTER_TAG, "Written Menu parameters part 1 ");
		}
		else
		{
			ESP_LOGE(MASTER_TAG, "=== Write value failure 1 ===");
		}
		/*-------------------------------------*/

		/*-------------------------------------*/
		request.command = MB_WRITE_HOLD_REG;
		request.slave_addr = MB_DEVICE_ADDR1;
		request.reg_start = PRODUCT_ID_MBREG;
		request.reg_size = 36;
		err = mbc_master_send_request(&request, (uint8_t*)(MBReg_Buff + PRODUCT_ID_CID));
		if (err == ESP_OK)
		{
			ESP_LOGI(MASTER_TAG, "Written Menu parameters part 2");
		}
		else
		{
			ESP_LOGE(MASTER_TAG, "=== Write value failure 2 ===");
		}
		/*-------------------------------------*/
		state = 0;
	break;
}

	#if 0
    ESP_LOGI(MASTER_TAG, "Destroy master...");
    ESP_ERROR_CHECK(mbc_master_destroy());
	#endif
}

### More Information.

[modbus.log](https://github.com/espressif/esp-idf/files/10792114/modbus.log)

alisitsyn, 2023-02-21:
@ AshvajitPXy ,

According to log. On some stage of the communication your application stops communication and even does not write a log for along time. Please check your active objects and try to understand what can block your processing algorithm for long time. Check the priorities of your tasks are not significatly bigger then Modbus task priority. Also please enable debug log severity in the kconfig menu and recompile project then place log here.

I (140774) MASTER_TEST: Written Menu parameters part 2 <<<<<<<<<<<<< some task has blocked processing for a long time
E (195124) MB_CONTROLLER_MASTER: mbc_master_send_request(93): Master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).

AshvajitPXy, 2023-02-21:
@alisitsyn 
I have disabled the logs manually so that I don't have to scroll too much to find these error logs.
 I tired to set log level to Verbose but it made all the process too slow and I was not able to reproduce this issue.
Is there any way to just able to set Debug level log for MODBUS only ?
I also recently found that these were false error as I was still able to send and view the sent data via MODBUS with interfaced controller during debugging.

alisitsyn, 2023-02-21:
@AshvajitPXy ,

Yes, I understand this but how else I can get information from your system/environment. According to log your issue is related to other code used in your application. In order to confirm this I need the log verbosity set to DEBUG. This will allow to see the messages from Modbus. 

> I also recently found that these were false error as I was still able to send and view the sent data via MODBUS with interfaced controller during debugging.

According to even included log the Modbus was able to send the request correctly then the task was blocked for ~55 sec and after unblocking the modbus completed processing of the previous request. Please check your other code because it is most likely algorithm/active objects priority issue.

dwarnow-xylem, 2023-02-22:
@alisitsyn 
I am a colleague of Ashvajit's and I'd like to add some details.  There is only one task of a higher priority than the Modbus task, and that spends most of it's time in the Blocked state.  Also, during the time that this gap occurs, there are other tasks that are of a lower priority that are still executing (screen display and button navigations never stop working for seconds at a time).  Hope that helps.

AshvajitPXy, 2023-03-03:
Hi ,
I changed the affinity to CPU1 (from CPU0) and I now seem unable to get the problem to occur.  I tried changing to No Affinity (which should be as good or better, allowing it to bounce between cores as needed) and increasing the priority, but I still was seeing the issue.  I do not know why this helps and do you have any recommendation for core affinity and task priority ?

[image]

alisitsyn, 2023-03-16:
Hi @AshvajitPXy, @dwarnow-xylem,
For some reason I did not get updates for this issue. Sorry for the delays.
According to your log the modbus sends the request and then serial task waits for the incoming response from slave during slave response time which is 500ms in your case but the task did not get response during response timeout and the timer triggers the timeout error. From the modbus standpoint everything looks good and it shows the timeout error like it should be when the master did not get respond. There are several possible reasons for this but I think the bluedroid component tasks prevents modbus serial task to work properly. By default Bluetooth tasks are running on core#0  but you can change this in [kconfig options](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-bt-bluedroid-pinned-to-core-choice). The solution is to run the Bluetooth tasks on one core and modbus tasks on other core. This will allow to solve this issue and actually you did it when changed the Modbus affinity option. My recommendations are below:
1. Update esp-modbus component to later version v1.0.10.
2. Change the [bluetooth affinity options](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html#built-in-task-priorities) on other core (opposite with modbus affinity options).
3. Set the affinity options of Modbus to opposite core compare to [Bluetooth tasks affinity](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html#built-in-task-priorities). Try to play with task priorities to get better result. 

> If the most significant bit of a write register is 1, the write fails and none of the registers in that write will update.

The error can be also related to coexistence with incorrectly configured bluetooth tasks.

```
D (4977) MBM_TIMER: Timer mode: (1) triggered
D (4977) MB_PORT_COMMON: eMBMasterPoll:EV_MASTER_ERROR_PROCESS
D (4977) MB_PORT_COMMON: vMBMasterErrorCBRespondTimeout:Callback respond timeout.
D (4977) MB_PORT_COMMON: eMBMasterWaitRequestFinish: returned event = 0x80
E (4997) MB_CONTROLLER_MASTER: mbc_master_send_request(93): Master send request failure error=(0x107) (ESP_ERR_TIMEOUT).
```
Please let me know the results. Thanks.

dwarnow-xylem, 2023-03-17:
Hi @alisitsyn 

Thanks for getting back to us.  As was mentioned, changing the affinity fixed the timeout problem.  The most significant bit problem ended up being a bug in the code for our Modbus slave device, so that was not related.

As for the issue with sharing tasks with the Bluetooth, is this going to be an issue with also using the WIFI?  I see from your links that the are on the same core with the same priority.  Also, is there any additional guidance for what tasks should be on which core?  Despite being built in, the link you shared does not mention that the Modbus task should be opposite of Bluetooth or any other high priority tasks.  Thanks for any additional guidance you can give.

alisitsyn, 2023-03-21:
Hi @dwarnow-xylem ,

Thank you for your feedback. 
> As for the issue with sharing tasks with the Bluetooth, is this going to be an issue with also using the WIFI? I see from your links that the are on the same core with the same priority.

What being said above applies to coexistence with WiFi tasks as well. The `esp-modbus` is now separate component and has its own documentation which mentions some aspects related to configuration options including the tasks affinity [here](https://docs.espressif.com/projects/esp-modbus/en/latest/esp32/applications_and_references.html#id9) . I can not clarify the exact task affinity for configuration there because the active objects configuration can be different in user projects but for sure it can be clarified more clearly and the link for task affinity above can be mentioned in the docs. This is added to the project todo list.

Let me know if you still have issues.

alisitsyn, 2024-03-05:
The issue is fixed in the latest version of [esp-modbus](https://github.com/espressif/esp-modbus/releases/tag/v1.0.11). The issue is closed.

closed 2024-03-05 by alisitsyn