espressif/esp-idf #10810: CMake SET variable accessible only after idf_component_register()

ddomnik, 2023-02-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.0 ((v5.0))

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

 Variables defined in a parent CMakeLists will be accessible in a child CMakeLists using ${VAR_NAME}.

### What is the actual behavior?

The variable value is only accessible after the idf_component_register() command in the child CMakeLists.

### Steps to reproduce.

Using the hello_world example:

Parent CMakeLists (hello_world):
```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

SET(TESTVAR ON)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(hello_world)
```

Child CMakeLists (main):
```
message(STATUS "\n\n\nTESTVAR 1. ${TESTVAR}\n\n\n")
 
idf_component_register(SRCS "hello_world_main.c"
                    INCLUDE_DIRS "")

target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format")

message(STATUS "\n\n\nTESTVAR 2. ${TESTVAR}\n\n\n")
```

### Build or installation Logs.

```plain
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.39.0.windows.1") 
[...]
Constraint file: C:\Espressif\espidf.constraints.v5.0.txt
Requirement files:
 - C:/Espressif/frameworks/esp-idf-v5.0/tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.0_py3.10_env\Scripts\python.exe
Python requirements are satisfied.
-- 

TESTVAR 1. 

-- Project sdkconfig file C:/Users/USER/Documents/workspace-idf/hello_world/sdkconfig
[...]
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- 

TESTVAR 1. ON

-- 

TESTVAR 2. ON

-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tester touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.0/components/app_trace C:/Espressif/frameworks/esp-idf-v5.0/components/app_update C:/Espressif/frameworks/esp-idf-v5.0/components/bootloader C:/Espressif/frameworks/esp-idf-v5.0/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.0/components/bt C:/Espressif/frameworks/esp-idf-v5.0/components/cmock C:/Espressif/frameworks/esp-idf-v5.0/components/console C:/Espressif/frameworks/esp-idf-v5.0/components/cxx C:/Espressif/frameworks/esp-idf-v5.0/components/driver C:/Espressif/frameworks/esp-idf-v5.0/components/efuse C:/Espressif/frameworks/esp-idf-v5.0/components/esp-tls C:/Espressif/frameworks/esp-idf-v5.0/components/esp_adc C:/Espressif/frameworks/esp-idf-v5.0/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.0/components/esp_common C:/Espressif/frameworks/esp-idf-v5.0/components/esp_eth C:/Espressif/frameworks/esp-idf-v5.0/components/esp_event C:/Espressif/frameworks/esp-idf-v5.0/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v5.0/components/esp_hid C:/Espressif/frameworks/esp-idf-v5.0/components/esp_http_client C:/Espressif/frameworks/esp-idf-v5.0/components/esp_http_server C:/Espressif/frameworks/esp-idf-v5.0/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v5.0/components/esp_https_server C:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.0/components/esp_lcd C:/Espressif/frameworks/esp-idf-v5.0/components/esp_local_ctrl C:/Espressif/frameworks/esp-idf-v5.0/components/esp_netif C:/Espressif/frameworks/esp-idf-v5.0/components/esp_partition C:/Espressif/frameworks/esp-idf-v5.0/components/esp_phy C:/Espressif/frameworks/esp-idf-v5.0/components/esp_pm C:/Espressif/frameworks/esp-idf-v5.0/components/esp_psram C:/Espressif/frameworks/esp-idf-v5.0/components/esp_ringbuf C:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.0/components/esp_system C:/Espressif/frameworks/esp-idf-v5.0/components/esp_timer C:/Espressif/frameworks/esp-idf-v5.0/components/esp_wifi C:/Espressif/frameworks/esp-idf-v5.0/components/espcoredump C:/Espressif/frameworks/esp-idf-v5.0/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.0/components/fatfs C:/Espressif/frameworks/esp-idf-v5.0/components/freertos C:/Espressif/frameworks/esp-idf-v5.0/components/hal C:/Espressif/frameworks/esp-idf-v5.0/components/heap C:/Espressif/frameworks/esp-idf-v5.0/components/http_parser C:/Espressif/frameworks/esp-idf-v5.0/components/idf_test C:/Espressif/frameworks/esp-idf-v5.0/components/ieee802154 C:/Espressif/frameworks/esp-idf-v5.0/components/json C:/Espressif/frameworks/esp-idf-v5.0/components/log C:/Espressif/frameworks/esp-idf-v5.0/components/lwip C:/Users/USER/Documents/workspace-idf/hello_world/main C:/Espressif/frameworks/esp-idf-v5.0/components/mbedtls C:/Espressif/frameworks/esp-idf-v5.0/components/mqtt C:/Espressif/frameworks/esp-idf-v5.0/components/newlib C:/Espressif/frameworks/esp-idf-v5.0/components/nvs_flash C:/Espressif/frameworks/esp-idf-v5.0/components/openthread C:/Espressif/frameworks/esp-idf-v5.0/components/partition_table C:/Espressif/frameworks/esp-idf-v5.0/components/perfmon C:/Espressif/frameworks/esp-idf-v5.0/components/protobuf-c C:/Espressif/frameworks/esp-idf-v5.0/components/protocomm C:/Espressif/frameworks/esp-idf-v5.0/components/pthread C:/Espressif/frameworks/esp-idf-v5.0/components/sdmmc C:/Espressif/frameworks/esp-idf-v5.0/components/soc C:/Espressif/frameworks/esp-idf-v5.0/components/spi_flash C:/Espressif/frameworks/esp-idf-v5.0/components/spiffs C:/Espressif/frameworks/esp-idf-v5.0/components/tcp_transport C:/Users/USER/Documents/workspace-idf/hello_world/components/tester C:/Espressif/frameworks/esp-idf-v5.0/components/touch_element C:/Espressif/frameworks/esp-idf-v5.0/components/ulp C:/Espressif/frameworks/esp-idf-v5.0/components/unity C:/Espressif/frameworks/esp-idf-v5.0/components/usb C:/Espressif/frameworks/esp-idf-v5.0/components/vfs C:/Espressif/frameworks/esp-idf-v5.0/components/wear_levelling C:/Espressif/frameworks/esp-idf-v5.0/components/wifi_provisioning C:/Espressif/frameworks/esp-idf-v5.0/components/wpa_supplicant C:/Espressif/frameworks/esp-idf-v5.0/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/USER/Documents/workspace-idf/hello_world/build
cmake --build . -- -v
[1/882] cmd.exe /C "cd /D C:\Users\USER\Documents\workspace-idf\hello_world\build && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -E touch C:/Users/USER/Documents/workspace-idf/hello_world/build/project_elf_src_esp32s3.c"
[2/882] cmd.exe /C "cd /D C:\Users\USER\Documents\workspace-idf\hello_world\build\esp-idf\esp_system && C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -C -P -x c -E -o C:/Users/USER/Documents/workspace-idf/hello_world/build/esp-idf/esp_system/ld/memory.ld -I C:/Users/USER/Documents/workspace-idf/hello_world/build/config -I C:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/ld C:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/ld/esp32s3/memory.ld.in"
```

### More Information.

I want to use a custom variable set via the SET() command to define which components should be included under the RRQUIRES keyword inside of the idf_component_register() function.

E.g.:

```
if(TESTVAR)
idf_component_register( SRCS "${srcs}" 
                        INCLUDE_DIRS "${incs}"
                        REQUIRES compA compB comC )
else()
idf_component_register( SRCS "${srcs}" 
                        INCLUDE_DIRS "${incs}"
                        REQUIRES compA )
endif()
```

or just

```
idf_component_register( SRCS "${srcs}" 
                        INCLUDE_DIRS "${incs}"
                        REQUIRES ${REQ_compA} ${REQ_compB} ${REQ_compC}  )
```

igrr, 2023-02-21:
Hi @ddomnik, indeed this does not work because global CMake variables are not passed to the component dependency expansion step, which is performed in CMake script mode. (The technical reasons for this two-pass component registration process used in IDF now was discussed recently in https://github.com/espressif/esp-idf/issues/9929 and https://github.com/espressif/esp-idf/issues/10579.)

As a workaround, you can use `idf_build_set_property` in the project CMakeLists.txt file to set a custom build property and then `idf_build_get_property` in the component CMakeLists.txt file to obtain it.

E.g., in the project CMakeLists.txt file:

```cmake
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
idf_build_set_property(TEST_VAR "some_value")
project(my_project)
```

In component CMakeLists.txt:
```cmake
idf_build_get_property(test_var TEST_VAR)
message(STATUS "TEST_VAR: ${test_var}")

idf_component_register(... something that uses ${test_var} ...)
```

Another option is to use the global variable, as you did, but instead of adding dependencies in the idf_component_register call, attach them later using `target_link_libraries`:
```cmake
idf_component_register(...)

if(REQ_compA)  # can be set to a component name, e.g. "driver"
  target_link_libraries(${COMPONENT_LIB} PRIVATE idf::${REQ_compA})   # or PUBLIC, if necessary
endif()
```

ddomnik, 2023-02-21:
Thank you very much for the incredible fast response. It worked!

closed 2023-02-21 by ddomnik