espressif/esp-idf #10813: ESP32-S3 ULP RISC-V running speed is not as documentation

ascillato, 2023-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I have a problem with the running speed of the ULP RISC-V on the ESP32-S3 which is 256 times slower than advertised.

According to the documentation, the ULP RISC-V on the ESP32-S3 has a working frequency of 17.5MHz. Also, according to the examples (like https://github.com/espressif/esp-idf/blob/master/examples/system/ulp_riscv/gpio_interrupt/main/ulp/main.c) it is stated that the speed of 17.5 MHz is correct due to the key `ULP_RISCV_CYCLES_PER_US` (which is 17.5):

```cpp
    /* Wakeup interrupt is a level interrupt, wait 1 sec to
       allow user to release button to avoid waking up the ULP multiple times */
    ulp_riscv_delay_cycles(1000*1000 * ULP_RISCV_CYCLES_PER_US);
```

But actually I can see that the real clock speed is 17.5 MHz / 256 = 68.359 Hz. Besides that, we have to take into account that each instruction can takes from 6 to 8 CPU cycles to be completed, so the resulting speed is lower. Just a simple program like the following one shows in an oscilloscope a frequency of 2.85 KHz which confirms this. 68.359Hz CPU frequency, 3 instructions of 8 cylcles each = 68.359 / (3 * 8) = 2.848 Hz.
```cpp
void main (void)
{
    ulp_riscv_gpio_init(GPIO_NUM_15);
    ulp_riscv_gpio_output_enable(GPIO_NUM_15);
    ulp_riscv_gpio_set_output_mode(GPIO_NUM_15);

    while(1) {
        ulp_riscv_gpio_output_level(GPIO_NUM_15, 0);
        ulp_riscv_gpio_output_level(GPIO_NUM_15, 1);
    }
}
```

So far I didn't find any option on the SDK config nor in the latest documentation (SDK 5.0.1) to increase this clock speed. The technical documentation for the ULP RISC-V doesn't say anything about the speed or postscalers. The only part in which is mentioned this postscaler is in the RTC Clock (https://docs.espressif.com/projects/esp-idf/en/v5.0.1/esp32s3/api-reference/kconfig.html#config-rtc-clk-src) but that is always with 256 postscaler.

Is there any way to use the clock for the ULP RISC-V without that 256 postscaler? Can the ULP RISC-V run faster than 68 KHz?

ESP-Marius, 2023-02-22:
As far as I know the clock should be 17.5 MHz, no prescaler is active. If there were then `ulp_riscv_delay_cycles` would be totally wrong as well.

How did you actually measure this, e.g. where do you get 3 instructions from?

My quick test with my logic analyzer give a toggle frequency of 250 KHz. Looking at the assembly there seems to be 10 instructions per loop iteration, so 10 * 7 (cycles per instructions) * 250 KHz = 17.5 Mhz

ascillato, 2023-02-22:
Thank you very much for your answer.

You are right. I made a mistake on my measurements. Sorry. I will close the issue.

The C code I posted seems that it generates a longer Assembly code than yours so it outputs 122KHz. Obviously could be improved to reach yours 250KHz.

So, the ULP RISC-V of the ESP32-S3 runs at 17.5MHz but as it uses around 7 cycles per assembly instruction (depending on the type of the instruction), it actually runs at 2.5 Mhz. Also, we have to take into account that going from C to assembly uses several instructions. I will need to optimize my application code. I will write it in assembly directly.

Do you know where is the assembly file generated from the C code? Or I have to disassemble the risc-v bin for checking that? Thanks.

closed 2023-02-22 by ascillato

ESP-Marius, 2023-02-23:
I disassembled the elf (you can find it under `build/esp-idf/main/ULP_PROJECT_NAME`)

Currently the binary is always built optimized for size (see `components/ulp/cmake/toolchain-ulp-riscv.cmake` flags) as we assumed the limited size would be the main problem for most users.

You can try to patch this to the optimization flag best suited for your ULP-application, and if you find that this end up making a big difference for your use-case then we can consider making the optimization level configurable from menuconfig.

ascillato, 2023-02-23:
@ESP-Marius 

Thank you so much for the information. Yes, changing the cmake flag of `components/ulp/cmake/toolchain-ulp-riscv.cmake` from `-Os` to `-O3` have duplicated the performance on my code!!! I didn't expect that huge change. If that option can be added to menuconfig would be awesome since it has shown to be really helpful.

Thank you for you help. Very appreciated!!!!

zhongpang, 2023-02-28:
@ESP-Marius @ascillato 

Dear both,

Greeting! I'm facing a performance issue similar as ascillato's report, and I tried to update the file: <esp-idf install dir>/components/ulp/cmake/toolchain-ulp-riscv.cmake from -Os to -O3.
But when I issued "idf.py build" in my project folder, I saw the compiler flag still be -Os, instead of -O3 as I expected.
Did I miss any approaches to make the change take effect? Thanks very much! 

> @ESP-Marius
> 
> Thank you so much for the information. Yes, changing the cmake flag of `components/ulp/cmake/toolchain-ulp-riscv.cmake` from `-Os` to `-O3` have duplicated the performance on my code!!! I didn't expect that huge change. This is great.
> 
> If the compiler optimization option for ULP-RISC-V can be added to menuconfig would be awesome since it has shown to be really helpful.
> 
> Thank you for your help. Very appreciated!!!!

> I disassembled the elf (you can find it under `build/esp-idf/main/ULP_PROJECT_NAME`)
>
> Currently the binary is always built optimized for size (see `components/ulp/cmake/toolchain-ulp-riscv.cmake` [...]

ESP-Marius, 2023-03-01:
@zhongpang I think it should work, at least from my testing it seems OK.

Just make sure you are clearing your build files before building to make sure nothing is stuck from before your change.

zhongpang, 2023-03-01:
@ESP-Marius 

Yes, you're correct! after remove the "build" folder, the change of toolchain-ulp-riscv.cmake took effect. I did  "idf.py clean" in previous tries, but seem it hasn't really "clean"ed the compiler caches. Thanks!
Unfortunately, change the optimization level still not helped on my case, because all parameters except -Os failed with same reason:  "of ulp_main section `.bss' is not within region `ram' ". With -Os, my code generated a binary file with 7888 bytes, which already closed to maximum RTC slow memory size, so I was not surprised.
But I'm eager to know is there any otherway to enhance the GPIO operation speed in risc-v ulp?
My scenario is to bigbang two SPI devices in the ulp, one is a ADC chip with 125 samples per second, another is Nand Flash, the slow speed of GPIO operation caused the write of flash couldn't complete in an ADC samples break, which makes some loses of ADC data ..., so I think/hope there's way to improve GPIO operation speed in riscv ulp? Thanks very much!

Jeffrey Zhong

ESP-Marius, 2023-03-02:
@zhongpang you can take a look at how the `ulp_riscv_gpio_output_level` function is implemented and do the register writes directly to avoid any uncessary function calls/instructions, look at the disassembled binary to spot any places were the code is inefficient etc.

But in the end it could also be that the ULP is just not fast enough to what you want to do. Maybe another option would be to use an i2c sensor instead and use the i2c hardware which I imagine would be faster.

Or reconsider if the flash access is needed, e.g. could you store the variables in memory instead, and let the main cpu occasionally do a flash write?

zhongpang, 2023-03-02:
@ESP-Marius 

Thanks for the timely reply!
Further question: Do you have any workable example to replace ulp_riscv_gpio_output_level() and ulp_riscv_gpio_get_level()  with register operation? :)

For other two suggestions:
For sensors, we can not use other sensors since it's a designed product. the only interface the sensor provides is SPI.
It generated 750 bytes of data per second, we actually use a 2K buffer to cache the data, and in normal model, main CPU read the buffer and write to flash 4K/time, which means the Flash in standby model in most of time.
To lower the power consuming, we're trying to add an ULP-Only mode, to do the data collection and flash write in ULP and let the main CPU in deep sleep. We actually already made it worked, but because of the low GPIO operation speed, plus the behavior of Nand Flash, we have to keep the Nand Flash active all the time, which cost 1.x ma of current almost all the time. It makes the current in the "ULP Only" model even higher than "Main CPU lightsleep and write Flash, only sensor in ULP".
So if we can enhance the io operate speed in ULP, we can significantly reduce the power consuming in "ULP only" model.

Thanks :)

ESP-Marius, 2023-03-02:
You can just see how it is done and copy from `components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h`, 

e.g. set output high would be : `REG_SET_FIELD(RTC_GPIO_OUT_W1TS_REG, RTC_GPIO_OUT_DATA_W1TS, BIT(gpio_num))`

zhongpang, 2023-03-02:
@ESP-Marius I have tried to use the macros directly, but the behavior was not as expected. I suspect the binary size would decrease, but it increased by just replacing the two functions. Any hints? Thank very much!

ESP-Marius, 2023-03-03:
> I have tried to use the macros directly, but the behavior was not as expected. I suspect the binary size would decrease, but it increased by just replacing the two functions. Any hints? Thank very much!

Yes, if a call to a function (shared code) was replaced by separate code wherever it is called then it will increase the binary size. In general it is "always" a trade off between speed and size.

zhongpang, 2023-03-03:
@ESP-Marius 
Thanks for the explanation! Seems the way above can not solve my problem, since my available RTC memory just 200 bytes now :(
It might be expect too much ... I still want find way to improve the gpio performance without binary size increase. Is it possible to use embeded ASM to reach this goal? If worth a try, any examples to embed ASM GPIO operation in ULP C code?
Thanks very much!