espressif/esp-idf #10814: libbtdm_app.a ASSERT_WARN(1 6), in lc_task.c at line 4658 when HCI mode change event

darthcloud, 2023-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3462-g045163a2ec

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC-32E

### Power Supply used.

USB

### What is the expected behavior?

No ASSERT in function  lc_task of the closed libbtdm_app.a BR/EDR library after receiving an HCI mode change event with Status: Command Disallowed (0x0c).

### What is the actual behavior?

ESP32 running as the host for BR/EDR HID devices sometimes throw the following assert:
ASSERT_WARN(1 6), in lc_task.c at line 4658

This happen after setting sniff mode with some BR/EDR HID devices.

This occur when mode change event HCI packet is received with status:
    Status: Command Disallowed (0x0c) (See wireshark trace packet # 481)

After this assert occur the HID device will stop sending HID report and their is no way to recover the connection.
The device stay connected but no data is received.

### Steps to reproduce.

1. Build and flash BlueRetro on ESP32-DevKitC-32E
```
git clone -b SW_REWORK2 https://github.com/darthcloud/BlueRetro.git`
cd BlueRetro
cp configs/hw1/gamecube sdkconfig
idf.py build
idf.py flash
```
2. Connect/Disconnect a BR/EDR HID device (like a Switch controller) until ASSERT appear.

### More Information.

Complete log including H4 HCI packet dump:
[switch_assert_after_sniff_mode_latest_idf.txt](https://github.com/espressif/esp-idf/files/10799167/switch_assert_after_sniff_mode_latest_idf.txt)

Log with packet dump removed:
[switch_assert_after_sniff_mode_latest_idf_nodump.txt](https://github.com/espressif/esp-idf/files/10799169/switch_assert_after_sniff_mode_latest_idf_nodump.txt)

Wireshark HCI H4 trace on ESP32 side:
[blueretro_switch_assert_after_sniff_mode_lastest_idf.zip](https://github.com/espressif/esp-idf/files/10799175/blueretro_switch_assert_after_sniff_mode_lastest_idf.zip)

Packets # 185 & # 188 are example were the sniff mode worked.
Packets # 479 & # 481 are example of disallowed sniff mode that trigger the assert.

BetterJincheng, 2023-03-22:
@darthcloud How to connect/disconnect to/from a device within this project?

darthcloud, 2023-03-22:
@BetterJincheng it is a host device and it goes into inquiry mode automatically on boot, so you connect game controller to it.

This list how to connect the various controller supported, this issue is easier to reproduce with Nintendo Switch controllers:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#6---switch-pairing-guide

BetterJincheng, 2023-03-23:
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/11046556/libbtdm_app.zip)
@darthcloud  Please reproduce with this bt-lib and attach the whole log. Thanks!

darthcloud, 2023-03-23:
Log with debug bt-lib and assert after sniff mode:
[switch_assert_after_sniff_mode_debug_bt.txt](https://github.com/espressif/esp-idf/files/11048315/switch_assert_after_sniff_mode_debug_bt.txt)

Log with debug bt-lib and sniff mode worked:
[switch_ok_after_sniff_mode_debug_bt.txt](https://github.com/espressif/esp-idf/files/11048328/switch_ok_after_sniff_mode_debug_bt.txt)

BetterJincheng, 2023-03-29:
@darthcloud Please test this bt-lib. Thanks!
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/11097792/libbtdm_app.zip)

darthcloud, 2023-03-29:
@BetterJincheng this look to have fixed it! :)

I reverted to last week lib and I reproduced easily and then upgraded back to this one and I could not reproduce the issue even after many attempts.

Here a couple log done with the new test lib for which the sniff mode was successful:
[switch_ok_after_sniff_mode_debug_bt_2ndlib.txt](https://github.com/espressif/esp-idf/files/11100560/switch_ok_after_sniff_mode_debug_bt_2ndlib.txt)
[switch_ok_after_sniff_mode_debug_bt_2ndlib2.txt](https://github.com/espressif/esp-idf/files/11100563/switch_ok_after_sniff_mode_debug_bt_2ndlib2.txt)

I will make more test tonight and report back here.

Thanks!
Jacques

darthcloud, 2023-03-29:
I made more tests and it always work now with this new bt lib, mode change is always successful.

Thanks a lot @BetterJincheng !

BetterJincheng, 2023-03-30:
@darthcloud We will merge it to release branch soon. Turn to us anytime!

AxelLin, 2023-05-05:
> @darthcloud We will merge it to release branch soon. Turn to us anytime!

@BetterJincheng How is the status for this fix?

BetterJincheng, 2023-09-18:
@AxelLin The fix has beed merged in all the release branches except for v4.4.

Alvin1Zhang, 2023-12-13:
github fix commit: 3fb242660d4b441f69f11aa396e5b9955a2e9e8e

closed 2023-12-13 by Alvin1Zhang