espressif/esp-idf #10815: It doesn't work when if enable hardware flow ctrl in openthread uart config

gl-chenxiaosheng, 2023-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF version：commit 886e98a2c1311556eb6be02775d49703d6050222 (HEAD -> release/v5.0, origin/release/v5.0)

I use esp32s3 as ot_br, and it radio config as flow:
```c
#define ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG()
{								
	.radio_mode = RADIO_MODE_UART_RCP,						
	.radio_uart_config = {									        
	.port = 1,												
	.uart_config =											
	{														
		.baud_rate = 115200,								
		.data_bits = UART_DATA_8_BITS,						
		.parity = UART_PARITY_DISABLE,						
		.stop_bits = UART_STOP_BITS_1,						
		.flow_ctrl = UART_HW_FLOWCTRL_CTS_RTS,				
		.rx_flow_ctrl_thresh = 0,							
		.source_clk = UART_SCLK_DEFAULT,					
	},														
	.rx_pin = 18,											
	.tx_pin = 17											
	},														
}
```

I use esp32h2 as ot_rcp, and it host config as flow:
```c
#define ESP_OPENTHREAD_DEFAULT_HOST_CONFIG()                    
{                                                           
	.host_connection_mode = HOST_CONNECTION_MODE_RCP_UART,  
	.host_uart_config = {                                   
	.port = 1,                                          
	.uart_config =                                      
	{                                               
		.baud_rate =  115200,                       
		.data_bits = UART_DATA_8_BITS,              
		.parity = UART_PARITY_DISABLE,              
		.stop_bits = UART_STOP_BITS_1,              
		.flow_ctrl = UART_HW_FLOWCTRL_CTS_RTS,      
		.rx_flow_ctrl_thresh = 0,                   
		.source_clk = UART_SCLK_DEFAULT,            
	},                                              
	.rx_pin = 5,                       
	.tx_pin = 4,                       
	},                                                      
}
````

Since the uart1 pin of H2 is any gpio, 1 and 2 are selected as cts/rts.
The two devices are connected as follows:
S3&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;H2
18(rx)&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;4(tx) 
17(tx)&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;5(rx)
19(rts)&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;1(cts)
20(cts)&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;2(rts)
gnd&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;&emsp;gnd

But it doesn't work,
```shell
W(2524) OPENTHREAD:[W] Platform------: Wait for response timeout
W(2524) OPENTHREAD:[W] Platform------: RCP failure detected
W(2524) OPENTHREAD:[W] Platform------: Trying to recover (1/1)
W(4544) OPENTHREAD:[W] Platform------: Wait for response timeout
W(4544) OPENTHREAD:[W] Platform------: RCP failure detected
```

And i switch UART_HW_FLOWCTRL_CTS_RTS to  UART_HW_FLOWCTRL_DISABLE it will work.
Does it not support flow control?

zwx1995esp, 2023-02-24:
请问一下，如您所述:
```
Since the uart1 pin of H2 is any gpio, 1 and 2 are selected as cts/rts.
The two devices are connected as follows:
S3       H2
18(rx)      4(tx)
17(tx)      5(rx)
19(rts)      1(cts)
20(cts)      2(rts)
gnd       gnd
```
这个配置您是改了openthread配置的代码嘛？
因为目前openthread port所实现的uart接口中没有对流控引脚进行支持
参考以下代码:
https://github.com/espressif/esp-idf/blob/master/components/openthread/port/esp_openthread_uart.c#L62-L64

请问您有需求: BR需要增加UART流控的支持嘛？

gl-chenxiaosheng, 2023-02-28:
Yes, support uart flow control mode will be better. 
Considering that in the subsequent scenarios, if there is a large amount of data or the data is sent/received quickly, it can be used to prevent data loss.

gl-chenxiaosheng, 2023-02-28:
Yes, It is necessary to support uart flow control mode. </br>
While the IEEE 802.15.4 2.4 GHz O-QPSK PHY supports 250 kbps, the actual data rate in practice is typically lower due to various overheads (e.g. MAC backoff, interframe spacing, RX-TX turnaround times, data-acknowledgment transmissions, etc). That said, increasing serial baudrate can help reduce communication latency between host and RCP processors. </br>
So when I improve the serial baudrate like 460800, uart flow control is necessary.

zwx1995esp, 2023-02-28:
So, you can try with the serial baudrate like 460800 without flow control(we have already done a lot of test on it) to see if it can meet your requirements.

closed 2025-03-14 by Alvin1Zhang