espressif/esp-idf #10817: Crash after Nimble initialisation on ESP-IDFv.5.0.1, works on v.5.0

jacek12345, 2023-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

Nimble initialisation

### What is the actual behavior?

Crash

### Steps to reproduce.

Used with RainMaker based project.
1. after start, when `Nimble_init();` it crashes.

### Debug Logs.

```plain
I (19450) app_main: MQTT Published. Msg id: 49289.
I (19450) app_main: MQTT Published. Msg id: 27609.
I (22300) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
I (22520) app_main: MQTT Published. Msg id: 2855.
I (34250) esp_rmaker_time: SNTP Synchronised.
I (34250) esp_rmaker_time: The current time is: Wed Feb 22 07:12:01 2023 +0100[CET], DST: No.
btdm: bss start 0x3fcef180, len 36
btdm: data start 0x3fcef174, data start rom 0x40057350, len 12
MAGIC fadebead VERSION 00010006
I (34710) BT_INIT: BT controller compile version [80abacd]
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4038525f  PS      : 0x00060833  A0      : 0x80383f53  A1      : 0x3fcf3840
0x4038525f: uxListRemove at F:/GitHub/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/list.c:197

A2      : 0x0000858c  A3      : 0x3fca0688  A4      : 0x00060820  A5      : 0x00060823
A6      : 0x00000001  A7      : 0x0000cdcd  A8      : 0x8038543c  A9      : 0x3fcf3810
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x01260267
A14     : 0x00000200  A15     : 0x00260000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000859c  LBEG    : 0x4037ed44  LEND    : 0x4037ed50  LCOUNT  : 0x00000000
0x4037ed44: ram_wifi_tx_dig_gain at ??:?

0x4037ed50: ram_wifi_tx_dig_gain at ??:?

Backtrace: 0x4038525c:0x3fcf3840 0x40383f50:0x3fcf3860 0x4202df39:0x3fcf3880 0x4203de0e:0x3fcf38a0 0x4203edf9:0x3fcf38c0 0x4202e633:0x3fcf3910 0x420374a2:0x3fcf3950 0x4200e438:0x3fcf39b0 0x4200ef14:0x3fcf39d0 0x420efe97:0x3fcf3b20 0x403852b6:0x3fcf3b50
0x4038525c: uxListRemove at F:/GitHub/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/list.c:194

0x40383f50: vTaskDelete at F:/GitHub/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/tasks.c:1348 (discriminator 4)

0x4202df39: task_delete_wrapper at F:/GitHub/esp-idf-v5.0.1/components/bt/controller/esp32s3/bt.c:646

0x4203de0e: ble_sm_alg_gen_key_pair at ??:?

0x4203edf9: btdm_controller_init at ??:?

0x4202e633: esp_bt_controller_init at F:/GitHub/esp-idf-v5.0.1/components/bt/controller/esp32s3/bt.c:1312

0x420374a2: nimble_port_init at F:/GitHub/esp-idf-v5.0.1/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:145

0x4200e438: Nimble_init at F:/GitHub/esp-rainmaker/examples/j_switch/main/app_main.c:3179

0x4200ef14: app_main at F:/GitHub/esp-rainmaker/examples/j_switch/main/app_main.c:2010

0x420efe97: main_task at F:/GitHub/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x403852b6: vPortTaskWrapper at F:/GitHub/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

### More Information.

It seems that BT controller version is the problem?

When crashing (ESP-IDFv.5.0.1):
I (34710) BT_INIT: BT controller compile version [**80abacd**] 

When works (ESP-IDFv.5.0):
I (25431) BT_INIT: BT controller compile version [**76c24c9**]

IshaESP, 2023-02-28:
Hi @jacek12345,
I tried running switch example using ESP-IDFv5.0.1 (BT controller compile version [80abacd]), no crash was observed.
Can you send the sdkconfig of the application and the output with debug logs?

rahult-github, 2023-04-25:
Hi @jacek12345 , any updates ?

jacek12345, 2023-06-14:
Now tried with ESP-IDF-v.5.1-rc1 and still have a problem. 
The last version when problem didn't occure is v.5.0
`btdm: bss start 0x3fcef180, len 36
btdm: data start 0x3fcef174, data start rom 0x40057350, len 12
MAGIC fadebead VERSION 00010006
I (22326) BLE_INIT: BT controller compile version [85b425c]
MAGIC fadebead VERSION 00010001
HLI Magic mismatch: 3fc9c9e0, fadebead
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40384abb  PS      : 0x00060e33  A0      : 0x80384495  A1      : 0x3fcaa4f0
0x40384abb: block_size at F:/GitHub/esp-idf-v5.1-rc1/components/heap/tlsf/tlsf_block_functions.h:20
 (inlined by) block_locate_free at F:/GitHub/esp-idf-v5.1-rc1/components/heap/tlsf/tlsf.c:566
 (inlined by) tlsf_malloc at F:/GitHub/esp-idf-v5.1-rc1/components/heap/tlsf/tlsf.c:1004

A2      : 0x3fcef96c  A3      : 0x000000dc  A4      : 0x3fce02e5  A5      : 0x00000000
A6      : 0x3fccb1d0  A7      : 0x3fcca254  A8      : 0xfd42a0b3  A9      : 0x00000008
A10     : 0xffffffc0  A11     : 0x00000003  A12     : 0x00000000  A13     : 0x00060e23
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c
EXCVADDR: 0xfd42a0b7  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x40384ab8:0x3fcaa4f0 0x40384492:0x3fcaa510 0x403762dd:0x3fcaa530 0x40376335:0x3fcaa550 0x4202815c:0x3fcaa570 0x4203bbbe:0x3fcaa590 0x42037acf:0x3fcaa5b0 0x420287d7:0x3fcaa600 0x42030f62:0x3fcaa640 0x4200d36f:0x3fcaa6a0 0x4200ddb8:0x3fcaa6c0 0x420f0c62:0x3fcaa7f0 0x40381d32:0x3fcaa820
0x40384ab8: block_locate_free at F:/GitHub/esp-idf-v5.1-rc1/components/heap/tlsf/tlsf.c:564
 (inlined by) tlsf_malloc at F:/GitHub/esp-idf-v5.1-rc1/components/heap/tlsf/tlsf.c:1004

0x40384492: multi_heap_malloc_impl at F:/GitHub/esp-idf-v5.1-rc1/components/heap/multi_heap.c:207

0x403762dd: heap_caps_malloc_base at F:/GitHub/esp-idf-v5.1-rc1/components/heap/heap_caps.c:176

0x40376335: heap_caps_malloc at F:/GitHub/esp-idf-v5.1-rc1/components/heap/heap_caps.c:197

0x4202815c: malloc_internal_wrapper at F:/GitHub/esp-idf-v5.1-rc1/components/bt/controller/esp32c3/bt.c:649

0x4203bbbe: r_llm_env_init at ??:?

0x42037acf: btdm_controller_init at ??:?

0x420287d7: esp_bt_controller_init at F:/GitHub/esp-idf-v5.1-rc1/components/bt/controller/esp32c3/bt.c:1305

0x42030f62: nimble_port_init at F:/GitHub/esp-idf-v5.1-rc1/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:153

0x4200d36f: Nimble_init at F:/GitHub/esp-rainmaker/examples/j_switch/main/app_main.c:3300

0x4200ddb8: app_main at F:/GitHub/esp-rainmaker/examples/j_switch/main/app_main.c:2122

0x420f0c62: main_task at F:/GitHub/esp-idf-v5.1-rc1/components/freertos/app_startup.c:208 (discriminator 13)

0x40381d32: vPortTaskWrapper at F:/GitHub/esp-idf-v5.1-rc1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
`

IshaESP, 2023-06-15:
Hi @jacek12345 ,
Can you please send us the debug logs and sdkconfig for us to debug it properly. 

To enable debug logs:
1. idf.py menuconfig
2. Component config → Bluetooth → NimBLE Options → NimBLE Host log verbosity → Debug logs
3. Component config → Log output → Default log verbosity → Debug

jacek12345, 2023-06-15:
I'm trying to run with enabled debug logs but have:
***ERROR*** A stack overflow in task sys_evt has been detected.

AxelLin, 2023-06-15:
> I'm trying to run with enabled debug logs but have: _**ERROR**_ A stack overflow in task sys_evt has been detected.

Increase your CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE setting.

IshaESP, 2023-06-15:
Hey,
Try increasing stack size by this:
Component config -> ESP System Settings->ESP_SYSTEM_EVENT_TASK_STACK_SIZE to 4000 for testing purposes

jacek12345, 2023-06-15:
Thank You @AxelLin and @IshaESP 
Now i see:

```
btdm: bss start 0x3fcef180, len 36
btdm: data start 0x3fcef174, data start rom 0x40057350, len 12
MAGIC fadebead VERSION 00010006
I (27283) BLE_INIT: BT controller compile version [85b425c]

assert failed: heap_caps_free heap_caps.c:387 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace: 0x40375cba:0x3fcad000 0x40381a75:0x3fcad020 0x40388b55:0x3fcad040 0x403766db:0x3fcad160 0x40384dc5:0x3fcad180 0x403825e3:0x3fcad1a0 0x42029dc5:0x3fcad1c0 0x420392d1:0x3fcad1e0 0x4203a0b4:0x3fcad200 0x4202a63f:0x3fcad250 0x42032e5e:0x3fcad290 0x4200dc74:0x3fcad2f0 0x4200e700:0x3fcad310 0x420f69d7:0x3fcad440 0x4038450e:0x3fcad470
0x40375cba: panic_abort at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/esp_system/panic.c:452

0x40381a75: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/esp_system/port/esp_system_chip.c:84

0x40388b55: __assert_func at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/newlib/assert.c:81

0x403766db: heap_caps_free at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/heap/heap_caps.c:387 (discriminator 1)

0x40384dc5: vPortFree at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/freertos/heap_idf.c:91

0x403825e3: vQueueDelete at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/freertos/FreeRTOS-Kernel/queue.c:2321

0x42029dc5: semphr_delete_wrapper at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/bt/controller/esp32c3/bt.c:529

0x420392d1: ble_sm_alg_gen_key_pair at ??:?

0x4203a0b4: btdm_controller_init at ??:?

0x4202a63f: esp_bt_controller_init at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/bt/controller/esp32c3/bt.c:1305

0x42032e5e: nimble_port_init at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:153

0x4200dc74: Nimble_init at F:/GitHub/esp-rainmaker/examples/j_switch/main/app_main.c:3301

0x4200e700: app_main at F:/GitHub/esp-rainmaker/examples/j_switch/main/app_main.c:2122

0x420f69d7: main_task at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/freertos/app_startup.c:208 (discriminator 13)

0x4038450e: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

Is it enough or You meed whole log? I'm aware that there can be some secret information. Eventually can send priv?

jacek12345, 2023-06-15:
@IshaESP This is not switch example. This is only based on switch example project.
After RainMaker initialisation I'm waiting 20 sec. and starting NimBle. I'm using NimBle for communication with sensors.
Crasch occures on nimble_port_init();

```

........
err = app_wifi_start(POP_TYPE_NONE);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Could not start Wifi. Aborting!!!");
        vTaskDelay(5000/portTICK_PERIOD_MS);
        abort();
    }
vTaskDelay(2000);
nimble_port_init();
 //Initialize the NimBLE host configuration.
 ble_hs_cfg.reset_cb = bleprph_on_reset;
 ble_hs_cfg.sync_cb = bleprph_on_sync;
 nimble_port_freertos_init(bleprph_host_task);
```

jacek12345, 2023-06-15:
I made one more time install.bat and export.bat in v5.1-rc1, and here is log

```
I (27) boot: ESP-IDF v5.1-rc1 2nd stage bootloader
I (27) boot: compile time Jun 15 2023 18:24:27
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (72) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (79) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (87) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (94) boot:  4 ota_0            OTA app          00 10 00020000 003c0000
I (102) boot:  5 ota_1            OTA app          00 11 003e0000 003c0000
I (109) boot:  6 fctry            WiFi data        01 02 007a0000 00006000
I (117) boot:  7 devdata          WiFi data        01 02 007a6000 00003000
I (124) boot:  8 coredump         Unknown data     01 03 007a9000 00010000
I (132) boot: End of partition table
I (136) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=55af8h (350968) map
I (208) esp_image: segment 1: paddr=00075b20 vaddr=3fc9b000 size=04e9ch ( 20124) load
I (212) esp_image: segment 2: paddr=0007a9c4 vaddr=40374000 size=05654h ( 22100) load
I (219) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=f5704h (1005316) map
I (402) esp_image: segment 4: paddr=0017572c vaddr=40379654 size=118e8h ( 71912) load
I (428) boot: Loaded app from partition at offset 0x20000
I (428) boot: Disabling RNG early entropy source...
I (440) cpu_start: Multicore app
D (440) flash HPM: HPM with dummy, status is 3
I (440) cpu_start: Pro cpu up.
I (441) cpu_start: Starting app cpu, entry point is 0x40375678
0x40375678: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
D (456) clk: RTC_SLOW_CLK calibration value: 3806566
I (468) cpu_start: Pro cpu start user code
I (468) cpu_start: cpu freq: 160000000 Hz
I (468) cpu_start: Application information:
I (471) cpu_start: Project name:     neox1
I (476) cpu_start: App version:      1.9
I (480) cpu_start: Compile time:     Jun 15 2023 18:23:48
I (486) cpu_start: ELF file SHA256:  1588ec57254788a4...
I (492) cpu_start: ESP-IDF:          v5.1-rc1
I (497) cpu_start: Min chip rev:     v0.0
I (502) cpu_start: Max chip rev:     v0.99
I (507) cpu_start: Chip rev:         v0.1
D (512) memory_layout: Checking 5 reserved memory ranges:
D (517) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc9b000
D (524) memory_layout: Reserved memory range 0x3fc9b000 - 0x3fca8230
D (530) memory_layout: Reserved memory range 0x3fceee34 - 0x3fcf0000
D (536) memory_layout: Reserved memory range 0x40374000 - 0x4038b000
0x40374000: _WindowOverflow4 at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:2013

D (543) memory_layout: Reserved memory range 0x600fe000 - 0x600fe010
D (549) memory_layout: Building list of available memory regions:
D (555) memory_layout: Available memory region 0x3fca8230 - 0x3fcb0000
D (562) memory_layout: Available memory region 0x3fcb0000 - 0x3fcc0000
D (569) memory_layout: Available memory region 0x3fcc0000 - 0x3fcd0000
D (575) memory_layout: Available memory region 0x3fcd0000 - 0x3fce0000
D (582) memory_layout: Available memory region 0x3fce0000 - 0x3fce9710
D (588) memory_layout: Available memory region 0x3fce9710 - 0x3fceee34
D (595) memory_layout: Available memory region 0x3fcf0000 - 0x3fcf8000
D (602) memory_layout: Available memory region 0x600fe010 - 0x60100000
I (608) heap_init: Initializing. RAM available for dynamic allocation:
D (616) heap_init: New heap initialised at 0x3fca8230
I (621) heap_init: At 3FCA8230 len 000414E0 (261 KiB): DRAM
I (627) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
[... 941 lines cut ...]
0x4037e4d0: ram_wifi_tx_dig_gain at ??:?

0x4037e4dc: ram_wifi_tx_dig_gain at ??:?

Backtrace: 0x420299ce:0x3fcabda0 0x42038ee5:0x3fcabdc0 0x42039cc4:0x3fcabde0 0x4202a253:0x3fcabe30 0x42032a72:0x3fcabe70 0x4200dbe0:0x3fcabed0 0x4200e65f:0x3fcabef0 0x420f4dc3:0x3fcac020 0x403831ae:0x3fcac050
0x420299ce: semphr_delete_wrapper at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/bt/controller/esp32c3/bt.c:521

0x42038ee5: ble_sm_alg_gen_key_pair at ??:?

0x42039cc4: btdm_controller_init at ??:?

0x4202a253: esp_bt_controller_init at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/bt/controller/esp32c3/bt.c:1305

0x42032a72: nimble_port_init at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:153

0x4200dbe0: Nimble_init at F:/GitHub/esp-rainmaker/examples/j_switch/main/app_main.c:3301

0x4200e65f: app_main at F:/GitHub/esp-rainmaker/examples/j_switch/main/app_main.c:2124

0x420f4dc3: main_task at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/freertos/app_startup.c:208 (discriminator 13)

0x403831ae: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.1-rc1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```
Attached also sdkconfig
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/11760634/sdkconfig.zip)

jacek12345, 2023-06-21:
Do You have any idea? Or any example with reusing NimBle after provisioning?

IshaESP, 2023-06-22:
Hey @jacek12345 ,
We tried on both the versions 5.0 and 5.1, and the IDF and rainmaker examples work cleanly.

Looking at your log, we see that you have somehow by passed the internal nimble initialization which happens via the wifi_provisioning -- protocomm layer endpoint. 

Since there appears to be code changes, is it possible for you to suggest any way to use IDF examples to reproduce the issue you have mentioned ?  Any minimalistic change we can do to simulate the same scenario as seen at your end.   

We would have performed git bisect to find failing commit, but we can do it once we are able to reproduce the issue at our end

jacek12345, 2023-07-03:
I just updated ESP-IDF to 5.1 release. Still have a problem.
I reproduced this issue on Rainmaker example 'switch' with little code addition.
Here is what i added:
1. Added 'bleprph.h' to main folder in 'switch' example
2. Added two blocks of code in app_main.c file:

On the beginning:
```
#include "app_priv.h"

//========= My code added =================
	    #include "esp_nimble_hci.h"
    #include "nimble/nimble_port.h"
    #include "nimble/nimble_port_freertos.h"
    #include "host/ble_hs.h"
    #include "host/util/util.h"
    #include "console/console.h"
    #include "services/gap/ble_svc_gap.h"
    #include "bleprph.h"
	

static void bleprph_on_reset(int reason)
{
    ESP_LOGE("app_main", "Resetting state; reason=%d", reason);
}

/**
 * The nimble host executes this callback when a GAP event occurs.  The
 * application associates a GAP event callback with each connection that is
 * established.  blecent uses the same callback for all connections.
 *
 * @param event                 The event being signalled.
 * @param arg                   Application-specified argument; unused by
 *                                  blecent.
 *
 * @return                      0 if the application successfully handled the
 *                                  event; nonzero on failure.  The semantics
 *                                  of the return code is specific to the
 *                                  particular GAP event being signalled.
 * 
 * 
 */
static int blecent_gap_event(struct ble_gap_event *event, void *arg)
{
        
    switch (event->type) {
    case BLE_GAP_EVENT_DISC:
		
	ESP_LOGI("app_main", "Event detected");
        
        return 0;

    case BLE_GAP_EVENT_DISC_COMPLETE:
        MODLOG_DFLT(INFO, "discovery complete; reason=%d\n",
                    event->disc_complete.reason);
        return 0;

    case BLE_GAP_EVENT_MTU:
        MODLOG_DFLT(INFO, "mtu update event; conn_handle=%d cid=%d mtu=%d\n",
                    event->mtu.conn_handle,
                    event->mtu.channel_id,
                    event->mtu.value);
        return 0;

    case BLE_GAP_EVENT_REPEAT_PAIRING:
        return 0;
    default:
        return 0;
    }
}

/**
 * Initiates the GAP general discovery procedure.
 */
static void blecent_scan(void)
{
    uint8_t own_addr_type;
    struct ble_gap_disc_params disc_params;
    int rc;

    /* Figure out address to use while advertising (no privacy for now) */
    //rc = ble_hs_id_infer_auto(0, &own_addr_type);
	rc = ble_hs_id_infer_auto(1, &own_addr_type);
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "error determining address type; rc=%d\n", rc);
        return;
    }

    /* Tell the controller to filter duplicates; we don't want to process
     * repeated advertisements from the same device.
     */
    disc_params.filter_duplicates = 0;//1;

    /**
     * Perform a passive scan.  I.e., don't send follow-up scan requests to
     * each advertiser.
     */
    disc_params.passive = 1;

    disc_params.itvl = 800;
    disc_params.window = 480;
    disc_params.filter_policy = 0;
    disc_params.limited = 0;

    rc = ble_gap_disc(own_addr_type, BLE_HS_FOREVER, &disc_params, blecent_gap_event, NULL);
    
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "Error initiating GAP discovery procedure; rc=%d\n",
                    rc);
    }
}

static void bleprph_on_sync(void)
{
    blecent_scan();
}

void bleprph_host_task(void *param)
{
    ESP_LOGI("app_main", "BLE Host Task Started");
    /* This function will return only when nimble_port_stop() is executed */
    nimble_port_run();

    nimble_port_freertos_deinit();
}
//END My code added =============

static const char *TAG = "app_main";
esp_rmaker_device_t *switch_device;
```

On the end:
```
   err = app_wifi_start(POP_TYPE_RANDOM);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Could not start Wifi. Aborting!!!");
        vTaskDelay(5000/portTICK_PERIOD_MS);
        abort();
    }
	
	// =========== My code added =================
    vTaskDelay(2000);
    esp_err_t ret;

	ret = nimble_port_init();
	if (ret != ESP_OK)
	{
		ESP_LOGE(TAG, "Failed to init nimble %d ", ret);
		return;
	}
	
   
	//Initialize the NimBLE host configuration.
	ble_hs_cfg.reset_cb = bleprph_on_reset;
	ble_hs_cfg.sync_cb = bleprph_on_sync;

	nimble_port_freertos_init(bleprph_host_task);
//END My code added ==============	
```

Zipped 'main' folder from Rainmaker 'switch' example.
[main.zip](https://github.com/espressif/esp-idf/files/11935255/main.zip)

IshaESP, 2023-07-04:
Hi @jacek12345,
There is a flag `WIFI_PROV_KEEP_BLE_ON_AFTER_PROV` which keeps the BT stack on after provisioning is done. By default, this flag is disabled. When disabled, it cleans all the memory required for BT stack to init. So now after provisioning is done, the flag is disabled, it cleans up all the memory thus preventing BT stack initializing.

The patch shared above has API's name as bleprph but the code is actually performing blecent role of scanning.

The solution to this is:
1. idf.py menuconfig
2. Enable  `WIFI_PROV_KEEP_BLE_ON_AFTER_PROV` from Component config -> Wi-Fi Provisioning Manager -> WIFI_PROV_KEEP_BLE_ON_AFTER_PROV
3. Replace the BT initialing code with just blecent_scan(), like below:

``` 
     /* vTaskDelay(2000);
      esp_err_t ret;
      ret = nimble_port_init();
      if (ret != ESP_OK)
      {
           ESP_LOGE(TAG, "Failed to init nimble %d ", ret);
           return;
     }
     
     //Initialize the NimBLE host configuration.
     ble_hs_cfg.reset_cb = bleprph_on_reset;
     ble_hs_cfg.sync_cb = bleprph_on_sync;
     nimble_port_freertos_init(bleprph_host_task); */
     
     //As BT stack is already up, only scanning the device is remaining
     blecent_scan();
 }
```

jacek12345, 2023-07-05:
@IshaESP , after proposed solution, there is problem:

```
I (27) boot: ESP-IDF v5.1 2nd stage bootloader
I (27) boot: compile time Jul  4 2023 16:17:38
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (72) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (79) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (87) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (94) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (101) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (109) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (117) boot: End of partition table
I (121) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=4bbf8h (310264) map
I (185) esp_image: segment 1: paddr=0006bc20 vaddr=3fc9c400 size=043f8h ( 17400) load
I (189) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=f1a14h (989716) map
I (368) esp_image: segment 3: paddr=00161a3c vaddr=3fca07f8 size=00a34h (  2612) load
I (369) esp_image: segment 4: paddr=00162478 vaddr=40374000 size=18334h ( 99124) load
I (406) boot: Loaded app from partition at offset 0x20000
I (406) boot: Disabling RNG early entropy source...
I (418) cpu_start: Multicore app
I (418) cpu_start: Pro cpu up.
I (418) cpu_start: Starting app cpu, entry point is 0x4037576c
0x4037576c: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.1/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (436) cpu_start: Pro cpu start user code
I (436) cpu_start: cpu freq: 160000000 Hz
I (436) cpu_start: Application information:
I (439) cpu_start: Project name:     switch
I (444) cpu_start: App version:      1.0
I (449) cpu_start: Compile time:     Jul  4 2023 16:17:06
I (455) cpu_start: ELF file SHA256:  fa9bb3dee9101ff0...
I (461) cpu_start: ESP-IDF:          v5.1
I (465) cpu_start: Min chip rev:     v0.0
I (470) cpu_start: Max chip rev:     v0.99
I (475) cpu_start: Chip rev:         v0.1
I (480) heap_init: Initializing. RAM available for dynamic allocation:
I (487) heap_init: At 3FCA7930 len 00041DE0 (263 KiB): DRAM
I (493) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (500) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (506) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (513) spi_flash: detected chip: generic
I (517) spi_flash: flash io: dio
W (521) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
W (534) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (545) sleep: Configure to isolate all GPIO pins in sleep state
I (552) sleep: Enable automatic switching of GPIO sleep configuration
I (559) coexist: coex firmware version: ebddf30
I (564) coexist: coexist rom version e7ae62f
I (569) app_start: Starting scheduler on CPU0
I (574) app_start: Starting scheduler on CPU1
I (574) main_task: Started on CPU0
I (584) main_task: Calling app_main()
I (584) esp_rmaker_console: Initialising UART on port 0
I (594) uart: queue free spaces: 8
I (594) esp_rmaker_commands: Registering command: reboot
I (604) esp_rmaker_commands: Registering command: up-time
I (614) esp_rmaker_commands: Registering command: mem-dump
I (614) esp_rmaker_commands: Registering command: task-dump
I (624) esp_rmaker_commands: Registering command: cpu-dump
I (624) esp_rmaker_commands: Registering command: sock-dump
[... 84 lines cut ...]
I (6504) esp_rmaker_param: Reporting params (init): {"Switch":{"Name":"Switch","Power":true},"Time":{"TZ":"Europe/Warsaw","TZ-POSIX":"CET-1CEST,M3.5.0,M10.5.0/3"},"Schedule":{"Schedules":[]},"Scenes":{"Scenes":[]},"Local Control":{"POP":"b590bfe7","Type":1}}
I (6524) esp_rmaker_ota_using_topics: Subscribing to: node/3485184ACC98/otaurl
I (6794) app_main: MQTT Published. Msg id: 39465.
I (7034) app_main: MQTT Published. Msg id: 39914.
I (7034) app_main: MQTT Published. Msg id: 13128.
I (11534) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
I (11804) app_main: MQTT Published. Msg id: 63757.
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception).
Debug exception reason: BREAK instr
Core  0 register dump:
PC      : 0x403743c0  PS      : 0x00040736  A0      : 0x00000000  A1      : 0x3fcab6a0
0x403743c0: _DoubleExceptionVector at C:/Espressif/frameworks/esp-idf-v5.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:564

A2      : 0x00000001  A3      : 0x3fcab6c0  A4      : 0x3fca174c  A5      : 0x0000018a
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x82025b42  A9      : 0x3fcab680
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x8038493e  A13     : 0x3fcabd30
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x0000001a  EXCCAUSE: 0x00000001
EXCVADDR: 0xffffffe0  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff
0x40056f5c: memcpy in ROM

0x40056f72: memcpy in ROM

Backtrace: 0x403743bd:0x3fcab6a0
0x403743bd: _UserExceptionVector at ??:?
```

IshaESP, 2023-07-05:
Hi @jacek12345,
Yes if the device is already provisioned, BLE stack won't initialize. The current application is suited for most common rainmaker use case. However, you can refer to existing nimble sample applications to how to re - init BT (stack and controller ).
You can use `wifi_prov_mgr_is_provisioned` to check if it's already provisioned and then take appropriate steps for initialize or not initialize BT.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/controller_vhci.html#_CPPv418esp_bt_mem_release13esp_bt_mode_t  

This is the documentation . The current use supported is to use BLE stack for provisioning and then turn off stack and release memory . Therefore , memory too is released . But when memory is released, it cannot be reversed.  
The use case being suggested here involves rebooting the device. If you enable the flag, and keep running the code, it should work as configured.

jacek12345, 2023-07-05:
Thank You for response.
Please check if this flow will be correct:
1.  Using wifi_prov_mgr_is_provisioned  after NVS initialisation and set flag if it is provisioned (FLAG_PROVISIONED=true)
2.   After app_wifi_start(), check FLAG_PROVISIONED. If set (device was already provisioned so BT was not initialised), initialise nimble. If cleared (device was unprovisioned so provisioning was started and BT was initialised and remains initilaised), ommit BT initialisation.

But if i understood, now, when device is already provisioned it should work with my own BT initilaisation regardless of WIFI_PROV_KEEP_BLE_ON_AFTER_PROV, but it crasches. Why?

IshaESP, 2023-07-05:
Hi @jacek12345 ,
So you don't have to explicitly set provisioned flag. The API sets the flag if the device is already provisioned.

Here are the 2 cases that need to be addressed:
1. Device is not provisioned:
- Keep the WIFI_PROV_KEEP_BLE_ON_AFTER_PROV flag enabled. After provisioning no need of doing anything extra, just start scan/adv according to the use case. 

2. Device is provisioned:
- Across reboots, this flag WIFI_PROV_KEEP_BLE_ON_AFTER_PROV will be of no use. So if device is already provisioned , you need to initialize BLE explicitly as reboot will start a clean slate. I have attached a patch which is working for me. Try it once.
[crash_rainmaker.zip](https://github.com/espressif/esp-idf/files/11957274/crash_rainmaker.zip)

jacek12345, 2023-07-05:
I have `error: corrupt patch at line 20`

jacek12345, 2023-07-05:
Finally, it looks like the problem was missing:
$ENV{IDF_PATH}/examples/bluetooth/nimble/common/nimble_central_utils
in CMakeLists.txt.
Now it does not crash, only put nimble initialisation error if provisioning was made in this boot (ble is already initialised).

But there is strange issue with WIFI_PROV_KEEP_BLE_ON_AFTER_PROV. If disabled, it crashes on ble initialisation even if there was not provisioning in this boot (ble was not already initialised). If enabled, initialisation take place correctly. This is strange, because it shouldn't be used when there was not provisioning in this boot.

IshaESP, 2023-07-12:
Hi
Regarding $ENV{IDF_PATH}/examples/bluetooth/nimble/common/nimble_central_utils, if this was the issue the code wouldn't have compiled earlier. Are there any stale files present?

So if provisioning is already done, you need to enable WIFI_PROV_KEEP_BLE_ON_AFTER_PROV for using BLE. During init process, if the flag is disabled, it releases all the memory related to ble thus causing a crash (irrespective of provisioning).

Attaching a log when WIFI_PROV_KEEP_BLE_ON_AFTER_PROV is enabled and provisioning is already done.
[github_10817_2.txt](https://github.com/espressif/esp-idf/files/12026047/github_10817_2.txt)

rahult-github, 2023-11-06:
Hi @jacek12345  , any further updates regarding WIFI_PROV_KEEP_BLE_ON_AFTER_PROV macro usage ?

PhLuReh, 2023-11-09:
I ran into the same issue working on an [esp-matter](https://github.com/espressif/esp-matter/) based project. It ocures while booting a commissioned device (this should be the normal state). And let me tell it is very urgent.

I looked around in the files and was thinking about why there is no sanity-check if the element in the [linked list](https://github.com/espressif/esp-idf/blob/b90dfe075923870a8d1f8d85a75318781455f57a/components/freertos/FreeRTOS-Kernel/list.c#L209) actually has a next element. The next element will just be used... I really don't know if there are some specialties on this linked list (like being a closed loop), but anyhow, it is just a guess... Even if it should not appear in reality, there should be a check for `NULL`.

Here is my stack trace:
```
E (6086) chip[DL]: Long dispatch time: 1433 ms, for event type 2
I (6086) chip[DL]: WIFI_EVENT_STA_START
I (5686) wifi:state: auth -> assoc (0)
I (6096) chip[DL]: Done driving station state, nothing else to do...
I (6096) wifi:Association refused temporarily, comeback time 1000 (TUs)
I (6106) chip[DL]: CHIPoBLE advertising stopped
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40095ac3  PS      : 0x00060933  A0      : 0x800941a9  A1      : 0x3fff33d0
0x40095ac3: uxListRemove at /opt/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/list.c:197

A2      : 0x00000006  A3      : 0x00060920  A4      : 0x00000000  A5      : 0x00060923
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x80095140  A9      : 0x3fff33a0
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000a  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
0x4000c2e0: memcpy in ROM

0x4000c2f6: memcpy in ROM

Backtrace: 0x40095ac0:0x3fff33d0 0x400941a6:0x3fff33f0 0x401516cd:0x3fff3410 0x401516e7:0x3fff3430 0x4013ab95:0x3fff3450 0x40094ff6:0x3fff3470
0x40095ac0: uxListRemove at /opt/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/list.c:192
0x400941a6: vTaskDelete at /opt/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:1434 (discriminator 4)
0x401516cd: esp_nimble_disable at /opt/espressif/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/nimble_port_freertos.c:55
0x401516e7: nimble_port_freertos_deinit at /opt/espressif/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/nimble_port_freertos.c:80
0x4013ab95: chip::DeviceLayer::Internal::BLEManagerImpl::bleprph_host_task(void*) at /home/smart/projects/smp_matter/build/esp-idf/chip/../../../../../../../opt/espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:864
0x40094ff6: vPortTaskWrapper at /opt/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

jacek12345, 2023-11-09:
> Hi @jacek12345 , any further updates regarding WIFI_PROV_KEEP_BLE_ON_AFTER_PROV macro usage ?

Thank You, I have this enabled and everything works. You can close.

PhLuReh, 2023-11-09:
I would say, the flag may ease the symptom by the root is somewhere else. As I wrote, the Matter implementation from Espressif also uses the Nimble-Stack, at completely controls it. So no interaction by my application code. And it still appears.

Should I guide someone from the esp-matter repo here?

rahult-github, 2023-11-10:
Hi @PhLuReh  , i believe the issue you have raised and the one being discussed are not same.  

I suggest to please open a fresh ticket on esp-matter repo. Also request you :
1. Mention commit id being used ( Please sync to latest sources to confirm if issue is observed)
2. Provide complete console log
3. Mention steps done / chip used in the description to assist looking at at the issue. 

I will be closing the current issue.

closed 2023-11-10 by rahult-github

PhLuReh, 2023-11-10:
@rahult-github I will do so. Just to conclude from my side. I added the following line to the list.c code and the abort get's executed:
`if (pxItemToRemove < (ListItem_t* const)0xff) abort();`

So I would assume, pxItemToRemove get's corrupted, is not NULL but according to the BT `pxItemToRemove` was `0x0000000a`. Did not get any clue how up to now.