espressif/esp-idf #10818: 512KB flash size Support for esp32s3

5ami, 2023-02-22:
### Is your feature request related to a problem?

I am writing to seek your assistance with a problem I am encountering while building and flashing for a custom board using the ESP-IDF v5.0.
I could flash the program somehow  but I get this message and the board keeps restarting. 

E (265) spi_flash: Detected size(512k) smaller than the size in the binary image header(2048k). Probe failed.

Specifically, my custom board has a flash size of 512KB, but I am unable to find the option for this size in the "sdkconfig".
I could see options to detect the flash size, but it doesnt seem to work. 

The esptool.py has --flash_size argument where I can specify 512KB, but  what about the build ?

[image]

### Describe the solution you'd like.

 To be able to select an option 512KB in sdkconfig

Icarus113, 2023-02-22:
The `Detected size(512k)` indicates your flash chip physical size. `binary image header(2048k)` can be controlled by [this](https://github.com/espressif/esp-idf/blob/master/components/esptool_py/Kconfig.projbuild#L150) but the least one is 1MB. IIRC we have some way to tune the size in the image header...

Besides, what's the actual image size? (You may use idf.py size to know this)

5ami, 2023-02-22:
How to tune the size the in the image header ?

That's what I get when I use idf.py size.  I am compiling Project with an Empty main. 

**Total sizes:**
Used static IRAM:   48562 bytes ( 313678 remain, 13.4% used)
      .text size:   47535 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   10128 bytes ( 162800 remain, 5.9% used)
      .data size:    7864 bytes
      .bss  size:    2264 bytes
Used Flash size :  103403 bytes
      .text     :   74123 bytes
      .rodata   :   29024 bytes
**Total image size:**  159829 bytes (.bin may be padded larger)

ginkgm, 2023-02-24:
I highly doubt if this size of flash can be used in real products. If you needs OTA, then two OTA partitions are required, I guess this will increase the size requirements to larger than 512K.

That's why Espressif hasn't introduced official module smaller than 2MB. 

Would you please try to link all the features you need, and evaluate the size requirement again? I guess WiFi will cost a lot. If during the process, you would like to run some tests first, you may switch the SIZE config to 1MB and then bypass (manual modify) the check in IDF. 
As long as the partition table is smaller than 512KB, I think there is no other place that will forbid your program from starting up and execute normally.

5ami, 2023-02-24:
I am sure 512KB is more then  enough for our application.  We don't use Wi-Fi, We don't use Bluetooth and we barely use any ESP-IDF component (We made our own peripheral drivers). We choose to use esp32s3 because it has a dual 240Mhz processor with a cheap price. 

`If during the process, you would like to run some tests first, you may switch the SIZE config to 1MB and then bypass (manual modify) the check in IDF, As long as the partition table is smaller than 512KB, I think there is no other place that will forbid your program from starting up and execute normally.`

I already did that and bypassed a check in the bootloader, and did some size optimizations, and my application boots, but I want to find an option or a command to make the binary image header 512KB. 

[image]

ceerqingtingml, 2024-03-20:
@5ami Are you using ESP32-S3-WROOM-2-N16R8V or ESP32-S3-WROOM-2-N32R8V? If yes, you need to enable Octal flash via menuconfig.
Config Path: (Top) -> Serial flasher config -> Enable Octal Flash

5ami, 2024-04-01:
I am using a custom board with an esp32s3 chip. Not using a WROOM module

muchajan, 2025-06-22:
@5ami I'm facing the same issue - I want to use 512KB flash in my custom board. Have you found a solution?