espressif/esp-idf #10819: esp_hf_ag_init() API call always fails

DirkGmailMeintjies, 2023-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When calling esp_hf_ag_init() after successful completion of esp_bluedroid_enable() the result is always ESP_FAIL. Even if I open the hfp_ag example app and run it on an ESP32 dev board the result is the same. The example source code does not check the return code of esp_hf_ag_init(), so I added a line to check the return code and it is always ESP_FAIL.

So, in the example code when trying to use the connection option the result is also always:

[0;33mW (132025) BT_BTIF: CHECK_HF_INIT: btc_hf_connect_audio: HF AG not initialized

How to make HF AG init pass in the example project? Is the example outdated? Or, is there an updated example available?

BetterJincheng, 2023-02-22:
@DirkGmailMeintjies Which branch did you use ? And the commit ID ?

DirkGmailMeintjies, 2023-02-23:
> 

I'm using IDE V2.7.0 Build 20221110-1527
ESP-IDF v5.0

BetterJincheng, 2023-02-23:
@DirkGmailMeintjies I need the commit ID of ESP-IDF. Thanks

DirkGmailMeintjies, 2023-02-24:
I'm sorry I'm not sure where to get the commit ID? Please indicate to me where I can see it. I installed the whole IDF and IDE from this download link: https://dl.espressif.com/dl/esp-idf/

BetterJincheng, 2023-02-28:
@DirkGmailMeintjies Could you please show me the source code of esp_hf_ag_init() ?

DirkGmailMeintjies, 2023-02-28:
Here we go, copied from esp_hf_ag_api.c:

`esp_err_t esp_hf_ag_init(void)
{
    if (esp_bluedroid_get_status() != ESP_BLUEDROID_STATUS_ENABLED) {
        return ESP_ERR_INVALID_STATE;
    }
    btc_msg_t msg;
    msg.sig = BTC_SIG_API_CALL;
    msg.pid = BTC_PID_HF;
    msg.act = BTC_HF_INIT_EVT;

    /* Switch to BTC context */
    bt_status_t status = btc_transfer_context(&msg, NULL, sizeof(btc_hf_args_t), NULL);
    return (status == BT_STATUS_SUCCESS) ? ESP_OK : ESP_FAIL;
}
`

DirkGmailMeintjies, 2023-02-28:
I see a problem with the call to btc_transfer_context(), because arg is NULL but arg_len is not 0! This means that btc_transfer_context() will return BT_STATUS_PARM_INVALID, and then esp_hf_ag_init() will return ESP_FAIL. Has this been fixed in a later version yet?

BetterJincheng, 2023-02-28:
@DirkGmailMeintjies  Yes, that is exactly the problem. And, this problem has been fixed. Please try with the lastet version.

DirkGmailMeintjies, 2023-02-28:
Thank you I will install latest version and retry.

DirkGmailMeintjies, 2023-03-01:
I can confirm that IDF V5.0.1 fixes the problem, thanks!

closed 2023-04-25 by Alvin1Zhang