espressif/esp-idf #10823: Got CMake issues with set-target ESP32C6 command

Quynh-Nguyen, 2023-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-master-045163a2ec99eb3cb7cc69e2763afd145156c4cf

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Able to set-target to the chip. It's working well with version 4.3
But version 4.3 not supported ESP32C6 yet.

### What is the actual behavior?

```
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/connectedhomeip/.environment/pigweed-venv/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world"...
-- Found Git: /opt/homebrew/bin/git (found version "2.34.0")
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/whyn4/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/whyn4/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/whyn4/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/tools/cmake/build.cmake:542 (message):
  usage: __main__.py [-h] [--project_dir PROJECT_DIR]

                     [--interface_version {0,1}]
                     {prepare_dependencies,inject_requirements,inject_requrements}
                     ...

  __main__.py: error: argument --interface_version: invalid choice: 2 (choose
  from 0, 1)

Call Stack (most recent call first):
  /Users/whyn4/Projects/RESEARCH/MATTER_IDF_MASTER/esp-idf/tools/cmake/project.cmake:453 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_12445 and /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_12445
The test for idf_size has failed. Please examine the artifacts.
```

### Steps to reproduce.

1. Execute `./install.sh`
[... 77 lines cut ...]
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/whyn4/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/tools/cmake/build.cmake:542 (message):
  usage: __main__.py [-h] [--project_dir PROJECT_DIR]

                     [--interface_version {0,1}]
                     {prepare_dependencies,inject_requirements,inject_requrements}
                     ...

  __main__.py: error: argument --interface_version: invalid choice: 2 (choose
  from 0, 1)

Call Stack (most recent call first):
  /Users/whyn4/Projects/RESEARCH/MATTER_IDF_MASTER/esp-idf/tools/cmake/project.cmake:453 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_12445 and /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_12445
The test for idf_size has failed. Please examine the artifacts.
```

dobairoland, 2023-02-23:
Hi @Quynh-Nguyen.

The reported issues are not even happening for ESP32-C6. All of them are for ESP32:

> -- Building ESP-IDF components for target esp32

BTW, why would you want to run `./tools/test_idf_size/test.sh`? That is an internal test script.

If you want to try ESP32-C6 support then please got to an ESP-IDF project, e.g. https://github.com/espressif/esp-idf/tree/master/examples/get-started/hello_world and run there `idf.py --preview set-target esp32c6`. It should work.

Please be aware that full ESP32-C6 support is not ready yet (see https://github.com/espressif/esp-idf/issues/10423). This chip will be supported in ESP-IDF v5.1 which will be released in a couple of Months.

Quynh-Nguyen, 2023-02-23:
@dobairoland 

I'm getting the same issue with the test run

```
MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world > master  idf.py --preview set-target esp32c6                                      
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c6, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/connectedhomeip/.environment/pigweed-venv/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32c6 -DCCACHE_ENABLE=0 /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world"...
-- Found Git: /opt/homebrew/bin/git (found version "2.34.0")
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/whyn4/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/whyn4/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/whyn4/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c6
CMake Error at /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/tools/cmake/build.cmake:542 (message):
  usage: __main__.py [-h] [--project_dir PROJECT_DIR]

                     [--interface_version {0,1}]
                     {prepare_dependencies,inject_requirements,inject_requrements}
                     ...

  __main__.py: error: argument --interface_version: invalid choice: 2 (choose
  from 0, 1)

Call Stack (most recent call first):
  /Users/whyn4/Projects/RESEARCH/MATTER_IDF_MASTER/esp-idf/tools/cmake/project.cmake:453 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_72083 and /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_72083
```

Quynh-Nguyen, 2023-02-24:
@dobairoland 
Do we have any under-development branch able to work with, especially ESP32C6, Roland?

dobairoland, 2023-02-24:
Have you run `install.sh` and `source export.sh` in the terminal where you are attempting to run `idf.py set-target`? 

> Do we have any under-development branch able to work with, especially ESP32C6

That would be the master branch you already using.

Quynh-Nguyen, 2023-02-24:
@dobairoland 

>Have you run install.sh and source export.sh in the terminal where you are attempting to run idf.py set-target?

Yes, I did with no error, Roland.

dobairoland, 2023-02-24:
What is the output of `python -m pip freeze | grep component`?

Maybe the following will solve the issue:
```
python -m pip install -U  idf-component-manager
```

Quynh-Nguyen, 2023-02-24:
> 

Yes,

This is the result

```
python -m pip freeze | grep component
idf-component-manager==1.1.4
```

Quynh-Nguyen, 2023-02-24:
@dobairoland 
Oh wow, it worked when I upgraded IDF manager.
But I'm getting another error with `kconfgen` module, which I am trying to resolve it now.

Quynh-Nguyen, 2023-02-24:
This is the issue I got

```
 No module named kconfgen.__main__; 'kconfgen' is a package and cannot be directly executed
CMake Error at /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/tools/cmake/kconfig.cmake:176 (message):
  Failed to run kconfgen
  (/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/connectedhomeip/.environment/pigweed-venv/bin/python;-m;kconfgen;--list-separator=semicolon;--kconfig;/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/Kconfig;--sdkconfig-rename;/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/sdkconfig.rename;--config;/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/sdkconfig;--env-file;/usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/examples/get-started/hello_world/build/config.env).
  Error 1
Call Stack (most recent call first):
  /usr/local/var/www/RESEARCH/MATTER_IDF_MASTER/esp-idf/tools/cmake/build.cmake:613 (__kconfig_generate_config)
  /Users/whyn4/Projects/RESEARCH/MATTER_IDF_MASTER/esp-idf/tools/cmake/project.cmake:453 (idf_build_process)
  CMakeLists.txt:6 (project)
```

dobairoland, 2023-02-24:
You can try the same:

```sh
python -m pip freeze | grep kconfig
```

and 

```sh
python -m pip install -U esp-idf-kconfig
```

dobairoland, 2023-02-24:
Note: The component manager issue was resolved by setting `idf-component-manager~=1.2` at https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.1.txt. The change will appear soon.

Quynh-Nguyen, 2023-02-24:
@dobairoland 
Thank you so much, Roland
It's worked well now.

dobairoland, 2023-02-24:
Oh, great!

closed 2023-02-24 by dobairoland

jiapei100, 2024-02-05:
@dobairoland 
@Quynh-Nguyen 

Any other work-around if I insist on using `idf-component-manager>=1.4` ??

dobairoland, 2024-02-05:
Re-running the install script will install the latest version of the component manager.