espressif/esp-idf #10825: Failed to connect STA in some ESP32 but same code working in other.

abhi-ratnman, 2023-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using ESP32 in one of the product. 

While using the same code to a STA it works on some ESP32 while on the other ESP32 it does not work.

Tested it on 10 different ESP32 from that 6 were able to connect to WiFi STA and 4 were unable to connect to the STA. 

I purchased the modules from Campus Components Pvt Ltd India, the website of the company says they are authorised dealer of espressif. 

Still such a failure is observed. Also I am not the only one facing the issues there are many other threads that face the same issue (for refrence. [issue 6427](https://github.com/espressif/esp-idf/issues/6427) they are also facing the same issue. Still no resolve of error. 

How can one manufacture product using such chip if they are not even sure after development of a product for 8-10 months. And while going in mass production such an issue arise. We have ordered 200pcs and from that tested 10pcs. Request to team espressif to not take this issue lightly because such failures are not expected from a company of your scale. 

Thanks,
Abhi
luckyabhi49@gmail.com

Xiehanxin, 2023-02-28:
hi Abhi, have you tried the station example? and what type of pruduct do you use, single chip or module? if you use the single chip, maybe you can share the schematic to sales@espressif.com and describe your problem in the mail, we can double check your design

AxelLin, 2023-06-04:
@abhi-ratnman 
Could you share the connect failure log? (Could you check if the faulty devices all have the same connect failure?)

AxelLin, 2023-12-23:
@abhi-ratnman 
How is the status now? Have you tried recent esp-idf releases?
If it still does not work, could you share the connect failure log?

closed 2024-03-18 by Alvin1Zhang