espressif/esp-idf #10829: IDF 4.4迁移到IDF 5.0 启动时程序校验失败，但是运行正常

zbqxyz, 2023-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF Version:v5.0-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

校验正常，dump时能自动重启

### What is the actual behavior?

IDF 4.4迁移到IDF 5.0 启动时程序校验失败,运行中程序出现异常后打印完dump，提示booting，但是也不会自动重启，就死在哪里，之前在idf4.4，程序运行异常打印完dump是可以自动重启的。

### Steps to reproduce.

IDF 4.4迁移到IDF 5.0 ，用命令idf.py -p /dev/ttyS5 -b 1152000 flash 烧录完成，启动时输出log:
SP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b9b
SPIWP:0xee
mode:DIO, clock div:1
**load:0x3fce3810,len:0x18ac
load:0x403c9700,len:0xce8
load:0x403cc700,len:0x2fc8
SHA-256 comparison failed:
Calculated: f243ac715f2300e33d363593b0d4c766b7430d87643f567792a4145b8906eb8c
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...**
entry 0x403c992c
I (35) HOOK: This hook is called BEFORE bootloader initialization
I (54) boot: ESP-IDF v5.0-dirty 2nd stage bootloader
I (54) boot: compile time 10:33:06
I (54) boot: chip revision: v0.1
I (56) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (64) boot.esp32s3: Boot SPI Speed : 80MHz
I (68) boot.esp32s3: SPI Mode       : DIO
I (73) boot.esp32s3: SPI Flash Size : 8MB
I (78) boot: Enabling RNG early entropy source...
I (83) HOOK: This hook is called AFTER bootloader initialization
I (90) boot: Partition Table:
I (93) boot: ## Label            Usage          Type ST Offset   Length
I (101) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (108) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (116) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (123) boot:  3 ota_0            OTA app          00 10 00010000 00300000
I (131) boot:  4 ota_1            OTA app          00 11 00310000 00300000
I (139) boot:  5 user_save        WiFi data        01 02 00610000 00003000
I (146) boot: End of partition table
I (150) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=4884ch (297036) map
I (159) esp_image: segment 1: paddr=00058874 vaddr=3fc9a900 size=06254h ( 25172) load
I (169) esp_image: segment 2: paddr=0005ead0 vaddr=40374000 size=01548h (  5448) load
I (176) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=dcbech (904172) map
[... 102 lines cut ...]
I (304) heap_init: At 600FE03C len 00001FB4 (7 KiB): RTCRAM
I (311) spi_flash: detected chip: generic
I (315) spi_flash: flash io: dio
W (319) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (327) sleep: Configure to isolate all GPIO pins in sleep state
I (334) sleep: Enable automatic switching of GPIO sleep configuration
I (341) esp_core_dump_uart: Init core dump to UART
I (346) coexist: coexist rom version e7ae62f
I (352) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (372) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (372) gpio: GPIO[11]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
I (382) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:3 
I (392) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:1 
I (402) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:1 
E (412) RTCIO: rtc_gpio_hold_dis(163): RTCIO number error
I (422) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (432) gpio: GPIO[45]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (442) i2c: i2c_driver_delete(433): i2c driver install error
E (442) i2c: i2c_driver_delete(433): i2c driver install erro
```

### More Information.

[image]

ESP-Marius, 2024-11-19:
@zbqxyz sorry for the late response here. 

Are you still experiencing issues? Are you able to produce a minmal example that lets us reproduce this?

ESP-Marius, 2024-12-04:
Closing due to lack of updates

closed 2024-12-04 by ESP-Marius