espressif/esp-idf #10830: GPIO_NUM_18 not changing level ESP32-S3

Divraj-Dev, 2023-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

GPIO_NUM_18 not changing levels

`gpio_config_t pump_config = {
        .pin_bit_mask = (1ULL << GPIO_NUM_18),
        .mode = GPIO_MODE_OUTPUT,
        .pull_up_en = GPIO_PULLUP_DISABLE,
        .pull_down_en = GPIO_PULLDOWN_DISABLE,
        .intr_type = GPIO_INTR_DISABLE,
    };
    gpio_config(&pump_config);
    gpio_set_level(GPIO_NUM_18, 0);`

"Triggered by Button Press"
`gpio_set_level(GPIO_NUM_18, 1);`

Using the same code I am able to change other GPIO pins however GPIO_NUM_18 will not change. Reading with a multimeter, the value is always 1.65V.

Tried with IDF 5.0.1 and 4.4

CC-MASTER05, 2023-03-02:
Hello,

I am have same issue with rainmaker have you found solution?

Thanks 
Fenil

Divraj-Dev, 2023-03-05:
Hi @CC-MASTER05 

Not yet, Im also monitoring the other thread for any esp engineer to reply but nothing yet.

CC-MASTER05, 2023-05-17:
Hi,
Now all the GPIO's working fine in IDFV5.0.1 if you have still question or not able to operate can you share more details if i able to solve then i will help.
Thanks

readmodifywrite, 2023-06-05:
I can confirm that GPIO 18 does not work on the ESP32-S3 as of ca8bf608df6caf9f83df2e2f98ff71815ad5c186 on June 1.  I've tried 5.0, 5.0.1, and 5.0.2 as well.  

GPIO 13 also does not work.

Other GPIOs do, a seemingly random set.  GPIO 2, 4, 5, 9, 10, and 12 are all definitely working on my board.

This along with a few other pretty serious GPIO problems on the S3 are bad enough that I am probably designing the ESP32-S3 out of a production design in favor of an STM32.