espressif/esp-idf #10831: HTTP_CLIENT: esp_transport_read returned:-1 and errno:128

franz-ms-muc, 2023-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

```
W (08:14:20.210) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128
E (08:14:20.211) esp_https_ota: data read -1, errno 128
```

i look up Error Numbers in https://github.com/espressif/esp-lwip/blob/10197b212a95c49c733fb18ffed56cafb0d196d4/src/include/lwip/errno.h

but this list end at 124. 

pls Help. Need to know what 128 is. 

Thanks!

AxelLin, 2023-02-24:
Try printing strerror(errno)

ESP-YJM, 2023-02-27:
@franz-ms-muc 128 means the peer send `FIN` to close the tcp connection.

franz-ms-muc, 2023-02-28:
The question is who is the peer (embedded device or the backend)..

Should be the ESP since it is receiving the data. Please , correct me if I am wrong.

Alvin1Zhang, 2023-03-27:
@franz-ms-muc Thanks for reporting, would you please help share if any updates for the issue? Thanks.

laukik-hase, 2023-03-31:
Closing this issue due to lack of feedback, feel free to reopen with more updates. Thanks.