espressif/esp-idf #10832: 请问LWIP计时器API的文档问题

WangShuoran, 2023-02-23:
### Is your feature request related to a problem?

about the documentation of LWIP timer API:
https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32s3/api-guides/lwip.html#lwip-timers-api
When users are not using WiFi, these APIs provide users with the ability to turn off LwIP timer to reduce power consumption.

I don't know LWIP turn off timer is all of outbound or only wifi outbound?

### Describe the solution you'd like.

None

### Describe alternatives you've considered.

None

### Additional context.

None

xueyunfei998, 2023-02-27:
hi @WangShuoran 

Lwip and wifi are two tasks respectively, and there are several timers in the lwip protocol stack. When wifi is not working, the timer of lwip can be turned off to reduce power consumption.

WangShuoran, 2023-02-27:
But if my cat1 or rj45 outbound with wifi open in the esp32 serial, if wifi close, other outbound rely on lwip. And if lwip timer close because wifi close, does the cat1 or rj45 run ok?

xueyunfei998, 2023-02-28:
hi @WangShuoran 
If your peripheral, such as Ethernet port, uses lwip, lwip timer cannot be turned off at this time.

closed 2023-03-01 by WangShuoran