espressif/esp-idf #10833: SPI Master driver crashes when using shared interrupts (ESP_INTR_FLAG_SHARED)

Erlkoenig90, 2023-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

After instantiating the SPI master driver and passing `ESP_INTR_FLAG_SHARED` to `spi_bus_config_t::intr_flags`, SPI should work and use the shared interrupt.

### What is the actual behavior?

[This assertion](https://github.com/espressif/esp-idf/blob/v5.0.1/components/driver/spi_master.c#L592) is triggered when an SPI interrupt occurs.

### Steps to reproduce.

Instantiate the SPI driver like this:
```c++
static spi_bus_config_t spiConfig = {
	...
	.intr_flags = ESP_INTR_FLAG_SHARED
};
ESP_ERROR_CHECK (spi_bus_initialize (SPI3_HOST, &spiConfig, SPI_DMA_CH_AUTO));
```
Perform some transactions like
```c++
spi_transaction_t trans {
	.flags = SPI_TRANS_USE_RXDATA,
	.cmd = static_cast<std::uint8_t> (0x80 | addr),
	.addr = 0,
	.length = 16,
	.rxlength = 8,
	.user = nullptr,
	.tx_buffer = nullptr,
	.rx_buffer = nullptr
};
ESP_ERROR_CHECK (spi_device_transmit (devHandle, &trans));
```
And simultaneously use some other periphery which uses the shared interrupt.

### Debug Logs.

```plain
assert failed: spi_intr spi_master.c:592 (spi_hal_usr_is_done(&host->hal))

Backtrace: 0x403760ce:0x3fc96c20 0x4037d4fd:0x3fc96c40 0x40383c2d:0x3fc96c60 0x40378846:0x3fc96d80 0x40376735:0x3fc96db0 0x403773f5:0x3fc96dd0 0x400559dd:0x3fc9ce50 |<-CORRUPTED
0x403760ce: panic_abort at esp-idf-v5.0.1/components/esp_system/panic.c:423

0x4037d4fd: esp_system_abort at esp-idf-v5.0.1/components/esp_system/esp_system.c:153

0x40383c2d: __assert_func at esp-idf-v5.0.1/components/newlib/assert.c:78

0x40378846: spi_intr at esp-idf-v5.0.1/components/driver/spi_master.c:592 (discriminator 1)

0x40376735: shared_intr_isr at esp-idf-v5.0.1/components/esp_hw_support/intr_alloc.c:436

0x403773f5: _xt_lowint1 at esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118
```
```

### More Information.

I think the issue is caused by using `esp_intr_alloc` for a (possibly) shared interrupt. The shared interrupt handler (`shared_intr_isr`) can't check whether the interrupt was actually triggered by the SPI periphery, so `spi_intr` is called even though there was no SPI interrupt. This then triggers the assertion.

So we should either:
* state in the documentation that `ESP_INTR_FLAG_SHARED` in `spi_bus_config_t::intr_flags` is **not** supported (and possibly add an assertion to check that)
* Change the assertion to a simple return instead of crashing: `if (!spi_hal_usr_is_done(&host->hal)) return;`
* Use `esp_intr_alloc_intrstatus` instead and pass the SPI register to only call `spi_intr` when an SPI interrupt actually occured, e.g.:

```c
err = esp_intr_alloc_intrstatus(spicommon_irqsource_for_host(host_id),
					bus_attr->bus_cfg.intr_flags | ESP_INTR_FLAG_INTRDISABLED,

					(uint32_t) &host->hal.hw->dma_int_raw,
					SPI_TRANS_DONE_INT_RAW,

					spi_intr, host, &host->intr);

```

This block then needs to be moved *below* the call to `spi_hal_init`.

wanckl, 2023-04-19:
@Erlkoenig90 Hello, not all interrupt source is able to be shared, it decided by chip hardware.    

For example, SPI2 interrupt source of ESP32S3 is not able to be shared, only SPI2 can alert this interrupt, there is no means to share this, you may enter `spi_intr` by accident, but other peripherals you shared can't alert this interrupt.

Erlkoenig90, 2023-04-19:
Hi, in this case the SPI documentation should state that `ESP_INTR_FLAG_SHARED` in `spi_bus_config_t::intr_flags` is not supported, and an assertion to check for that would be helpful. Currently it is not clear from the IDF documentation that this interrupt can't be shared.

KaeLL, 2023-04-19:
@Erlkoenig90's got a point. Such a check should also be replicated to other peripherals components' initialization routines.

wanckl, 2023-04-25:
@Erlkoenig90 
You are right, it is understandable, I think espressif official can accept this.

wanckl, 2023-05-04:
@Erlkoenig90 Hello, Sorry for my last replay.   

For the newest information, I heard that this interrupt can be shared, it is a bug on SPI_Master driver, and they will fix it recently :fire:

Erlkoenig90, 2023-05-06:
Great, I appreciate it!

closed 2025-12-04 by Alvin1Zhang