espressif/esp-idf #10834: Wrong reset cause on brownout (ESP_RST_WDT instead of ESP_RST_BROWNOUT)

kriegste, 2023-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO (ESP32)

### Power Supply used.

USB

### What is the expected behavior?

After a brownout and the subsequent restart, ESP_RST_BROWNOUT should be returned by esp_reset_reason().

### What is the actual behavior?

Instead, esp_reset_reason() returns ESP_RST_WDT.

### Steps to reproduce.

Run any code that displays the result of esp_reset_reason(), then provoke a supply voltage instability. The console will show "Brownout detector was triggered", the ESP32 will restart.

### More Information.

The code sets the reset reason correctly:
https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_system/port/brownout.c#L66
But it seems to get overwritten by the way the CPUs are reset (using the watchdog).

igrr, 2023-02-24:
>  The console will show "Brownout detector was triggered", the ESP32 will restart and show the wrong reason.

@kriegste By "show the wrong reason", are you referring to the message printed by the ROM bootloader (`rst:0x1 (RTC_WDT_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)`) or the return value of `esp_reset_reason` function on the subsequent boot?

kriegste, 2023-02-24:
The return value of `esp_reset_reason()` is wrong.

I have not checked the bootloader message yet. But it is irrelevant since the app relies on esp_reset_reason(). Some background why this is important:
To make sure all is well after an OTA update, the app monitors each reset reason. If the reason is either ESP_RST_PANIC, ESP_RST_INT_WDT, ESP_RST_TASK_WDT or ESP_RST_WDT five times in a row, this is interpreted as being stuck in a boot loop. The app then attempts a roll back. Now, ESP_RST_BROWNOUT is a hardware problem and should not be the cause for a firmware roll back. So this cause should be distinguishable from the (regular) ESP_RST_WDT.

ESP-Marius, 2023-02-28:
@kriegste the reset reason seems to be set correctly on v4.4.4 after a brownout is detected when I tested.

What could be happening though is that after setting the brownout hint the restart code could be getting stuck somewhere. (we enable the RTC WDT while doing the reset to protect us from this).

Are you able to reproduce this using a minimal example, e.g. hello world?

kriegste, 2023-02-28:
Strange. It seems to be a different issue.

I had the device running for days, developing and doing OTA updates, but apart from provoking brownouts, I never did a real power cycle.
After a proper power cycle (hardware reset), the problem disappeared... I can no longer reproduce it... The reset reason is correct now (ESP_RST_BROWNOUT).

Could it be that I messed up the RTC WDT so bad that even the (software) resets didn't fix it? I wonder how I can debug this.

ESP-Marius, 2023-03-01:
@kriegste do you remember what the reset reason printed from the bootloader was? Do you have a full log of the whole process? If it was a SW reset then yes, the RTC related configs would not be cleared, but IDF should still reconfigure the RTC WDT on next boot.

Did your SW touch the RTC in any way (menuconfig/calling rtc wdt in your code/change)?

kriegste, 2023-03-01:
No, I do not see the bootloader log, I redirect the console messages to RAM using `esp_log_set_vprintf()` and `esp_rom_install_channel_putc()` access it via a HTTP server. So whatever happens before that, is lost.

The app does not touch the RTC WDT. The only "RTC" thing it uses is variables in RTC memory, aka `RTC_NOINIT_ATTR`, and lots of it.

It sounds plausible that brownouts can make the chip hang. But the message "Brownout detector was triggered" appeared every time. And now, try as I may, I cannot reproduce the issue any more.

One thing, though, I will attempt to do: Running an older version of my app compiled with 4.4.3 and then OTA upgrading to the newer 4.4.4 based version. This is what I did in the past weeks. I had the device running on my desk and also the IDF got updated during the time.

ESP-Marius, 2023-04-26:
@kriegste any updates on this one? Are you still seeing this issue?

kriegste, 2023-04-28:
I have not seen it since upgrading to 4.4.4 and doing a proper power-cycle.

ESP-Marius, 2023-05-04:
@kriegste That's good! Closing this one for now then, feel free to reopen if there is any new information.

VitorBFreitas, 2023-05-22:
I have reproduced it powering esp with a regulated dc power supply and dropping it's voltage to something around 1.3V.

szmodz, 2023-05-26:
@igrr This actually makes sense.  What will happen if the flash chip (3.3v) will stop working after the initial reset, during ROM code execution? Will the reset reason stay the same (brownout), or get overwritten by WDT timeout?

szmodz, 2023-05-28:
From my observation, a brownout sometimes results in a RTCWDT_RTC_RESET, same as the spurious RTCWDT_RTC_RESET during initial poweron on ESP32 rev1.

An RTCWDT_RTC_RESET will replace the original reset cause.

Also, I've noticed that the reset hint stored in RTC_RESET_CAUSE_REG is cleared by RTCWDT_RTC_RESET, adding to the confusion.

(this applies to the ESP32)

kriegste, 2023-05-28:
Is there already an issue (link?) for the
> spurious RTCWDT_RTC_RESET during initial poweron
?
TTGO T-Display's ESP32 is indeed revision 1 as well.

szmodz, 2023-05-29:
> Is there already an issue (link?) for the
> 
> > spurious RTCWDT_RTC_RESET during initial poweron
> 

https://www.espressif.com/sites/default/files/documentation/esp32_errata_en.pdf

3.1

VitorBFreitas, 2023-05-29:
I don't know if that's the expected behavior, but from some tests i got this:

- If supply voltage drops to 2.3v and then back to 3.3v: got RTCWDT_BROWN_OUT_RESET

- If supply voltage drops to 1.3v and then back to 3.3v: first got RTCWDT_BROWN_OUT_RESET then after a while RTCWDT_RTC_RESET(I think it's like @szmodz told [here](https://github.com/espressif/esp-idf/issues/10834#issuecomment-1564962596))

- If supply voltage drops to 0.3v and then back to 3.3v: first got RTCWDT_BROWN_OUT_RESET then right after SW_CPU_RESET(This one I think doesn't make sense, not sure)

kriegste, 2023-05-29:
This should be re-opened.

ginkgm, 2023-05-31:
I can't tell accurately since there is no FULL log before and after the brownout... 

But here's my imagination:

1. The voltage drops under BOD threshold
2. Luckily BOD reset happen (or ISR triggered and do a SW reset as well as store the reason into RTC REG)
3. The flash fails to work. Most flash have a working voltage no larger than 2.3V. Some of them are 2.5, 2.7V. There is only a few allows 1.6V.
4. After reset, when 1st bootloader attempts to read from flash, it fails to read the contents correctly. If it takes too long, 1st bootloader will be reset by RTCWDT hardware, and try again. 

If you provide the FULL log and the interval between each significant log, maybe I can try to explain more.. And your sdkconfig will also help a lot.

btw, I think under the situation of brownout, trying to erase/write anything to the flash are very dangerous. Maybe we should try to avoid it.

szmodz, 2023-05-31:
@ginkgm according to the WROVER-B/E datasheet, the "Recommended Operating Conditions" voltage is 3.0 to 3.6V, which is a bit tight TBH.

If you look at the original ESP-WROVER datasheet, which used 1.8V flash, the "Recommended Operating Conditions"  are much less restrictive.

But yeah, I also imagine something like that happens. Point is, the 1st bootloader crash after the initial reboot overwrites the initial reset cause, and also clears the SW defined RTC_RESET_CAUSE_REG.

I don't think anybody mentioned anything about erasing / writing flash other than monitoring the status of OTA update. If the new FW crashed after an OTA update, you generally want to know why it crashed.  Seeing RTC WDT as the reset cause can be confusing.

Personally, I leave the RTC watchdog disabled in the main FW, so at least I know that any RTC WDT crash must have been before "app_main".

szmodz, 2023-05-31:
@ginkgm The BOD detector is pretty much useless if you never actually get "ESP_RST_BROWNOUT" as the cause.

Let's assume we're dealing with buggy prototype HW, for example using an RS-485 transceiver powered from the same regulator as the ESP32. A short on the A, B lines causes large current flow, resulting in a voltage drop on the ESP32 VDD33, resulting in brownout, and what you get on the FW side is RTC WDT.

The purpose of the BOD detector is to identify power supply problems, is it not? It would be better if you saw either brownout or poweron as the cause, not RTC WDT.

szmodz, 2023-05-31:
@ginkgm 
> and the interval between each significant log

You mean the interval between the ESP32 ROM bootloader messages? That would require a custom logger attached to the ESP32 UART, which could also be buggy.

szmodz, 2023-05-31:
@ginkgm I assume the tight WROVER-B/E VDD33 requirements are due to the additional voltage drop on the VDD_SDIO regulator.

szmodz, 2023-06-07:
@ginkgm 

The fact that the message string "\r\nBrownout detector was triggered\r\n\r\n" is placed in FLASH doesn't help either.

It should probably use the DRAM_STR macro.

https://github.com/espressif/esp-idf/blob/dc016f59877d13e6e7d4fc193aa5aa764547f16d/components/esp_system/port/brownout.c#L54

ginkgm, 2023-06-09:
@szmodz 

> according to the WROVER-B/E datasheet, the "Recommended Operating Conditions" voltage is 3.0 to 3.6V, which is a bit tight TBH. It's above the max BOD threshold.
> If you look at the original ESP-WROVER datasheet, which used 1.8V flash, the "Recommended Operating Conditions" are much less restrictive.

The main difference is that, ESP-Wrover uses 1.8V flash inside, while other modules, ESP-Wroom, ESP-Wrover-B/E uses 3.3V flash. 

The 3.3V power is supplied via the input of ESP32 via a small resistor, while the 1.8V power is supplied via an internal regulator, which can tolerate lower input.

ginkgm, 2023-06-09:
Seeing from the TRM: https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

I know your requirements that knowing about a BOD reset is useful. But..

section 3.1.2, your reset seems to be a `System Reset` level one,  caused by `RWDT System Reset`. This will reset all the registers, including the one used to hold the reset reason. To avoid this behavior, you may try to turn it off , see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#rtc-watchdog-timeout. 

But when brownout issue occurs, the behavior become undetermined, you will not be able to do very complex workaround.

More importantly, I'd like to know what causes the RTCWDT reset after the BOD reset. That's why I asked for the FULL log. Why the console output cannot be provided?

VitorBFreitas, 2023-06-09:
@ginkgm Are all necessary information in this log?

```
ets Jul 29 2019 12:21:46

rst:0xf (RTCWDT_BROWN_OUT_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7008
load:0x40078000,len:15684
ho 0 tail 12 room 4
load:0x40080400,len:4516
0x40080400: _init at ??:?

entry 0x400806b8
I (30) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (30) boot: compile time 08:36:31
I (30) boot: chip revision: v3.0
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 1
I (40) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (50) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (83) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (90) boot:  2 app0             OTA app          00 10 00010000 001b0000
I (98) boot:  3 app1             OTA app          00 11 001c0000 001b0000
I (105) boot:  4 spiffs           Unknown data     01 82 00370000 00060000
I (113) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (120) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (128) boot: End of partition table
I (132) boot_comm: chip revision: 3, min. application chip revision: 1
I (139) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3b1d0h (242128) map
��וBets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7008
load:0x40078000,len:15684
ho 0 tail 12 room 4
load:0x40080400,len:4516
0x40080400: _init at ??:?

entry 0x400806b8
I (29) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (29) boot: compile time 08:36:31
I (30) boot: chip revision: v3.0
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 1
I (40) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (50) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (82) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (90) boot:  2 app0             OTA app          00 10 00010000 001b0000
I (97) boot:  3 app1             OTA app          00 11 001c0000 001b0000
I (105) boot:  4 spiffs           Unknown data     01 82 00370000 00060000
I (112) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (120) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (127) boot: End of partition table
I (132) boot_comm: chip revision: 3, min. application chip revision: 1
I (139) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3b1d0h (242128) map
I (214) esp_image: segment 1: paddr=0004b1f8 vaddr=3ffb0000 size=04e20h ( 20000) load
I (221) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=e95ech (955884) map
I (484) esp_image: segment 3: paddr=00139614 vaddr=3ffb4e20 size=00be8h (  3048) load
I (485) esp_image: segment 4: paddr=0013a204 vaddr=40080000 size=15c80h ( 89216) load
I (519) esp_image: segment 5: paddr=0014fe8c vaddr=00000000 size=000f4h (   244) 
I (530) boot: Loaded app from partition at offset 0x10000
I (530) boot: Disabling RNG early entropy source...
I (542) cpu_start: Pro cpu up.
I (542) cpu_start: Starting app cpu, entry point is 0x400811b4
0x400811b4: call_start_cpu1 at C:/esp/esp-idf/components/esp_system/port/cpu_start.c:149

I (0) cpu_start: App cpu up.
I (556) cpu_start: Pro cpu start user code
I (556) cpu_start: cpu freq: 240000000
I (556) cpu_start: Application information:
I (561) cpu_start: Project name:     mesh-shopBA
I (566) cpu_start: App version:      1.3.3
I (571) cpu_start: Compile time:     Jun  5 2023 13:15:21
I (577) cpu_start: ELF file SHA256:  dbb12184e25346a3...
I (583) cpu_start: ESP-IDF:          v4.4.4
I (588) heap_init: Initializing. RAM available for dynamic allocation:
I (595) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (601) heap_init: At 3FFBAC48 len 000253B8 (148 KiB): DRAM
I (607) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (614) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (620) heap_init: At 40095C80 len 0000A380 (40 KiB): IRAM
I (627) spi_flash: detected chip: generic
I (631) spi_flash: flash io: qio
I (635) esp_core_dump_flash: Init core dump to flash
I (641) esp_core_dump_flash: Found partition 'coredump' @ 3d0000 65536 bytes
D (682) esp_core_dump_checksum: Total length of hashed data: 40000
I (682) esp_core_dump_flash: Core dump data checksum is correct
I (684) esp_core_dump_flash: Found core dump 40004 bytes in flash @ 0x3d0000
I (692) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

szmodz, 2023-06-09:
The RTC WDT cannot be disabled in general. It's always activated by the ROM bootloader and the  ROM bootloader timeout cannot be changed.

19.3.1.3 Flash Boot Protection

https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

My workaround is to store the reset hint (RTC_RESET_CAUSE_REG) in RTC SRAM. This memory is not cleared by RTC WDT reset.

esp-idf already uses the RTC SRAM to store timekeeping info.

szmodz, 2023-07-08:
> My workaround is to store the reset hint (RTC_RESET_CAUSE_REG) in RTC SRAM.

Of course that's 100% not foolproof. The BOD threshold remains below the FLASH recommended operating conditions, so while it should work better most times, it's by no means reliable.

ginkgm, 2023-08-08:
Your log at least shows that there is a RTCWDT reset after the brownout reset. But the 1st/2nd bootloader doesn't help to print the PC before reset.. So we don't know what causes the RTCWDT before the reset.

Doubt whether there are some issue in the reset process that causes the 2nd bootloader failed to load the program again (in time) after reboot..