espressif/esp-idf #10835: Bluetooth Controller asserts (rwbt.c at line 381) after enable->disable->enable

mringwal, 2023-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3541-g778aeae99e

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Lyra T v4.3

### Power Supply used.

USB

### What is the expected behavior?

I expect the Bluetooth Controller to work after this sequence:
- esp_bt_controller_init
- esp_bt_controller_enable
- esp_bt_controller_disable
- esp_bt_controller_enable
- initialize the Bluetooth Host Stack (BTstack in this case)

While this enable->disable->enable sequence does not make sense as it is, it's a good test case as it reproduces 100% for me.

In a product, the user would like to disable the Bluetooth Controller to save energy and later enable it again. The crash then happens on the second startup. 

### What is the actual behavior?

During Bluetooth Host Stack startup, the following assert happens:

[00:00ASSERT_PARAM(2048 0), in rwbt.c at line 381
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

### Steps to reproduce.

When using BTstack, add the following lines between the existing call to esp_bt_controller_enable and the call to esp_vhci_host_register_callback:

esp_bt_controller_disable();
esp_bt_controller_enable(bt_mode);

Then start any application.

### Debug Logs.

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5740
ho 0 tail 12 room 4
[... 168 lines cut ...]
Core  0 register dump:
PC      : 0x4008f23f  PS      : 0x00060d34  A0      : 0x80083c31  A1      : 0x3ffc16b8
0x4008f23f: r_assert at /Users/mringwal/Projects/esp32/esp-idf/components/bt/controller/esp32/bt.c:1863

A2      : 0x00000000  A3      : 0x00000800  A4      : 0x00000000  A5      : 0x3f410a98
A6      : 0x0000017d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc1628
A10     : 0x00000000  A11     : 0x3ffc164b  A12     : 0x3ffc15f7  A13     : 0x00000031
A14     : 0x00000000  A15     : 0x3ffc15fc  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40083b39  LEND    : 0x40083b41  LCOUNT  : 0x00000000
0x40083b39: r_assert_param at ??:?

0x40083b41: r_assert_param at ??:?

Backtrace: 0x4008f23c:0x3ffc16b8 |<-CORRUPTED
0x4008f23c: r_assert at /Users/mringwal/Projects/esp32/esp-idf/components/bt/controller/esp32/bt.c:1862

ELF file SHA256: 23bf0c53217ec466

Initiating core dump!
I (1665) esp_core_dump_uart: Press Enter to print core dump to UART...
I (1672) esp_core_dump_uart: Print core dump to uart...
Core dump started (further output muted)
Received  10 kB...
Core dump finished!
```

mringwal, 2023-02-23:
Issue #8405 hits the same assert, but seems unrelated as they don't disable->enable the Bluetooth Controller.

xiongweichao, 2023-04-04:
Hi @mringwal ,

May I ask if it can be changed to the following way?
```
esp_bt_controller_init
esp_bt_controller_enable
esp_bt_controller_disable
esp_bt_controller_deinit
esp_bt_controller_init
esp_bt_controller_enable
initialize the Bluetooth Host Stack (BTstack in this case)
```

Thanks

xiongweichao, 2023-04-04:
Hi @mringwal ,

Please try if this lib can solve the problem.
replace path: $IDF_PATH/components/bt/controller/lib_esp32
[esp32.zip](https://github.com/espressif/esp-idf/files/11148057/esp32.zip)

Thanks

mringwal, 2023-04-04:
Thanks. I will try and let you know.

mringwal, 2023-04-04:
> esp_bt_controller_init
> esp_bt_controller_enable
> esp_bt_controller_disable
> esp_bt_controller_deinit
> esp_bt_controller_init
> esp_bt_controller_enable
> initialize the Bluetooth Host Stack (BTstack in this case)

Hi @xiongweichao. This seems to work, thanks!
Please let me know if you try to get enable-disable-enable to work.
If not, going over de-init is a valid solution for us.

Should esp_bt_controller_mem_release be called a second time if Classic is not used on ESP32:
- esp_bt_controller_mem_release
- esp_bt_controller_init
- esp_bt_controller_enable
- ... use Controller
- esp_bt_controller_disable
- esp_bt_controller_deinit
- .. long pause
- esp_bt_controller_mem_release again?
- esp_bt_controller_init
- esp_bt_controller_enable
.

xiongweichao, 2023-04-06:
Hi @mringwal , 

1. I reproduced this problem and there is no such problem after using the new lib.The new lib version information is as follows.
`BTDM_INIT: BT controller compile version [09b12ae]`

2. esp_bt_controller_mem_release again?
I don't understand you. But I think you can refer to [this.](https://github.com/espressif/esp-idf/blob/42261df71f12b0b995774d103ce68e40e8fba529/components/bt/include/esp32/include/esp_bt.h#L438)

Thanks

mringwal, 2023-04-06:
Hi @xiongweichao 

Thanks for checking. Sorry, I must have made a mistake yesterday. Just tried again and the Controller behaves correctly both with enable/disable/enable as well as enable/disabled/deinit/init/enable.  

Great!

Could you ping me here once it was merged into master? Thanks again!

closed 2023-04-17 by espressif-bot