espressif/esp-idf #10837: tusb_msc Sample fails when GPIO4 is low

franz-ms-muc, 2023-02-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc

This Sample runs fine on all Devices, unless GPIO4 is pulled low.

if we release GPIO4 then USB Device appears, 
if we pull it to low again it disappears again.

franz-ms-muc, 2023-02-24:
#10727

franz-ms-muc, 2023-02-24:
Resolved: with commit Meisterschulen-am-Ostbahnhof-Munchen/tusb\_msc@c839a1a

closed 2023-02-24 by franz-ms-muc

esp-saurabhbansal, 2023-04-14:
Hello,

Just for your information, We have recently released USB Device MSC driver. Please check if it is useful to you. We highly appreciate the feedback.

The driver makes ESP based device recognizable by USB-hosts as a USB Mass Storage Device.
It either allows the embedded application to access the partition or Host PC accesses the partition over USB MSC. They can't be allowed to access the partition at the same time.

The driver supports storage media of two types:
1. SPI Flash
2. SD MMC Card

Driver can be found [here](https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/tusb_msc_storage.c).
Sample application can be found [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc).

Thanks.