espressif/esp-idf #10838: Not rebooting after crash

QuentinFarizon, 2023-02-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-DevkitM1

### Power Supply used.

USB

### What is the expected behavior?

On a crash, the board is supposed to reboot.

### What is the actual behavior?

Log output ends in "Rebooting...", but it doesn't actually reboot.

### Steps to reproduce.

Call assert(0); somewhere.

### Debug Logs.

```plain
assert failed: peer_disc_complete peer.c:83 (0)

Backtrace: 0x40376036:0x3fcbf0c0 0x4038049d:0x3fcbf0e0 0x403867f5:0x3fcbf100 0x4200b0a6:0x3fcbf220 0x4200b382:0x3fcbf240 0x42015899:0x3fcbf260 0x420158cf:0x3fcbf280 0x42015f41:0x3fcbf2a0 0x42016d69:0x3fcbf2d0 0x42019403:0x3fcbf2f0 0x4201944c:0x3fcbf350 0x420152da:0x3fcbf3b0 0x420154b9:0x3fcbf3d0 0x42014b4d:0x3fcbf3f0 0x4037ea05:0x3fcbf410 0x40377dfd:0x3fcbf430 0x42009aa6:0x3fcbf450
0x40376036: panic_abort at /home/quentin/esp/esp-idf/components/esp_system/panic.c:412
(...)

ELF file SHA256: 2d91cc9f437f536d

Rebooting...

```

Then nothing more
```

igrr, 2023-02-24:
Could you please try updating IDF to v5.0.1 release? From the [release notes](https://github.com/espressif/esp-idf/releases/tag/v5.0.1), it seems that the issue might be fixed there:

* Panic: Fixed bug where reboot is stuck after the ESP32-S3 core1 enters the panic (https://github.com/espressif/esp-idf/commit/c280d4b56e1429ff9aa1f7f8bf6afa6381a099f7, https://github.com/espressif/esp-idf/issues/10320)

QuentinFarizon, 2023-02-24:
Perfect

closed 2023-02-24 by QuentinFarizon