espressif/esp-idf #10840: ESP32-WROOM doesn't work TCP connection in some routers

castrofernando, 2023-02-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.0.1-Release

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Currently, i'm using the simple HTTP Request example provided by espressif in the latest release 5.x. I expected that it will be able to connect to wifi and do the basic 4 request verb (GET, POST, PUT and DELETE) to any web server.

### What is the actual behavior?

Currently, the ESP32-WROOM connect to the router, get the DHCP ip properly, but fail all requests when try to connect to the host. It fail to connect to the socket always i'm using the router FastLink FLT718 using 2.4ghz network (https://fastlinktelecom.com.br/roteador-lte-4g/) . I already checked the connection , DNS. All ok.
 

### Steps to reproduce.

Just run the example in this router model. There are others i also got problem like this, but didn't remember the exactly model. I also have tried using arduino, the same behavior. It connect to the internet , when it does the tcp connection it fails to connect always. 
If i ran the exactly same code in other router, it works without any change, some others router fail.

### Debug Logs.

```plain
it doesn't crash, it only don't connect with the socket (I already checked, and the router has internet and all my others home devices connect on it and perform properly (e.g: cellphone, notebook, etc.)

(4353) esp_netif_handlers: example_netif_sta ip: 192.168.1.176, mask: 255.255.255.0, gw: 192.168.1.1
I (4353) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.1.176
I (4673) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:0a3a:f2ff:fe4f:1b54, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4673) example_common: Connected to example_netif_sta
I (4683) example_common: - IPv4 address: 192.168.1.176,
I (4683) example_common: - IPv6 address: fe80:0000:0000:0000:0a3a:f2ff:fe4f:1b54, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4693) HTTP_CLIENT: Connected to AP, begin http example
I (4713) wifi:<ba-add>idx:0 (ifx:0, a4:d4:b2:6a:f8:ff), tid:0, ssn:2, winSize:64
E (10193) esp-tls: [sock=54] select() timeout
E (10193) transport_base: Failed to open a new connection: 32774
E (10193) HTTP_CLIENT: Connection failed, sock < 0
E (10193) HTTP_CLIENT: HTTP GET request failed: ESP_ERR_HTTP_CONNECT
E (15223) esp-tls: [sock=54] select() timeout
E (15223) transport_base: Failed to open a new connection: 32774
E (15223) HTTP_CLIENT: Connection failed, sock < 0
E (15223) HTTP_CLIENT: HTTP POST request failed: ESP_ERR_HTTP_CONNECT
E (20253) esp-tls: [sock=54] select() timeout
E (20253) transport_base: Failed to open a new connection: 32774
E (20253) HTTP_CLIENT: Connection failed, sock < 0
E (20253) HTTP_CLIENT: HTTP PUT request failed: ESP_ERR_HTTP_CONNECT
```

### More Information.

At first, i thought could be some problem with old sdk running in arduino. Then i resolved to test example in lates IDF SDK 5.0 release version, and same problem happens. It fail when try to connect tcp with any port 80, 443, etc.

ESP-YJM, 2023-02-27:
@castrofernando It seems the TCP SYN connection is time-out.  I am not sure if the server received SYN packets or if the server replied with SYN+ACK, ESP32 did not receive. Could you please help capture the TCP three-way handshake packet, preferably can provide a comparison of failure and success packet.

castrofernando, 2023-03-01:
Dear @ESP-YJM ,

I've tested using wireshark doing a simple get to the server. For demo purpose i didn't sent any response to ESP. But as you can see below, at the router that it works the ESP32(192.168.15.3) send a SYN and server(192.168.15.230) response SYN + ACK
[image]

At the router where connection doesn't work(FastLink FLT718), if i point it to my local network, it works. Follow the TCP packages. Seems same of above:
[image]

Unfortunatly, i can't debug using wireshak to debug in the internet, since i dont have a switch with port mirroring or hub.

Could it be related to latency? I'm sure both routers has internet, but for any reason ESP got problem to do the handshake connetion at the FastLink FLT718 (pc, mobile, others pcbs, all works fine). Anything can i try to investigate?

ESP-YJM, 2023-03-01:
@castrofernando I think you can use PC to visit the http url by internet with FastLink FLT718. And capture the packet in PC with wireshark. The PC need connect FastLink FLT718 with wifi in 2.4G channel.

castrofernando, 2023-03-01:
@ESP-YJM i have monitored using my pc to do a GET request at the URL. See the log below:
[image]

I also tried to do same with some urls from ESPRESSIF example, and before you see the result: GET -> (http://httpbin.org/)
[image]

Then, below the test with same URL  GET -> (http://httpbin.org/) , but in a router that it works properly:
[image]

**At the router where ESP doesn't work, it there is a answer [PSH,ACK].**

ESP-YJM, 2023-03-02:
@castrofernando Could you please disable your IPv6 address in your PC and try again. Besides, i need the full capture file, not screenshots. Thanks.

castrofernando, 2023-03-02:
Dear @ESP-YJM , i guess i have found the problem with your help, but i don't know how to solve. I also tested others hardware at same router and problem still same. It's seems that it only can get internet if IPV6 is enabled. If i disable IPv6 in my computer, it also stop to work. If i connect my pc at router that esp32 also works with ipv6 disabled, i have no problem..all works.
So, i imagine that for any reason, this router need that hardware have IPv6 to works.
[Wireshark packets.zip](https://github.com/espressif/esp-idf/files/10876590/Wireshark.packets.zip)

In attachment the packets captured with IPv6 enabled and with IPv6 disabled. Any other test i can do to try to do it works? Please, use filter tcp.port == 80 to see the specific logs of my test

ESP-YJM, 2023-03-03:
@castrofernando Thanks for your useful information. I check the packet that with IPv4, it can't connect to. The SYN sent by PC has no response and reach the max SYN retry. This phenomenon is the same as ESP32. So i guess the internet access with IPv4 is not available. You need check your Internet provider whether support IPv4 network? Or your router whether disable IPv4 network in WAN.

closed 2023-05-19 by ESP-YJM