espressif/esp-idf #10841: esp32-s2使用tinyusb的问题

Szeroy, 2023-02-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1.在menuconfig里面看到最多只能创建出两个usb设备，如何才能创建出更多的设备？
2.在串口调试助手里面显示的是usb串行设备，如何才能修改名称？

Szeroy, 2023-02-25:
我想要虚拟多个串口出来做多合一的调试器用

peter-marcisovsky, 2025-01-21:
Hi @Szeroy is this still a relevant issue?

> 1.在menuconfig里面看到最多只能创建出两个usb设备，如何才能创建出更多的设备？

If I got it correctly, you want to create a composite USB Device with multiple USB interfaces? You can use [tusb_composite_msc_serialdevice](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_composite_msc_serialdevice) example here. But keep in mind, you will be limited by number of avaliable [endpoints](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/peripherals/usb_device.html#features) on esp32s3 (or esp32s2) 

> 2.在串口调试助手里面显示的是usb串行设备，如何才能修改名称？

Can you provide some info (screenshot) of the debugger you are using?

Indastri, 2025-03-27:
No response. If you are encountering the issue please feel free to reopen.

Cheers!

closed 2025-03-27 by Indastri