espressif/esp-idf #10842: qsort() in app_main() run error?

WangShuoran, 2023-02-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When I use ESP32S3 with IDFv4.4.3 to run qsort() have error, That is my test code:
```
#include <stdio.h>
#include <stdlib.h>
typedef struct
{
    int date;
    int value;
}X_S;

/* a->z */
int cmpfunA2Z(const void * a, const void * b)
{
    X_S * p1 = (X_S *)a;
    X_S * p2 = (X_S *)b;

    return p1->value > p2->value;
}

/* z -> a */
int cmpfunZ2A(const void * a, const void * b)
{
    X_S * p1 = (X_S *)a;
    X_S * p2 = (X_S *)b;

    return p1->value < p2->value;
}

int quickSortOfCpp()
{
    X_S Xlist[100];
    int Ilist[100];
    int i = 0;
    for(i = 0; i < 100; i++)
    {
        Xlist[i].date = i+1;
        double t = (double)i - 50.3;
        Xlist[i].value = (int)(t * t + 5.6);
    }

    for(i = 0; i < 100; i++)
    {
        printf("num : %3d, value : %4d\n", Xlist[i].date, Xlist[i].value);
    }

    qsort(Xlist, 100, sizeof(X_S), cmpfunA2Z);
    printf("\033[034m-------------sorted-------------\033[0m\n");

    for(i = 0; i < 100; i++)
    {
        printf("num : %3d, value : %4d\n", Xlist[i].date, Xlist[i].value);
    }

    return 0;
}

int app_main(void)
{
    quickSortOfCpp();
    return 0;
}
```
When I use Linux or online runner is OK, I don't know whether or not the ESP-IDF's gcc can not run this function. I fine #9018 somebody use [musl](https://musl.libc.org/), I don't know why too use musl.

That is in esp32s3 output:
```
num :   1, value :  2535
num :   2, value :  2436
num :   3, value :  2338
num :   4, value :  2242
num :   5, value :  2149
num :   6, value :  2057
num :   7, value :  1968
num :   8, value :  1880
num :   9, value :  1794
num :  10, value :  1711
num :  11, value :  1629
num :  12, value :  1550
num :  13, value :  1472
num :  14, value :  1396
num :  15, value :  1323
num :  16, value :  1251
num :  17, value :  1182
num :  18, value :  1114
num :  19, value :  1048
num :  20, value :   985
num :  21, value :   923
num :  22, value :   864
num :  23, value :   806
num :  24, value :   750
num :  25, value :   697
num :  26, value :   645
num :  27, value :   596
num :  28, value :   548
num :  29, value :   502
num :  30, value :   459
num :  31, value :   417
num :  32, value :   378
num :  33, value :   340
num :  34, value :   304
num :  35, value :   271
num :  36, value :   239
num :  37, value :   210
num :  38, value :   182
num :  39, value :   156
num :  40, value :   133
num :  41, value :   111
num :  42, value :    92
num :  43, value :    74
num :  44, value :    58
num :  45, value :    45
num :  46, value :    33
num :  47, value :    24
num :  48, value :    16
num :  49, value :    10
num :  50, value :     7
num :  51, value :     5
num :  52, value :     6
num :  53, value :     8
num :  54, value :    12
num :  55, value :    19
num :  56, value :    27
num :  57, value :    38
num :  58, value :    50
num :  59, value :    64
num :  60, value :    81
num :  61, value :    99
num :  62, value :   120
num :  63, value :   142
num :  64, value :   166
num :  65, value :   193
num :  66, value :   221
num :  67, value :   252
num :  68, value :   284
num :  69, value :   318
num :  70, value :   355
num :  71, value :   393
num :  72, value :   434
num :  73, value :   476
num :  74, value :   520
num :  75, value :   567
num :  76, value :   615
num :  77, value :   666
num :  78, value :   718
num :  79, value :   772
num :  80, value :   829
num :  81, value :   887
num :  82, value :   948
num :  83, value :  1010
num :  84, value :  1074
num :  85, value :  1141
num :  86, value :  1209
num :  87, value :  1280
num :  88, value :  1352
num :  89, value :  1426
num :  90, value :  1503
num :  91, value :  1581
num :  92, value :  1662
num :  93, value :  1744
num :  94, value :  1828
num :  95, value :  1915
num :  96, value :  2003
num :  97, value :  2094
num :  98, value :  2186
num :  99, value :  2280
num : 100, value :  2377
-------------sorted-------------
num :  51, value :     5
num :   2, value :  2436
num :   3, value :  2338
num :   4, value :  2242
num :   5, value :  2149
num :   6, value :  2057
num :   7, value :  1968
num :   8, value :  1880
num :   9, value :  1794
num :  10, value :  1711
num :  11, value :  1629
num :  12, value :  1550
num :  13, value :  1472
num :  14, value :  1396
num :  15, value :  1323
num :  16, value :  1251
num :  17, value :  1182
num :  18, value :  1114
num :  19, value :  1048
num :  20, value :   985
num :  21, value :   923
num :  22, value :   864
num :  23, value :   806
num :  24, value :   750
num :  25, value :   697
num :  26, value :   645
num :  27, value :   596
num :  28, value :   548
num :  29, value :   502
num :  30, value :   459
num :  31, value :   417
num :  32, value :   378
num :  33, value :   340
num :  34, value :   304
num :  35, value :   271
num :  36, value :   239
num :  37, value :   210
num :  38, value :   182
num :  39, value :   156
num :  40, value :   133
num :  41, value :   111
num :  42, value :    92
num :  43, value :    74
num :  44, value :    58
num :  45, value :    45
num :  46, value :    33
num :  47, value :    24
num :  48, value :    16
num :  49, value :    10
num :  50, value :     7
num :   1, value :  2535
num :  52, value :     6
num :  53, value :     8
num :  54, value :    12
num :  55, value :    19
num :  56, value :    27
num :  57, value :    38
num :  58, value :    50
num :  59, value :    64
num :  60, value :    81
num :  61, value :    99
num :  62, value :   120
num :  63, value :   142
num :  64, value :   166
num :  65, value :   193
num :  66, value :   221
num :  67, value :   252
num :  68, value :   284
num :  69, value :   318
num :  70, value :   355
num :  71, value :   393
num :  72, value :   434
num :  73, value :   476
num :  74, value :   520
num :  75, value :   567
num :  76, value :   615
num :  77, value :   666
num :  78, value :   718
num :  79, value :   772
num :  80, value :   829
num :  81, value :   887
num :  82, value :   948
num :  83, value :  1010
num :  84, value :  1074
num :  85, value :  1141
num :  86, value :  1209
num :  87, value :  1280
num :  88, value :  1352
num :  89, value :  1426
num :  90, value :  1503
num :  91, value :  1581
num :  92, value :  1662
num :  93, value :  1744
num :  94, value :  1828
num :  95, value :  1915
num :  96, value :  2003
num :  97, value :  2094
num :  98, value :  2186
num :  99, value :  2280
num : 100, value :  2377
```

WangShuoran, 2023-02-25:
And I use qsort_r() but the output is still error. I don't know why.

igrr, 2023-02-25:
I think the issue is that your comparison callback doesn't satisfy the requirements for qsort function. From the [docs](https://pubs.opengroup.org/onlinepubs/009695399/functions/qsort.html):

> The contents of the array shall be sorted in ascending order according to a comparison function. The compar argument is a pointer to the comparison function, which is called with two arguments that point to the elements being compared. The application shall ensure that the function returns an integer **less than, equal to, or greater than 0, if the first argument is considered respectively less than, equal to, or greater than the second**. If two members compare as equal, their order in the sorted array is unspecified.

If you modify the comparison callback to meet this requirement, the result will be sorted as you expect:

```c++
int cmpfunA2Z(const void * a, const void * b)
{
    X_S * p1 = (X_S *)a;
    X_S * p2 = (X_S *)b;

    return p1->value - p2->value;
}
```

WangShuoran, 2023-02-27:
Thank you. You are right, I try to find doc with qsort(), Thank you give me the URL.

closed 2023-02-27 by WangShuoran