espressif/esp-idf #10843: Can  "esp_adc_cal_characterize" configure ADC_UNIT1 with two attens?

casitelwithdean, 2023-02-25:
### Is your feature request related to a problem?

I am using two channels of ADC_UINT1,I want use ADC_ATTEN_DB_11 for one and ADC_ATTEN_DB_0 for the other.  But how to use "esp_adc_cal_characterize "to configure two attens?
Thanks !

Icarus113, 2023-02-27:
Hi, @casitelwithdean . Which chip and which idf version are you using? If you're using v4.4, you may refer to [this](https://github.com/espressif/esp-idf/blob/release/v4.4/examples/peripherals/adc/single_read/single_read/main/single_read.c#L68), there you call this `esp_adc_cal_characterize` twice, with 2 different attens. Then you'll get two different set of characters. And when doing calibration, you just give the characters to `esp_adc_cal_raw_to_voltage` respectively.

casitelwithdean, 2023-02-27:
I am using esp32s3wroom1   idf4.4.3
Thanks very much

casitelwithdean, 2023-02-27:
> Hi, @casitelwithdean . Which chip and which idf version are you using? If you're using v4.4, you may refer to [...]
I am using esp32s3wroom1 idf4.4.3
Thanks very much