espressif/esp-idf #10844: Simple gpio output not working

devotip, 2023-02-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v4.4.4-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32 wroom

### Power Supply used.

USB

### What is the expected behavior?

with the provided sample I expect a squarewave on pin 21 period 200ms

### What is the actual behavior?

every 100ms there is a short bit stream burst every 100ms with an apparent clock of 100kHz

### Steps to reproduce.

1. run provided sample
2. check pin21 with oscilloscope
[gpio21.zip](https://github.com/espressif/esp-idf/files/10831636/gpio21.zip)

### Debug Logs.

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6668
ho 0 tail 12 room 4
load:0x40078000,len:14944
load:0x40080400,len:3816
0x40080400: _init at ??:?

entry 0x40080698
I (29) boot: ESP-IDF v4.4.4-dirty 2nd stage bootloader
I (29) boot: compile time 16:35:42
I (29) boot: chip revision: v3.0
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=086e4h ( 34532) map
I (125) esp_image: segment 1: paddr=0001870c vaddr=3ffb0000 size=02430h (  9264) load
I (128) esp_image: segment 2: paddr=0001ab44 vaddr=40080000 size=054d4h ( 21716) load
I (140) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=15bb0h ( 89008) map
I (173) esp_image: segment 4: paddr=00035bd8 vaddr=400854d4 size=063b4h ( 25524) load
I (189) boot: Loaded app from partition at offset 0x10000
I (189) boot: Disabling RNG early entropy source...
I (201) cpu_start: Pro cpu up.
I (201) cpu_start: Starting app cpu, entry point is 0x4008102c
0x4008102c: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (215) cpu_start: Pro cpu start user code
I (215) cpu_start: cpu freq: 160000000
I (215) cpu_start: Application information:
I (220) cpu_start: Project name:     generic_gpio
I (225) cpu_start: App version:      v4.4.4-dirty
I (230) cpu_start: Compile time:     Feb 25 2023 16:35:23
I (236) cpu_start: ELF file SHA256:  dd549a210dccab57...
I (242) cpu_start: ESP-IDF:          v4.4.4-dirty
I (248) heap_init: Initializing. RAM available for dynamic allocation:
I (255) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (261) heap_init: At 3FFB2D18 len 0002D2E8 (180 KiB): DRAM
I (267) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (274) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (280) heap_init: At 4008B888 len 00014778 (81 KiB): IRAM
I (288) spi_flash: detected chip: generic
I (291) spi_flash: flash io: dio
W (295) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (309) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (319) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
cnt: 0
cnt: 1
cnt: 2
cnt: 3
cnt: 4
cnt: 5
cnt: 6
```

devotip, 2023-02-25:
see also https://github.com/espressif/esp-idf/issues/10816

devotip, 2023-02-27:
see also https://github.com/espressif/esp-idf/issues/10830

closed 2023-03-01 by devotip

devotip, 2023-03-01:
misplaced wire

Divraj-Dev, 2023-03-05:
Hi @devotip 

Were we able to find a solution?