espressif/esp-idf #10847: ESP MDNS doesn't respond

TheElectronicMusicProject, 2023-02-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

I attach my simple project. [15_2_mdns.zip](https://github.com/espressif/esp-idf/files/10834084/15_2_mdns.zip)

I created a simple application where the esp32 connects to a router. On the other hand I connect to the same router.
If I try to use the given IP address that I see on the terminal, `192.168.1.104` in my case, it works well and it prints "hello world" as a web page.

I added a MDNS with "esp32" as hostname. So I expect I can connect to it through `http://esp32.local`.

I want to underline that, to achieve this, I wrote the following:
```c
static void
start_mdns_service (void)
{
    ESP_ERROR_CHECK(mdns_init());
    ESP_ERROR_CHECK(mdns_hostname_set("esp32"));
    ESP_ERROR_CHECK(mdns_instance_name_set("Learn esp32 thing"));

    if (false == mdns_hostname_exists("esp32"))
    {
        ESP_LOGE(TAG, "Error on hostname");
    }
}
```

### What is the actual behavior?

Insted of working, it appears an error of page not found.
I tried different browsers (Firefox and Chrome) and a different device (my smartphone) but anything seems to work.
In particular, I also tried an example of Arduino IDE for MDNS applications, but same problem.

If I use the given IP address it works well and responds.

If I try `ping 192.168.1.104` from the command prompt it responds, but `ping esp32.local` or `ping esp32` says that it is impossible to find the host.

I switched to another ESP32, same problem.

### Steps to reproduce.

If I use the given IP address it works well and responds.

If I try `ping 192.168.1.104` from the command prompt it responds, but `ping esp32.local` or `ping esp32` says that it is impossible to find the host.

I switched to another ESP32, same problem.

### Debug Logs.

```plain
I (0) cpu_start: App cpu up.
I (406) cpu_start: Pro cpu start user code
I (406) cpu_start: cpu freq: 160000000
I (406) cpu_start: Application information:
I (411) cpu_start: Project name:     15_2_mdns
I (416) cpu_start: App version:      3ee9388-dirty
I (421) cpu_start: Compile time:     Feb 26 2023 15:57:17
I (428) cpu_start: ELF file SHA256:  9b7f3ec30bb5bc88...
I (433) cpu_start: ESP-IDF:          -128-NOTFOUND
I (439) heap_init: Initializing. RAM available for dynamic allocation:
I (446) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (452) heap_init: At 3FFB7C20 len 000283E0 (160 KiB): DRAM
I (458) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (465) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (471) heap_init: At 40094600 len 0000BA00 (46 KiB): IRAM
I (479) spi_flash: detected chip: generic
I (482) spi_flash: flash io: dio
I (488) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (560) wifi:wifi driver task: 3ffc086c, prio:23, stack:6656, core=0
I (560) system_api: Base MAC address is not set
I (560) system_api: read default base MAC address from EFUSE
I (570) wifi:wifi firmware version: 71cb2c8
I (570) wifi:wifi certification version: v7.0
I (570) wifi:config NVS flash: enabled
I (570) wifi:config nano formating: disabled
I (580) wifi:Init data frame dynamic rx buffer num: 32
I (580) wifi:Init management frame dynamic rx buffer num: 32
I (590) wifi:Init management short buffer num: 32
I (590) wifi:Init dynamic tx buffer num: 32
I (600) wifi:Init static rx buffer size: 1600
I (600) wifi:Init static rx buffer num: 10
I (600) wifi:Init dynamic rx buffer num: 32
I (610) wifi_init: rx ba win: 6
I (610) wifi_init: tcpip mbox: 32
I (620) wifi_init: udp mbox: 6
I (620) wifi_init: tcp mbox: 6
I (620) wifi_init: tcp tx win: 5744
I (630) wifi_init: tcp rx win: 5744
I (630) wifi_init: tcp mss: 1440
I (640) wifi_init: WiFi IRAM OP enabled
I (640) wifi_init: WiFi RX IRAM OP enabled
I (650) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (750) wifi:mode : sta (ec:94:cb:7b:2b:a4)
I (750) wifi:enable tsf
I (750) wifi: connecting...
I (770) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (1510) wifi:state: init -> auth (b0)
I (1520) wifi:state: auth -> assoc (0)
I (1530) wifi:state: assoc -> run (10)
I (1660) wifi:connected with TP-Link_FD14, aid = 4, channel 1, 40U, bssid = 54:af:97:96:fd:14
I (1670) wifi:security: WPA2-PSK, phy: bgn, rssi: -63
I (1670) wifi:pm start, type: 1

I (1670) wifi: connected
W (1690) wifi:<ba-add>idx:0 (ifx:0, 54:af:97:96:fd:14), tid:0, ssn:3, winSize:64
I (1730) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3040) wifi: got ip
I (3040) esp_netif_handlers: sta ip: 192.168.1.104, mask: 255.255.255.0, gw: 192.168.1.1
W (156160) wifi:<ba-del>idx
```
```

ESP-YJM, 2023-02-27:
@TheElectronicMusicProject You need to make sure that your windows PC supports mdns query. In order to check this, You can install wireshark to capture packets and check whether your computer support multicast domain name resolution. Of course, you can also provide the packet attached to this github issue, we glad to help you analyze.

TheElectronicMusicProject, 2023-02-27:
> @TheElectronicMusicProject You need to make sure that your windows PC supports mdns query. In order to check this, You can install wireshark to capture packets and check whether your computer [...]

Today I tried the same application with a different computer connected to a different network (also the ESP32 was connected to this network). It worked without problems.

I thought it could be a problem due to my router's properties: 30 minutes ago I have tested this project in the network I was using yesterday and it **worked without problems**!!! Without touching any parameter.

I'm going to close this issue, however I'd like to understand why this happened to me... Any idea?

closed 2023-02-27 by TheElectronicMusicProject